robotology/gz-sim-yarp-plugins #28: Add install step

traversaro, 2023-08-08:
We should calls to the [install CMake function](https://cmake.org/cmake/help/latest/command/install.html), and document on which environment variables to modify to ensure that Gazebo can find the installed plugins.

traversaro, 2023-08-08:
Probably we can just install the plugin using the same command used in the gazebo-yarp-plugins repo (https://github.com/robotology/gazebo-yarp-plugins/blob/9ce261e8c0f3433d48de8032820afabc0b33afcb/cmake/AddGazeboYarpPluginTarget.cmake#L77):
~~~
# Add install target
install(TARGETS ${GAZEBO_PLUGIN_LIBRARY_NAME}
        RUNTIME DESTINATION "${CMAKE_INSTALL_BINDIR}"
        LIBRARY DESTINATION "${CMAKE_INSTALL_LIBDIR}"
        ARCHIVE DESTINATION "${CMAKE_INSTALL_LIBDIR}")
~~~

And then we should document that the user needs to install the `<install_prefix>/lib`  or `<install_prefix>/bin` to `GZ_SIM_SYSTEM_PLUGIN_PATH` environment variable.

lucapa17, 2023-08-28:
After having set the `GZ_SIM_SYSTEM_PLUGIN_PATH` environment variable correctly to `<install_prefix>/lib`, we still have the following error:
~~~
Error while loading the library [/usr/local/lib/libgz-sim-yarp-forcetorque-system.so]: libHandler.so: cannot open shared object file: No such file or directory
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gzyarp::ForceTorque]
- Requested library name: [gz-sim-yarp-forcetorque-system]
- Resolved library path: [/usr/local/lib/libgz-sim-yarp-forcetorque-system.so]
Error while loading the library [/usr/local/lib/libgz-sim-yarp-robotinterface-system.so]: libHandler.so: cannot open shared object file: No such file or directory
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gzyarp::RobotInterface]
- Requested library name: [gz-sim-yarp-robotinterface-system]
- Resolved library path: [/usr/local/lib/libgz-sim-yarp-robotinterface-system.so]
~~~

We fixed it setting the `LD_LIBRARY_PATH`  to `<install_prefix>/lib`, but maybe there is a better solution.
@traversaro what do you suggest?

traversaro, 2023-08-28:
Can you change the name of libHandler.so to be something more unique? As these files get installed in a location shared by all libraries, `libHandler.so` is a bit generic name and prome to name collisions. Perhaps we can use `gz-sim-yarp-handler` (note that it does not end in `-system` as this shared library is not a system)?

For the linking error, probably you can just enable rpath as done in https://github.com/robotology/how-to-export-cpp-library/blob/afb21efb655e7b1cc11636c3d42aad9e02fb626f/CMakeLists.txt#L106 ? Note that you can use the update file at https://github.com/robotology/ycm/blob/master/modules/AddInstallRPATHSupport.cmake .