robotology/gz-sim-yarp-plugins #40: Test if LIBGL_ALWAYS_SOFTWARE=1 is still necessary on WSLg and if it is not necessary anymore get rid of it

traversaro, 2023-08-18:
CMake code and READMEs are full of references to `LIBGL_ALWAYS_SOFTWARE=1`. However, that is a pretty specific workaround of one setup (Linux on WSLg) that should be solved by gz-rendering 7.4.1 (see https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1683223479). 

I suggest that @lucapa17 @GiacomoBisio do the following:
* 1. Update via `sudo apt update` and `sudo apt upgrade` and verify that gz-rendering 7.4.1 is installed (`apt list --installed | grep gz-rendering` should give you 7.4.1 as version installed)
* 2. Get rid of any `LIBGL_ALWAYS_SOFTWARE` set in `.bashrc` or similar (i.e. `env | grep LIBGL_ALWAYS_SOFTWARE` should not print anything)
* 3. At this point, run gz-sim . If everything works, remove any reference to LIBGL_ALWAYS_SOFTWARE in the repo. If it still does not work, please report the error that you get, as probably the proper solution is to update mesa. In that case, we should document the proper fix for WSLg (i.e. update mesa) instead of setting LIBGL_ALWAYS_SOFTWARE everywhere.

GiacomoBisio, 2023-08-19:
We followed your directions, but we still have the following error:

~~~
giaco@PC-di-Giacomo:/mnt/c/Windows/System32$ gz sim shapes.sdf
libEGL warning: failed to open /dev/dri/renderD128: Permission denied

libEGL warning: failed to open /dev/dri/card0: Permission denied

terminate called after throwing an instance of 'Ogre::UnimplementedException'
  what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at ./RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 685)
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f9c3323875a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f9c332930b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f9c30dcf3e2, in g_main_context_iteration
#28   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f9c30e27257, in
#27   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f9c30dd1d3a, in g_main_context_dispatch
#26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f9c33293a66, in
#25   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f9c3323cf26, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#24   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f9c33239e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#23   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f9c32991712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f9c3326741d, in QObject::event(QEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f9c002a5513, in gz::gui::plugins::RenderWindowItem::Ready()
#20   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f9c002a5120, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
#19   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f9c002b1c7d, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
#18   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f9c002a8c87, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
#17   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f9bf877aa08, in gz::rendering::v7::BaseRenderEngine::CreateScene(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#16   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f9bf878200e, in gz::rendering::v7::BaseRenderEngine::CreateScene(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#15   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f9bf87959fa, in gz::rendering::v7::BaseScene::CreateMaterials()
#14   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f9bbec9e79a, in gz::rendering::v7::Ogre2Material::SetTextureMapImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::PbsTextureTypes)
#13   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f9bbe903224, in Ogre::TextureGpuManager::_waitFor(Ogre::TextureGpu*, bool)
#12   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f9bbe902e4a, in Ogre::TextureGpuManager::_update(bool)
#11   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f9bbe7d055c, in Ogre::ObjCmdBuffer::execute()
#10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f9bbe7ce8cd, in Ogre::ObjCmdBuffer::NotifyDataIsReady::execute()
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f9bbe8f2d9b, in Ogre::TextureFilter::GenerateHwMipmaps::_executeSerial(Ogre::TextureGpu*)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0x7f9bbe279ac3, in
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0x7f9bbe2763da, in
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9c34c9b4d7, in __cxa_throw
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9c34c9b276, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9c34c9b20b, in
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9c34c8fb9d, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9c389747f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9c3898e475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9c389e2a7c, in pthread_kill
Aborted (Signal sent by tkill() 8267 1000)
~~~

We also tried to update mesa drivers.

traversaro, 2023-08-19:
Can you list apt the installed gz-rendering-related packages, in particular: 

~~~
sudo apt list --installed | grep ogre
~~~

and 

~~~
sudo apt list --installed | grep gz-rendering
~~~

?

GiacomoBisio, 2023-08-19:
> ```
> sudo apt list --installed | grep ogre
> ```

Output of the command above:
~~~
libgz-rendering7-ogre1-dev/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre1/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre2-dev/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre2/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libogre-1.9-dev/jammy,now 1.9.0+dfsg1-12.1ubuntu1 amd64 [installed]
libogre-1.9.0v5/jammy,now 1.9.0+dfsg1-12.1ubuntu1 amd64 [installed,automatic]
libogre-next-2.3-dev/unknown,now 2.3.1-8osrf~jammy amd64 [installed]
libogre-next-2.3.0/unknown,now 2.3.1-8osrf~jammy amd64 [installed,automatic]
libogre-next-dev/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed]
libogrenexthlmspbs2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
libogrenexthlmsunlit2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
libogrenextmain2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
libogrenextmeshlodgenerator2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
libogrenextoverlay2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
libogrenextplanarreflections2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
libogrenextsceneformat2.2.5/jammy,now 2.2.5+dfsg3-0ubuntu2 amd64 [installed,automatic]
ruby-progressbar/jammy,now 1.11.0-1 all [installed,automatic]
~~~

> ```
> sudo apt list --installed | grep gz-rendering
> ```

Output of the command above:
~~~
libgz-rendering7-core-dev/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-dev/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre1-dev/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre1/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre2-dev/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7-ogre2/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
libgz-rendering7/unknown,now 7.4.1-1~jammy amd64 [installed,automatic]
~~~

traversaro, 2023-08-19:
Interesting, the version you have installed of ogre-next-2.3 is `2.3.1-8osrf~jammy`, while the version with the fix is https://build.osrfoundation.org/job/ogre-2.3-debbuilder/ . In theory `2.3.1-8osrf` should be newer and contan the fix, but perhaps something went wrong.

traversaro, 2023-12-19:
This should not be necessary anymore: https://community.gazebosim.org/t/wslg-with-gpu-support-available-on-latest-version-of-gazebo-garden-and-harmonic/2360 .