robotology/gz-sim-yarp-plugins #42: Failing in running Camera tests on Conda (Linux and macOS)

GiacomoBisio, 2023-08-19:
There are the following errors:
## conda-forge ubuntu
~~~
Error while loading the library [/home/runner/micromamba/envs/gzyarppluginsdev/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so]: libOpenGL.so.0: cannot open shared object file: No such file or directory
[Err] [RenderEngineManager.cc:534] Failed to load plugin [gz-rendering-ogre2] : couldn't load library on path [/home/runner/micromamba/envs/gzyarppluginsdev/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so].
~~~

## conda-forge macOS
~~~
Err] [Ogre2RenderEngine.cc:629] Unable to load Ogre Plugin[/Users/runner/micromamba/envs/gzyarppluginsdev/lib/OGRE-Next]. Rendering will not be possible.Make sure you have installed OGRE properly.
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Out of GPU memory or driver refused.
~~~

The problem is the same for the Laser tests #41.
See also #38 for other information.

traversaro, 2023-08-21:
If this is true I am not sure why it is not a problem with apt dependencies, but perhaps we should call the `ServerConfig::SetHeadlessRendering(/*_headless=*/true)` method (see https://gazebosim.org/api/sim/7/headless_rendering.html)?

lucapa17, 2023-08-26:
> If this is true I am not sure why it is not a problem with apt dependencies, but perhaps we should call the `ServerConfig::SetHeadlessRendering(/*_headless=*/true)` method (see https://gazebosim.org/api/sim/7/headless_rendering.html)?

If we try to use headless rendering, we have the following error both for Camera and Laser:
~~~
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device EGL_EXT_device_drm EGL_EXT_device_drm_render_node #0 /dev/dri/card0 in EGLSupport::getGLDisplay at ./RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 320)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1131]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[Err] [Ogre2RenderEngine.cc:1139] Unable to create the rendering window after [11] attempts.
[Err] [Ogre2RenderEngine.cc:1048] Failed to create dummy render window.
[Err] [Ogre2RenderEngine.cc:1049] Please see the troubleshooting page for possible fixes: https://gazebosim.org/docs/fortress/troubleshooting
Stack trace (most recent call last) in thread 81308:
#22   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#21   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6a60dbf9ff, in 
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6a60d2db42, in 
#19   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f6a5d2ff252, in 
#18   Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-sensors-system.so", at 0x7f6a2dffbc29, in gz::sim::v7::systems::SensorsPrivate::RenderThread()
#17   Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-sensors-system.so", at 0x7f6a2dffa21f, in gz::sim::v7::systems::SensorsPrivate::WaitForInit()
#16   Object "/lib/x86_64-linux-gnu/libgz-sim7-rendering.so.7", at 0x7f6a2df51c20, in gz::sim::v7::RenderUtil::Init()
#15   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f6a2ddb0c2b, in gz::rendering::v7::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#14   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f6a2ddb0a0e, in gz::rendering::v7::RenderEngineManagerPrivate::Engine(EngineInfo, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_tra [... cut ...] har_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x220])
Segmentation fault
~~~