robotology/gz-sim-yarp-plugins #45: Familiarize with the concept of YARP devices and other related concepts (xela-95 version)

traversaro, 2023-12-20:
From: https://github.com/robotology/gz-sim-yarp-plugins/issues/8 .

References:
* Day 1 and Day 2 of https://github.com/icub-training/icub-training.github.io/wiki/Online-Program
* Section 4 of https://www.frontiersin.org/articles/10.3389/frobt.2016.00024/full 
* https://yarp.it/latest/group__note__devices.html
* http://wiki.icub.org/wiki/File:Icub_control_modes_slides_1_1.pdf
* http://wiki.icub.org/wiki/File:ICub_Control_Modes_1_1.pdf
* https://github.com/vvv-school/tutorial_joint-interface
* https://yarp.it/latest//group__nws__and__nwc__architecture.html

Questions to answer: 
* *Why some code in the iCub infrastructure is encapsulated using YARP devices that communicate over C++ interface rather than RFModule that communicate using YARP ports?* 

Assignment to do: 
*  https://github.com/vvv-school/tutorial_joint-interface

traversaro, 2023-12-20:
cc @xela-95

traversaro, 2023-12-20:
> https://github.com/vvv-school/tutorial_joint-interface

The idea is to do this tutorial by installing dependencies via conda (https://github.com/robotology/robotology-superbuild/blob/master/doc/install-mambaforge.md).

xela-95, 2023-12-22:
In order to do the tutorials in https://github.com/icub-training/icub-training.github.io/wiki/Online-Program I have created the following conda environment:

```
mamba create -c conda-forge -c robotology -n yarp_tutorials_env  cmake ninja pkg-config cmake compilers yarp icub-contrib-common 
```

xela-95, 2024-01-02:
> * Day 1 and Day 2 of https://github.com/icub-training/icub-training.github.io/wiki/Online-Program

Day 1 lessons and tutorials completed.

xela-95, 2024-01-03:
I've completed the joint interface tutorial https://github.com/vvv-school/tutorial_joint-interface, however I had issues in the setup of the exercise. 

@mebbaid helped me and we understood that some packages were missing from the conda environment, namely `icub-models` and `gazebo-yarp-plugins`. 

Moreover I had to build the source files of the tutorial and run it from the terminal, since I was not able to run them from 'yarpmanager'. It seems it is not able to find them (the log reports: `[ERR] (Tutorial_on_Joint_Interface_App) cannot run tutorial_joint-interface-velocity on localhost : Can't execute command because No such file or directory`).

In the end it was really cool to make icub move!

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/07de455b-7fc3-4a9a-8036-a932e66bf36c

xela-95, 2024-01-03:
While trying the impedance control tutorial https://github.com/vvv-school/tutorial_impedance-control, I installed in the conda environment the missing `idyntree` package and then builded the source files:

```bash
cd <tutorial-path>
mkdir build
cd build
cmake ../
make
```

I obtained the following error:

```log
/home/acroci/repos/tutorial_impedance-control/Module.cpp:30:9: error: invalid use of incomplete type ‘class yarp::os::LogStream’
   30 |         yError() << "convertDegToRad: wrong vector size";
      |         ^~~~~~
In file included from /home/acroci/mambaforge/envs/yarp_tutorials_env/include/yarp/sig/Vector.h:21,
                 from /home/acroci/repos/tutorial_impedance-control/Module.h:5,
                 from /home/acroci/repos/tutorial_impedance-control/Module.cpp:1:
/home/acroci/mambaforge/envs/yarp_tutorials_env/include/yarp/os/Log.h:41:7: note: forward declaration of ‘class yarp::os::LogStream’
   41 | class LogStream;
      |       ^~~~~~~~~
In file included from /home/acroci/mambaforge/envs/yarp_tutorials_env/include/yarp/sig/Vector.h:21,
                 from /home/acroci/repos/tutorial_impedance-control/Module.h:5,
                 from /home/acroci/repos/tutorial_impedance-control/Module.cpp:1:

```

CC @traversaro

traversaro, 2024-01-03:
I guess it is a bug exposes by a a new version of YARP. The code was not include what you use (IWYU) and it was working just by chance thanks to a transitive include in some YARP header, that was removed in some recent version of YARP. To fix the problem, you just need to add the correct include, if you need help ask in the lab to anyone with C++ experience.

traversaro, 2024-01-03:
After you fixed the problem locally, please open a PR, thanks!

xela-95, 2024-01-04:
I'm trying the gazebo simple model tutorial (https://github.com/vvv-school/tutorial_gazebo-simple-model), but Gazebo (I'm using Gazebo classic 11) does not find the model installed, neither from the Gazebo model pane nor launching gazebo with the `gazebo simple_object_world.sdf` command.

I think the issue is due to the fact that the model directory is installed in `<conda-env-path>/share/gazebo/models` while the `GAZEBO_MODEL_PATH` environment variable has value `<conda-env-path>/share/gazebo-11/models`.

@traversaro do you think this is a consequence of some missing configuration in my environment?

xela-95, 2024-01-04:
> I'm trying the gazebo simple model tutorial (https://github.com/vvv-school/tutorial_gazebo-simple-model), but Gazebo (I'm using Gazebo classic 11) does not find the model installed, neither from the [...]

I tried to add in my `.bashrc` file the missing path to the `GAZEBO_MODEL_PATH` env variable:
```bash
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:<conda-env-path>/share/gazebo/models
```

Then, launching Gazebo I'm able to see the model in the insert pane:
[image]

xela-95, 2024-01-04:
> I guess it is a bug exposes by a a new version of YARP. The code was not include what you use (IWYU) and it was working just by chance thanks to a transitive include in some YARP header, that was [...]

Regarding this tutorial, the problem was a missing include in `Module.cpp`:
```c++
#include <yarp/os/LogStream.h>
```

Using this, I was able to compile and install. 

I then started `yarpserver` and `yarpmanager` and from it I've launched the following applications:
**Tutorial_on_Impedance_Control_system**

[image]

**Tutorial_on_Impedance_Control**
[image]

The latter is failing, i think because icub is not appearing in Gazebo, probably an issue related to https://github.com/robotology/gz-sim-yarp-plugins/issues/45#issuecomment-1877415992.

xela-95, 2024-01-05:
> After you fixed the problem locally, please open a PR, thanks!

PR merged here https://github.com/vvv-school/tutorial_impedance-control/pull/7

xela-95, 2024-01-05:
> Why some code in the iCub infrastructure is encapsulated using YARP devices that communicate over C++ interface rather than RFModule that communicate using YARP ports?

From Ch. 4 of https://www.frontiersin.org/articles/10.3389/frobt.2016.00024/full:

> For such devices YARP defines a set of C++ interfaces that provide an abstraction layer that is independent of the specificities of the hardware components that implement them. Communication with the hardware is achieved by instantiating user-level devices,  which directly send messages to the hardware. These devices implement a set of interfaces that allow reading sensor values and controlling the motors at the joint level.

Hence, it is possible to decouple the main functionality for which a device is required from the specific implementation, allowing to replace and upgrade parts of the robot without changing the software. It is also possible to create fake devices, that simulates the behavior of the hardware ones enabling to test the software infrastructure in simulation instead of using the real robot.

traversaro, 2024-01-05:
> @traversaro do you think this is a consequence of some missing configuration in my environment?

Not really, more a difference between the robotology-superbuild-based installation and the conda-based one that you are testing. The culprit is indeed the values of the `GAZEBO_MODEL_PATH` env variable. When the robotology-superbuild is used, `<install_prefix>/share/gazebo/models` is added in `GAZEBO_MODEL_PATH` by the `setup.sh` script (see https://github.com/robotology/robotology-superbuild/blob/master/doc/environment-variables-configuration.md#gazebo and https://github.com/robotology/robotology-superbuild/blob/74d42b0065f64659575856d1dc71ce8fc4fdecf8/cmake/template/setup.sh.in#L29 . To be completely honest, I do not even remember why we were adding those directories to `GAZEBO_MODEL_PATH`, I guess to support long deprecated repos such as https://github.com/robotology/icub-gazebo-wholebody or https://github.com/robotology-legacy/icub-gazebo-legacy . So, as there is no conda package for this deprecate repos, nobody adds  `<install_prefix>/share/gazebo/models` to `GAZEBO_MODEL_PATH`, and so the tutorial does not work out of the box. I opened an issue on the robotology-superbuild to track this: , even if I do not have a clear solution.

xela-95, 2024-01-09:
You can probably also follow the tutorial linked in https://github.com/robotology/gz-sim-yarp-plugins/issues/1 before closing this issue.

xela-95, 2024-01-10:
CC @carloscp3009

xela-95, 2024-01-10:
> You can probably also follow the tutorial linked in #1 (i.e. https://github.com/robotology/gazebo-yarp-plugins/blob/35fa088fa993fec83f0a14eabc4a502d3253ff7f/tutorial/model/forcetorque/README.md) before closing this issue.

@traversaro I'm not succeeding in making gazebo find the `forcetorque_world.world` when launching gazebo as:

```bash
gazebo --verbose forcetorque_world.world
```

gazebo responds with:

```
[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.240.2.37
[Wrn] [SystemPaths.cc:460] File or path does not exist [""] [forcetorque_world.world]
[Err] [Server.cc:493] Could not open file[forcetorque_world.world]
[Wrn] [Server.cc:383] Falling back on worlds/empty.world
[Msg] Loading world file [/home/acroci/mambaforge/envs/gyp_test/share/gazebo-11/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.240.2.37
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/acroci/repos/gazebo-yarp-plugins/tutorial/model/model_pose_publisher"
```

I've added the tutorial/model folder to the `GAZEBO_MODEL_PATH` using:
```
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/home/acroci/repos/gazebo-yarp-plugins/tutorial/model
```

In the gazebo insert pane I'm able to see the models coming with the plugin, but the world is not spawned.

[image]

I've tried also to add the same tutorial/model folder to the GAZEBO_RESOURCE_PATH that is usually used to search for worlds but nothing changed.

traversaro, 2024-01-10:
I think the README assumes that one does change the directory in `tutorial/model/forcetorque` before running `gazebo --verbose forcetorque_world.world`, but indeed we should write this in the tutorial.

xela-95, 2024-01-10:
> I think the README assumes that one does change the directory in `tutorial/model/forcetorque` before running `gazebo --verbose forcetorque_world.world`, but indeed we should write this in the tutorial.

Ah ok! In the meantime adding to the GAZEBO_RESOURCE_PATH the path up to `forcetorque` folder made it work.

xela-95, 2024-01-19:
I think I acquired the base knowledge of Yarp devices and gazebo yarp plugins (and I know where to get further information when needed). Closing.

closed 2024-01-19 by xela-95

xela-95, 2024-01-22:
CC @LoreMoretti

xela-95, 2024-01-25:
CC @Gio-DS

xela-95, 2024-02-02:
CC @rob-mau

traversaro, 2024-02-05:
cc @giotherobot