robotology/gz-sim-yarp-plugins #46: Refresh CMake and C++ knowledge (xela-95 version)

traversaro, 2023-12-20:
Material:
* https://github.com/ami-iit/ami-commons/wiki/Get-familiar-with-CMake
* https://github.com/ami-iit/ami-commons/wiki/Get-familiar-with-Cxx

xela-95, 2023-12-20:
@traversaro can assign this issue to me?

traversaro, 2023-12-20:
Done!

traversaro, 2023-12-21:
(You should now have permission to assign issues yourself)

xela-95, 2023-12-21:
> (You should now have permission to assign issues yourself)

I confirm, thanks!

closed 2024-01-19 by xela-95

traversaro, 2024-02-01:
> I gained the base knowledge to needed to start working with gazebo and yarp.
> 
> Useful resources I used:
> 
>     * **[Accelerated C++: Practical Programming by Example](https://www.amazon.com/Accelerated-C-Practical-Programming-Example/dp/020170353X)** by Andrew Koenig and Barbara Moo (old book but one of the best I've read about programming and with a really steep learning curve) - available in lab;
> 
>     * **[C++ Crash Course: A Fast-Paced Introduction](https://amzn.eu/d/9Gkot12)** (I've just started reading this but this is referenced as one of the best books about Modern C++(17), a new edition is coming in 2025 to integrate also C++20);
> 
>     * **[Modern C++ (Lecture & Tutorials, 2020, Vizzo & Stachniss)](https://youtube.com/playlist?list=PLgnQpQtFTOGRM59sr3nSL8BmeMZR9GCIA&si=VomlEz8J1gjEtvnI)** suggested by @LoreMoretti (great video lectures from Bonn University)
> 
>     * **[CMake tutorial](https://cmake.org/cmake/help/latest/guide/tutorial/index.html)** (official tutorial about CMake)
> 
> 
> I think this issue can be closed.

Probably we can add these to https://github.com/ami-iit/ami-commons/wiki/Get-familiar-with-Git .

xela-95, 2024-02-01:
> Probably we can add these to https://github.com/ami-iit/ami-commons/wiki/Get-familiar-with-Cxx .

Sure! I will open a PR

traversaro, 2024-02-01:
> > Probably we can add these to https://github.com/ami-iit/ami-commons/wiki/Get-familiar-with-Cxx .
> 
> Sure! I will open a PR

(Unfortunatly) that is a wiki, so you need to directly edit it. Go ahead!