robotology/gz-sim-yarp-plugins #55: Fix README build instructions

xela-95, 2023-12-22:
I was following the README instructions for the conda installation on Ubuntu, but I think there is a mistake since a cd into the cloned directory is missing:
https://github.com/robotology/gz-sim-yarp-plugins/blob/3860be537cb50e171bf63d469fb5e28e3bff9499/README.md?plain=1#L22-L29

The same also for the Windows procedure:
https://github.com/robotology/gz-sim-yarp-plugins/blob/3860be537cb50e171bf63d469fb5e28e3bff9499/README.md?plain=1#L32-L39

CC @traversaro

xela-95, 2023-12-22:
I tried to put the build directory inside the cloned `gz-sim-yarp-plugins` folder and continue the installation process. But when executing the `ninja' command I get the following errors:

<details> <summary>Logs</summary>

~~~
(gz) $ ninja
[10/23] Linking CXX executable bin/ImuTest
FAILED: bin/ImuTest 
: && /usr/bin/c++   tests/imu/CMakeFiles/ImuTest.dir/ImuTest.cc.o -o bin/ImuTest  -Wl,-rpath,/home/acroci/mambaforge/envs/gz/lib  lib/libgtest_main.a  /home/acroci/mambaforge/envs/gz/lib/libgz-sim7.so.7.6.0  /home/acroci/mambaforge/envs/gz/lib/libYARP_robotinterface.so.3.9.0  lib/libgtest.a  /home/acroci/mambaforge/envs/gz/lib/libgz-fuel_tools8.so.8.1.0  /home/acroci/mambaforge/envs/gz/lib/libgz-gui7.so.7.2.1  /home/acroci/mambaforge/envs/gz/lib/libgz-plugin2-loader.so.2.0.2  /home/acroci/mambaforge/envs/gz/lib/libQt5QuickControls2.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5Quick.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5QmlModels.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5Qml.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5Network.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5Widgets.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5Gui.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libQt5Core.so.5.15.8  /home/acroci/mambaforge/envs/gz/lib/libtinyxml2.so  /home/acroci/mambaforge/envs/gz/lib/libtinyxml2.so  /home/acroci/mambaforge/envs/gz/lib/libgz-physics6.so.6.5.1  /home/acroci/mambaforge/envs/gz/lib/libgz-plugin2.so.2.0.2  /home/acroci/mambaforge/envs/gz/lib/libgz-rendering7.so.7.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-profiler.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-events.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-av.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libswscale.so  /home/acroci/mambaforge/envs/gz/lib/libswscale.so  /home/acroci/mambaforge/envs/gz/lib/libavdevice.so  /home/acroci/mambaforge/envs/gz/lib/libavdevice.so  /home/acroci/mambaforge/envs/gz/lib/libavformat.so  /home/acroci/mambaforge/envs/gz/lib/libavformat.so  /home/acroci/mambaforge/envs/gz/lib/libavcodec.so  /home/acroci/mambaforge/envs/gz/lib/libavcodec.so  /home/acroci/mambaforge/envs/gz/lib/libavutil.so  /home/acroci/mambaforge/envs/gz/lib/libavutil.so  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-io.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-testing.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-geospatial.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5-graphics.so.5.4.2  /home/acroci/mambaforge/envs/gz/lib/libgz-common5.so.5.4.2  -lpthread  -lrt  /home/acroci/mambaforge/envs/gz/lib/libgz-transport12-parameters.so.12.2.1  /home/acroci/mambaforge/envs/gz/lib/libgz-transport12.so.12.2.1  /home/acroci/mambaforge/envs/gz/lib/libuuid.so  /home/acroci/mambaforge/envs/gz/lib/libuuid.so  /home/acroci/mambaforge/envs/gz/lib/libgz-msgs9.so.9.5.0  /home/acroci/mambaforge/envs/gz/lib/libsdformat13.so.13.6.0  /home/acroci/mambaforge/envs/gz/lib/libgz-math7.so.7.3.0  /home/acroci/mambaforge/envs/gz/lib/libgz-utils2.so.2.1.0  /home/acroci/mambaforge/envs/gz/lib/libprotobuf.so.24.4.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_check_op.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_leak_check.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_die_if_null.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_conditions.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_message.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_nullguard.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_examine_stack.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_format.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_proto.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_log_sink_set.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_sink.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_entry.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_internal.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_marshalling.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_reflection.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_config.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_program_name.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_private_handle_accessor.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_commandlineflag.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_flags_commandlineflag_internal.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_initialize.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_globals.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_log_internal_globals.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_raw_hash_set.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_hash.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_city.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_low_level_hash.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_hashtablez_sampler.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_statusor.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_status.so.2308.0.0  /home/acroci/mambaforge/envs/gz/lib/libabsl_cord.so.2308.0.0  /home/acroci/mambaf [... cut ...]
/home/acroci/mambaforge/envs/gz/bin/ld: /home/acroci/mambaforge/envs/gz/bin/../x86_64-conda-linux-gnu/sysroot/lib64/libpthread.so.0: undefined reference to `__libc_current_sigrtmax_private@GLIBC_PRIVATE'
/home/acroci/mambaforge/envs/gz/bin/ld: /home/acroci/mambaforge/envs/gz/bin/../x86_64-conda-linux-gnu/sysroot/lib64/libdl.so.2: undefined reference to `_dl_vsym@GLIBC_PRIVATE'
collect2: error: ld returned 1 exit status
[18/23] Building CXX object plugins/laser/CMakeFiles/gz-sim-yarp-laser-system.dir/Laser.cc.o
ninja: build stopped: subcommand failed.
~~~

</details>

traversaro, 2023-12-22:
> I was following the README instructions for the conda installation on Ubuntu, but I think there is a mistake since a cd into the cloned directory is missing: [...]

Good catch, feel free to open a PR.

traversaro, 2023-12-22:
> I tried to put the build directory inside the cloned `gz-sim-yarp-plugins` folder and continue the installation process. But when executing the `ninja' command I get the following errors:

Interesting, I wonder why it does not happen in CI, this is a good example why using pixi could make sense. Anyhow, to be more pragmatic, can you paste the output of `mamba list` in your environment, so we can compare it with the `mamba list` in the CI?

traversaro, 2023-12-22:
> so we can compare it with the `mamba list` in the CI?

This was missing, I added it in https://github.com/robotology/gz-sim-yarp-plugins/pull/56 .

xela-95, 2023-12-22:
> Interesting, I wonder why it does not happen in CI, this is a good example why using pixi could make sense. Anyhow, to be more pragmatic, can you paste the output of `mamba list` in your environment, so we can compare it with the `mamba list` in the CI?

Here it is:

```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.1.2                h59595ed_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
aom                       3.7.1                h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
azure-core-cpp            1.10.3               h91d86a7_0    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_0    conda-forge
azure-storage-common-cpp  12.5.0               hb858b4b_2    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.24.0               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2023.11.17           hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
cfitsio                   4.3.1                hbdc6101_0    conda-forge
cli11                     2.3.2                hcb278e6_0    conda-forge
cmake                     3.28.1               hcfe8598_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
dartsim                   6.13.0               h7abe67e_5    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
elfutils                  0.189                h6f2b95c_1    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
fcl                       0.7.0                hadc09e8_4    conda-forge
ffmpeg                    6.1.0           gpl_h186bccc_105    conda-forge
flann                     1.9.2                h2b5ea80_0    conda-forge
fmt                       10.1.1               h00ab1b0_1    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.7.0                heb67821_0    conda-forge
freeglut                  3.2.2                hac7e632_2    conda-forge
freeimage                 3.18.0              h7ff9cae_18    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
freexl                    2.0.0                h743c826_0    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gcc                       12.3.0               h8d2909c_2    conda-forge
gcc_impl_linux-64         12.3.0               he2b93b0_3    conda-forge
gcc_linux-64              12.3.0               h76fc315_2    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
geos                      3.12.1               h59595ed_0    conda-forge
geotiff                   1.7.1               h6b2125f_15    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
gfortran                  12.3.0               h499e0f7_2    conda-forge
gfortran_impl_linux-64    12.3.0               hfcedea8_3    conda-forge
gfortran_linux-64         12.3.0               h7fe76b4_2    conda-forge
giflib                    5.2.1                h0b41bf4_3    conda-forge
glib                      2.78.3               hfc55251_0    conda-forge
glib-tools                2.78.3               hfc55251_0    conda-forge
gmp                       6.3.0                h59595ed_0    conda-forge
[... 215 lines cut ...]
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
[... cut ...]
yarp                      3.9.0                ha770c72_0    conda-forge
yarp-python               3.9.0           py312hebe80a8_0    conda-forge
ycm-cmake-modules         0.16.2               h59595ed_0    conda-forge
zeromq                    4.3.5                h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

```

xela-95, 2024-01-09:
@xela-95 I will try on a fresh environment and I will report the outcome.

xela-95, 2024-01-11:
I've created a fresh environment and followed the README instructions for the conda installation. I've been able to build and install without errors.

For completeness, here is reported my 'conda list' output:

<details><summary>Expand</summary>

```
# packages in environment at /home/acroci/mambaforge/envs/gz7:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.1.2                h59595ed_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
aom                       3.7.1                h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
azure-core-cpp            1.10.3               h91d86a7_0    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_0    conda-forge
azure-storage-common-cpp  12.5.0               hb858b4b_2    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.25.0               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2023.11.17           hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
cfitsio                   4.3.1                hbdc6101_0    conda-forge
cli11                     2.3.2                hcb278e6_0    conda-forge
cmake                     3.28.1               hcfe8598_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
dartsim                   6.13.1               hdbb2bd4_2    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
elfutils                  0.189                h6f2b95c_1    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
fcl                       0.7.0                hadc09e8_4    conda-forge
ffmpeg                    6.1.1           gpl_h186bccc_100    conda-forge
flann                     1.9.2                h2b5ea80_0    conda-forge
fmt                       10.2.1               h00ab1b0_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.7.0                heb67821_0    conda-forge
freeglut                  3.2.2                hac7e632_2    conda-forge
freeimage                 3.18.0              h7ff9cae_18    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
freexl                    2.0.0                h743c826_0    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gcc                       12.3.0               h8d2909c_2    conda-forge
gcc_impl_linux-64         12.3.0               he2b93b0_3    conda-forge
gcc_linux-64              12.3.0               h76fc315_2    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
geos                      3.12.1               h59595ed_0    conda-forge
geotiff                   1.7.1               h6b2125f_15    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
gfortran                  12.3.0               h499e0f7_2    conda-forge
gfortran_impl_linux-64    12.3.0               hfcedea8_3    conda-forge
gfortran_linux-64         12.3.0               h7fe76b4_2    conda-forge
[... 222 lines cut ...]
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
[... cut ...]
yarp                      3.9.0                ha770c72_0    conda-forge
yarp-python               3.9.0           py312hebe80a8_0    conda-forge
ycm-cmake-modules         0.16.2               h59595ed_0    conda-forge
zeromq                    4.3.5                h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

```

</details>

traversaro, 2024-01-11:
Great! Feel free to close the issue then.

closed 2024-01-11 by xela-95