robotology/gz-sim-yarp-plugins #57: Run tutorial about gz-sim-yarp-plugins

xela-95, 2024-01-09:
https://github.com/robotology/gz-sim-yarp-plugins/tree/main/tutorial

xela-95, 2024-01-11:
I'm trying to run the **Camera model tutorial**, following https://github.com/robotology/gz-sim-yarp-plugins/tree/main/tutorial/camera/model.

When I launch the command `gz sim model.sdf ` gazebo sim open but the terminal writes the following errors:

```
[Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gzyarp::Camera]
- Requested library name: [gz-sim-yarp-camera-system]
- Library search paths:
  - /home/acroci/.gz/sim/plugins/
  - /home/acroci/mambaforge/envs/gz7/lib/gz-sim-7/plugins/
  - /home/acroci/.ignition/gazebo/plugins/
[Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gzyarp::RobotInterface]
- Requested library name: [gz-sim-yarp-robotinterface-system]
- Library search paths:
  - /home/acroci/.gz/sim/plugins/
  - /home/acroci/mambaforge/envs/gz7/lib/gz-sim-7/plugins/
  - /home/acroci/.ignition/gazebo/plugins/
```

It is similar to the errors I got in gazebo classic tutorials when the environment variables could not find the models and worlds of the tutorials. 

@traversaro do you know if gazebo sim works through the same environment variables used in gazebo classic? I could not find helpful documentation on this topic.

traversaro, 2024-01-11:
Did you set the GZ_SIM_SYSTEM_PLUGIN_PATH env variable as described in the readme of the repo?

xela-95, 2024-01-11:
> Did you set the GZ_SIM_SYSTEM_PLUGIN_PATH env variable as described in the readme of the repo?

Thank you that was the problem! I forgot this step when I created the new environment. I put the `export` command in my env activation scripts.

xela-95, 2024-01-11:
Here two things that happened when launching `gz sim model.sdf`:

**Segmentation fault**

```
gz sim model.sdf 
[INFO] gz-sim-yarp-camera-system: configuration of sensor  camera_sensor  loaded from yarpConfigurationString :  (yarpDeviceName camera_plugin_device) (parentLinkName link_1) (sensorName camera_sensor) (display_timestamp) 
[DEBUG] |yarp.dev.PolyDriver|gazebo_camera| Parameters are (device gazebo_camera) (display_timestamp) (parentLinkName link_1) (sensorName camera_sensor) (sensorScopedName "model/camera_model/link/link_1/sensor/camera_sensor") (sensor_name camera_sensor) (yarpDeviceName camera_plugin_device)
[INFO] |yarp.dev.PolyDriver|gazebo_camera| Created device <gazebo_camera>. See C++ class CameraDriver for documentation.
[INFO] gz-sim-yarp-camera-system: Registered YARP device with instance name: model/camera_model/link/link_1/sensor/camera_sensor/camera_plugin_device
[DEBUG] Reading file camera_nws.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file camera_nws.xml
[INFO] Yarprobotinterface was started using the following enable_tags: 
[INFO] Yarprobotinterface was started using the following disable_tags: 
[DEBUG] List of all enable attributes found in the include tags: 
[DEBUG] List of all disable attributes found in the include tags: 
[DEBUG] Preprocessor complete in:  7.15256e-07 s
[INFO] startup phase starting...
[INFO] Opening device camera_nws_yarp with parameters [("robotName" = "camera"), ("name" = "/camera"), ("period" = "0.033333"), ("capabilities" = "COLOR")]
[DEBUG] |yarp.dev.PolyDriver|camera_nws_yarp| Parameters are (capabilities COLOR) (device frameGrabber_nws_yarp) (id camera_nws_yarp) (name "/camera") (period 0.033333000000000001517) (robotName camera)
[INFO] |yarp.os.Port|/camera/rpc| Port /camera/rpc active at tcp://10.240.2.37:10002/
[INFO] |yarp.os.Port|/camera| Port /camera active at tcp://10.240.2.37:10003/
[INFO] |yarp.device.frameGrabber_nws_yarp| Running, waiting for attach...
[INFO] |yarp.dev.PolyDriver|camera_nws_yarp| Created wrapper <frameGrabber_nws_yarp>. See C++ class FrameGrabber_nws_yarp for documentation.
[INFO] Entering action level 5 of phase startup
[INFO] Executing attach action, level 5 on device camera_nws_yarp with parameters [("device" = "camera_plugin_device")]
[WARNING] |yarp.device.frameGrabber_nws_yarp| Targets has not IVisualParamInterface, some features cannot be available
[INFO] All actions for action level 5 of startup phase started. Waiting for unfinished actions.
[INFO] All actions for action level 5 of startup phase finished.
[INFO] startup phase finished.
Stack trace (most recent call last) in thread 22541:
#6    Object "", at 0xffffffffffffffff, in 
#5    Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f07cc92684f]
#4    Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f07cc894ac2]
#3    Source "../../../../../libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x7f07c7dd9e94]
#2    Object "/home/acroci/mambaforge/envs/gz7/lib/libgz-sim7.so.7.6.0", at 0x7f07c6dcc592, in gz::sim::v7::SimulationRunner::ProcessSystemQueue()::{lambda()#1}::operator()() const
#1    Object "/home/acroci/mambaforge/envs/gz7/lib/libgz-sim-yarp-camera-system.so", at 0x7f0797d4ee20, in non-virtual thunk to gzyarp::Camera::PostUpdate(gz::sim::v7::UpdateInfo const&, gz::sim::v7::EntityComponentManager const&)
#0    Source "../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S", line 801, in __memmove_sse2_unaligned_erms [0x7f07cc8c4c4d]
Segmentation fault (Address not mapped to object [0x7f07c62c25e0])
```

Closing and reopening the terminal tab seems to resolve the issue.

**Strange jitter in Gazebo sim GUI**

I noticed a strange behavior in gazebo sim GUI in which the play/pause button of the simulation seemed to change state in milliseconds and does not allow me to start the simulation properly. Maybe an issue with the gpu or some drivers?

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/b6919412-b6c2-4581-9212-0a34c8e82bd3

xela-95, 2024-01-11:
> **Strange jitter in Gazebo sim GUI**
> 
> I noticed a strange behavior in gazebo sim GUI in which the play/pause button of the simulation seemed to change state in milliseconds and does not allow me to start the simulation properly. Maybe an issue with the gpu or some drivers?

Rebooting my PC this is not happening anymore.

xela-95, 2024-01-12:
https://github.com/robotology/gz-sim-yarp-plugins/blob/main/tutorial/forcetorque/model_two_sensors/README.md?plain=1#L25-L41

Maybe it's more clear to name the two `/read` ports differently in order to not have clashing names if the user wants to read the two torque sensors simultaneously.

traversaro, 2024-01-12:
> https://github.com/robotology/gz-sim-yarp-plugins/blob/main/tutorial/forcetorque/model_two_sensors/README.md?plain=1#L25-L41
> 
> Maybe it's more clear to name the two `/read` ports differently in order to not have clashing names if the user wants to read the two torque sensors simultaneously.

Good idea, feel free to open a PR!

xela-95, 2024-01-12:
I've launched all the tutorials. They all worked as expected. 

Now I'm looking through the code of the plugins, for specific topics I will open new dedicated issues.

xela-95, 2024-01-12:
Turials have been ran successfully. The issues reported in https://github.com/robotology/gz-sim-yarp-plugins/issues/57#issuecomment-1886923601 have not reproduced. Closing.

closed 2024-01-12 by xela-95