robotology/gz-sim-yarp-plugins #58: Adapt gz-sim-yarp-plugins to work also (or just with, if it simpler) gz-sim8

xela-95, 2024-01-09:


xela-95, 2024-03-01:
I created a fresh new mamba environment with the following command:

```
mamba create -c conda-forge -n gz8 libgz-sim8 yarp cmake ninja pkg-config cmake compilers
```

When I activate this environment I get the following warning (at least I think):

```
WARNING: No ICDs were found. Either,
- Install a conda package providing a OpenCL implementation (pocl, oclgrind, intel-compute-runtime, beignet) or 
- Make your system-wide implementation visible by installing ocl-icd-system conda package. 
```

@traversaro are you familiar with this?

xela-95, 2024-03-01:
Unrelated to the previous message:

For my development setup I use additional packages, my mamba environment is in fact created as:

```
mamba create -c conda-forge -n gz8 libgz-sim8 yarp cmake ninja pkg-config cmake compilers clang clang-tools clang-format include-what-you-use 
```

Should I update the README with all those packages, or create separate environments.yaml for development and usage of the package?

traversaro, 2024-03-04:
As gz-sim7 is part of Gazebo Garden (see https://gazebosim.org/docs/garden/install) and Gazebo Garden is a short-term support release of Gazebo (support will stop in November 2024, see https://gazebosim.org/docs/harmonic/releases), I think it is ok to just support gz-sim8 . In the future we may want to to support multiple version of gz-sim at the same time, but I see little value in trying to remain compatible with gz-sim7 at this stage.

traversaro, 2024-03-04:
> Should I update the README with all those packages, or create separate environments.yaml for development and usage of the package?

Yes, I would separate the dependencies strictly required for the package usage, and the development dependencies. Anyhow, eventually we will add gz-sim-yarp-plugins will be added to conda-forge and to the robotology-superbuild, so the two main way of usage document in the README will be those one (see for example the idyntree README: https://github.com/robotology/idyntree?tab=readme-ov-file#installation) so eventually this will not be a big problem.

traversaro, 2024-03-04:
> hen I activate this environment I get the following warning:
> 
> ```
> WARNING: No ICDs were found. Either,
> - Install a conda package providing a OpenCL implementation (pocl, oclgrind, intel-compute-runtime, beignet) or 
> - Make your system-wide implementation visible by installing ocl-icd-system conda package. 
> ```
> 
> @traversaro are you familiar with this?

Yes, see https://github.com/conda-forge/ocl-icd-feedstock/issues/29 .

PR #105 (2024-03-07): Adapt gz-sim-yarp-plugins to work with gz-sim8
Closes #58
closed 2024-03-07 by xela-95