robotology/gz-sim-yarp-plugins #59: Port basestate plugin functionality from gazebo-yarp-plugins

xela-95, 2024-01-09:
Plugin: https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/basestate
Docs: https://github.com/robotology/gazebo-yarp-plugins/blob/c89280295d53279049bb7521cf6b6b3400130f23/plugins/basestate/include/yarp/dev/BaseStateDriver.h#L44
Examples (not accessible to everyone): https://github.com/ami-iit/component_ironcub/blob/a7117b65ccc4adc43f5991098d89e2266dff4870/models/iRonCub-Mk3/iRonCub/robots/iRonCub-Mk3_Gazebo/model_with-legs-fts.urdf#L1524

xela-95, 2024-01-12:
As this plugin share the same structure of the forcetorque one, probably we can copy the forcetorque directory in basestate, and then modify the plugin to get the same functionality of https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/basestate by copying the code in there.

xela-95, 2024-01-17:
The `basestate` plugin for gazebo classic have been updated in PR https://github.com/robotology/gazebo-yarp-plugins/pull/675

xela-95, 2024-01-17:
@traversaro I am progressing with the porting of the plugin, but up to now I have encountered an issue: my code builds (both manually configuring and building from terminal and using the vscode cmake tools extension). But when I try to install I get the following error:

```
(gz7) acroci@IITICUBLAP218:~/repos/gz-sim-yarp-plugins/build$ ninja install
[0/1] Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/lib/libgz-sim-yarp-handler.so
CMake Error at libraries/singleton-devices/cmake_install.cmake:52 (file):
  file INSTALL cannot copy file
  "/home/acroci/repos/gz-sim-yarp-plugins/build/lib/libgz-sim-yarp-handler.so"
  to "/usr/local/lib/libgz-sim-yarp-handler.so": Permission denied.
Call Stack (most recent call first):
  libraries/cmake_install.cmake:47 (include)
  cmake_install.cmake:47 (include)

FAILED: CMakeFiles/install.util 
cd /home/acroci/repos/gz-sim-yarp-plugins/build && /home/acroci/mambaforge/envs/gz7/bin/cmake -P cmake_install.cmake
ninja: build stopped: subcommand failed.
```

Probably I'm missing something stupid but I cannot figure out what it is 🙃

xela-95, 2024-01-17:
Another question: in order to ask code reviews about the work I'm doing, should I open a PR in draft mode (even if the code is not completed)?

xela-95, 2024-01-17:
> @traversaro I am progressing with the porting of the plugin, but up to now I have encountered an issue: my code builds (both manually configuring and building from terminal and using the vscode [...]

Ok the problem is solved by configuring the project (as written in the README) with:

```
cmake -GNinja -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_PREFIX_PATH=$CONDA_PREFIX ..
```

Then running:

```
ninja
ninja install
```

runs fine.

I'm a bit new with cmake, so as I can understand the problem was due to the fact that by default ninja tries to install the executables into the `/usr/local/lib` folder (for which it needs administrative privileges), while using the `$CONDA_PREFIX` location it is able to install.

traversaro, 2024-01-17:
> Another question: in order to ask code reviews about the work I'm doing, should I open a PR in draft mode (even if the code is not completed)?

Yes!

xela-95, 2024-01-18:
Some documentation to understand messaging in Gazebo sim: 
- https://gazebosim.org/docs/harmonic/architecture
- https://gazebosim.org/api/transport/13/tutorials.html

PR #65 (2024-01-26): Add gz-sim-yarp-basestate-system plugin to publish BaseState information
This PR adds the `gz-sim-yarp-basestate` plugin that, as the [correspondent plugin](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/basestate) for gazebo classic, serves the purpose of publishing state information (base pose, velocity and acceleration) of the model base link.

Closes #59
Closes #66
closed 2024-01-26 by xela-95