robotology/gz-sim-yarp-plugins #60: Port basic clock plugin functionality from gazebo-yarp-plugins

xela-95, 2024-01-09:
Clock plugin docs is https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/clock/include/gazebo/Clock.hh#L35 .

The clock functionality are two:
 * - It publishes the simulation time on a Yarp port `/clock`
 * - It opens a Yarp RPC Port (`/clock/rpc`) to allows control of the simulation time.

This issue focus on the first functionality, i.e. publication of simulation time on yarp port /clock.

xela-95, 2024-01-09:
Useful resource: https://github.com/robotology/gz-sim-yarp-plugins/issues/4 .

xela-95, 2024-01-24:
I'm gathering documentation related to this feature to understand how to perform the porting:

- Gazebo sim Classes:
  - [Clock ](https://gazebosim.org/api/transport/13/classgz_1_1transport_1_1Clock.html#pub-methods)
  - [WallClock](https://gazebosim.org/api/transport/13/classgz_1_1transport_1_1WallClock.html)
  - [NetworkClock](https://gazebosim.org/api/transport/13/classgz_1_1transport_1_1NetworkClock.html#details)
- [Gazebo physics documentation](https://gazebosim.org/api/physics/7/tutorials.html)
- Old Issue from @diegoferigo, potentially useful: https://github.com/gazebosim/gz-sim/issues/51

xela-95, 2024-01-24:
I've seen that the clock plugin makes use of the thrift definition for the Clock service: https://github.com/robotology/gazebo-yarp-plugins/blob/cb3521f7543a5a07e2a65a82dedd2eb2ec2cea12/thrift/clock/clock_rpc.thrift

This file gets compiled and produces a `ClockServer` class, that in turn is inherited by [ClockServerImpl](https://github.com/robotology/gazebo-yarp-plugins/blob/cb3521f7543a5a07e2a65a82dedd2eb2ec2cea12/plugins/clock/include/GazeboYarpPlugins/ClockServerImpl.h) class. This class is used as a data member of class [GazeboYarpClock](https://github.com/robotology/gazebo-yarp-plugins/blob/cb3521f7543a5a07e2a65a82dedd2eb2ec2cea12/plugins/clock/include/gazebo/Clock.hh) that implements the `gazebo::SystemPlugin` base class.

I have to understand how this logic maps to the new Gazebo sim paradigm and whether it is needed to use the thrift definitions for the clock service.

xela-95, 2024-01-25:
CC @traversaro

traversaro, 2024-01-25:
> I have to understand how this logic maps to the new Gazebo sim paradigm and whether it is needed to use the thrift definitions for the clock service.

If you want to implement the /clock/rpc part of the plugin, yes. However, as a first iteration you can also just concentrate on publishing the `/clock` port.

xela-95, 2024-01-25:
I leave here some refs to useful classes and tutorials for controlling the simulation that @diegoferigo showed me:
- [WorldControl class](https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/WorldControl.hh)
- [SimulationRunner class](https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/SimulationRunner.hh)
- [Reset simulation tutorial](https://github.com/gazebosim/gz-sim/blob/gz-sim8/tutorials/reset_simulation.md)

I think this will be useful in future to control the clock rpc part.

xela-95, 2024-01-29:
@traversaro, up to now the clock plugin sends on the `/clock` port the current simulation time at each simulation step, do you think it's useful to add a check to **avoid** writing a message if the current timestamp is the same of the previous sent? 

This could be useful to avoid sending thousands of messages of the same timestamp when gazebo server is launched or when in general the simulation is paused, but maybe there are cases that I don't have in mind for which it is better to always send the timestamp.

traversaro, 2024-01-29:
> @traversaro, up to now the clock plugin sends on the `/clock` port the current simulation time at each simulation step, do you think it's useful to add a check to **avoid** writing a message if the current timestamp is the same of the previous sent?

Yes, I think it make sense.

PR #70 (2024-01-29): Add clock plugin with basic functionality
This PR adds to the gz-sim-yarp-plugins a clock plugin, with the only basic functionality of writing the current simulation time on the yarp `/clock` port.

Closes #60 , #73
closed 2024-01-29 by xela-95