robotology/gz-sim-yarp-plugins #75: CameraTest flaky tests on apt CI

traversaro, 2024-01-29:
See for example https://github.com/robotology/gz-sim-yarp-plugins/actions/runs/7698197313/job/20976938788 .

xela-95, 2024-01-29:
Closed in #76

traversaro, 2024-01-29:
Actually I wanted to have an issue to track the underling issue of the CameraTest segfault in a non-deterministic way, but indeed probably an issue with a more appropriate name may be more indicated.

traversaro, 2024-01-30:
Thanks, I clarified the test.

Nicogene, 2025-03-10:
Hi!

I was developing the `DepthCameraTest` and I have encountered the same segmentation fault w/ apt dependencies (see ), I tried to get some info from valgrind and/or gdb, this is what I got after compiling both YARP and gz-sim-yarp-plugins in debug:
```

Thread 49 "DepthCameraTest" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffbb57c6c0 (LWP 104443)]
0x00007fffa0fdab10 in ?? ()
(gdb) bt
#0  0x00007fffa0fdab10 in ??? ()
#1  0x00007ffff743d340 in __GI___nptl_deallocate_tsd () at ./nptl/nptl_deallocate_tsd.c:73
#2  __GI___nptl_deallocate_tsd () at ./nptl/nptl_deallocate_tsd.c:22
#3  0x00007ffff7440890 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.
```

Nicogene, 2025-03-12:
I confirm that w/ `export LIBGL_ALWAYS_SOFTWARE=1` the test passes without crashing on closure!

```
14: [----------] Global test environment tear-down
14: [==========] 1 test from 1 test suite ran. (5407 ms total)
14: [  PASSED  ] 1 test.
1/1 Test #14: DepthCameraTest ..................   Passed    5.47 sec
```

traversaro, 2025-03-12:
Ack, then the problem was that one, on the one tracked here. For reference, an upstream related issue is https://github.com/microsoft/WSL/issues/11277 . If you want to use hardware acceleration but not nvidia due to this bug, you can also select the intel GPU by setting `export MESA_D3D12_DEFAULT_ADAPTER_NAME=Intel`, see https://github.com/microsoft/wslg/wiki/GPU-selection-in-WSLg .