robotology/gz-sim-yarp-plugins #101: Decouple sensor reading frequency from simulation one

xela-95, 2024-02-28:
See https://github.com/robotology/gz-sim-yarp-plugins/issues/83#issuecomment-1967601706

traversaro, 2024-02-28:
Note that probably this is only needed for the sensors that are part of the controlboard, for the SDF sensors (http://sdformat.org/spec?elem=sensor= they should have an `update_rate` parameter that already does that, at least in theory.

xela-95, 2024-02-28:
> Note that probably this is only needed for the sensors that are part of the controlboard, for the SDF sensors (http://sdformat.org/spec?elem=sensor= they should have an `update_rate` parameter that already does that, at least in theory.

Thank you I didn't noticed this parameter, I will use this issue to perform tests and check that the correct behavior is implemented.