robotology/gz-sim-yarp-plugins #122: Diary: make ergoCub walk in Gazebo

xela-95, 2024-03-15:
This issue will track all the technical details concerning #94.

xela-95, 2024-03-15:
# Conda environment for development

```
mamba create -n walking  -c robotology -c conda-forge bipedal-locomotion-framework idyntree yarp icub-contrib-common icub-main osqp-eigen qpoases libunicycle-footstep-planner libgz-sim8 catch2 whole-body-estimators icub-models ycm-cmake-modules ninja pkg-config cmake compilers clang clang-tools clang-format include-what-you-use
```

xela-95, 2024-03-15:
Since in the walking controller tutorial I should launch gazebo classic with:
```
export YARP_CLOCK=/clock
gazebo -slibgazebo_yarp_clock.so
```

I think this is no more possible on Gazebo. Should I update the world SDF used in #103 adding the clock plugin?

CC @traversaro

xela-95, 2024-03-15:
> By doing this, the port that gets registered is `registration name /IITICUBLAP218/gz/377448/clock` instead of simply `/clock` 🤔

The issue was caused by the `YARP_CLOCK` environment variable. If I do not set it the `/clock` port is opened and can be read.

xela-95, 2024-03-15:
I'm at step 4/7 of the [walking-controllers tutorial](https://github.com/robotology/walking-controllers/blob/39fb59969e0219f50e63b39664839f7ed02559af/README.md#computer-how-to-run-the-simulation).

The command:
```
yarprobotinterface --config conf/launch_wholebodydynamics_ecub.xml  
```

referring to https://github.com/icub-tech-iit/ergocub-software/blob/335ed25cab5ff24ccdc4647b4ea318e0f5f9f305/urdf/ergoCub/conf/launch_wholebodydynamics_ecub.xml#L4.

The output logs are the following:

<details><summary>Details</summary>

```
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[DEBUG] Reading file /home/acroci/ergocub_ws/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[WARNING] Invalid syntax while loading /home/acroci/ergocub_ws/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml at line 1 . Expected document of type robot . Found devices
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /home/acroci/ergocub_ws/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[INFO] Yarprobotinterface was started using the following enable_tags: 
[INFO] Yarprobotinterface was started using the following disable_tags: 
[DEBUG] List of all enable attributes found in the include tags: 
[DEBUG] List of all disable attributes found in the include tags: 
[DEBUG] Preprocessor complete in:  0.000169277 s
[WARNING] Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
[WARNING] *************************************************************************************
* yarprobotinterface 'portprefix' parameter does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name    *
*     name:    full port prefix name with leading '/',  e.g.  /robotName      *
* A temporary automatic fix will be done for you, but please fix your config file   *
*************************************************************************************
[INFO] |yarp.os.Port|/ergoCubGazeboV1/yarprobotinterface| Port /ergoCubGazeboV1/yarprobotinterface active at tcp://10.240.2.19:10046/
[INFO] startup phase starting...
[INFO] Opening device torso_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/torso"), ("local" = "/wholeBodyDynamics/torso")]
[DEBUG] |yarp.dev.PolyDriver|torso_mc| Parameters are (device remote_controlboard) (id torso_mc) (local "/wholeBodyDynamics/torso") (remote "/ergocubSim/torso") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/rpc:o| Port /wholeBodyDynamics/torso/rpc:o active at tcp://10.240.2.19:10047/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/command:o| Port /wholeBodyDynamics/torso/command:o active at tcp://10.240.2.19:10048/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/stateExt:i| Port /wholeBodyDynamics/torso/stateExt:i active at tcp://10.240.2.19:10049/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/rpc:o| Sending output from /wholeBodyDynamics/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/command:o| Sending output from /wholeBodyDynamics/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|torso_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device left_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/left_arm"), ("local" = "/wholeBodyDynamics/left_arm")]
[DEBUG] |yarp.dev.PolyDriver|left_arm_mc| Parameters are (device remote_controlboard) (id left_arm_mc) (local "/wholeBodyDynamics/left_arm") (remote "/ergocubSim/left_arm") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/rpc:o| Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://10.240.2.19:10050/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/command:o| Port /wholeBodyDynamics/left_arm/command:o active at tcp://10.240.2.19:10051/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/stateExt:i| Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://10.240.2.19:10052/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/rpc:o| Sending output from /wholeBodyDynamics/left_arm/rpc:o to /ergocubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/command:o| Sending output from /wholeBodyDynamics/left_arm/command:o to /ergocubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/left_arm/stateExt:i| Receiving input from /ergocubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|left_arm_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device right_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/right_arm"), ("local" = "/wholeBodyDynamics/right_arm")]
[DEBUG] |yarp.dev.PolyDriver|right_arm_mc| Parameters are (device remote_controlboard) (id right_arm_mc) (local "/wholeBodyDynamics/right_arm") (remote "/ergocubSim/right_arm") (robotName ergoCubGazeboV1)
[... cut ...]
[INFO] Closing device ergocub_left_arm_ft
[INFO] Closing device waist_inertial_hardware_device
[INFO] Closing device head_inertial_hardware_device
[INFO] Closing device head_mc
[INFO] Closing device right_leg_mc
[INFO] Closing device left_leg_mc
[INFO] Closing device right_arm_mc
[INFO] Closing device left_arm_mc
[INFO] Closing device torso_mc
[INFO] shutdown phase finished.
[ERROR] Error in shutdown phase... see previous messages for more info
[INFO] |yarp.os.RFModule| RFModule failed to open.
```

</details>

xela-95, 2024-03-15:
CC @S-Dafarra

traversaro, 2024-03-15:
It seems that wholebodydynamics is not able to start, in particular this line:
~~~
[INFO] wholeBodyDynamics : impossible to create ExtWrenchesAndJointTorquesEstimator from file  model.urdf  ( full path:    ) 
~~~

suggest that it was not able to find the model. Did you set `YARP_ROBOT_NAME` ?

traversaro, 2024-03-16:
This feature may be relevant:
* https://github.com/gazebosim/gz-physics/pull/582
* https://github.com/gazebosim/gz-sim/pull/2270

xela-95, 2024-03-18:
> It seems that wholebodydynamics is not able to start, in particular this line:
> 
> ```
> [INFO] wholeBodyDynamics : impossible to create ExtWrenchesAndJointTorquesEstimator from file  model.urdf  ( full path:    ) 
> ```
> 
> suggest that it was not able to find the model. Did you set `YARP_ROBOT_NAME` ?

In Step 1, I've done `export YARP_ROBOT_NAME="ergoCubGazeboV1"`. We are using the model in the folder named `ergoCubGazeboV1_1`. But also trying to update the `YARP_ROBOT_NAME` variable to this name produces the same error

traversaro, 2024-03-18:
Ahh, probably you did not installed `ergocub-software` as you are using your local model?

xela-95, 2024-03-18:
> Ahh, probably you did not installed `ergocub-software` as you are using your local model?

Yep exactly! Do I have to install it and update some of the variables?

xela-95, 2024-03-18:
This is the output of `yarp resource find model.urdf` (having set YARP_ROBOT_NAME="ergoCubGazeboV1_1"):

```
[DEBUG] |yarp.os.ResourceFinder| configuring
[DEBUG] |yarp.os.ResourceFinder| finding file [from]
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/ergocub_ws/install/share/ergoCub/conf/from] (pwd)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/.config/yarp/robots/ergoCubGazeboV1_1] (robot YARP_CONFIG_HOME)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/.local/share/yarp/robots/ergoCubGazeboV1_1] (robot YARP_DATA_HOME)
[DEBUG] |yarp.os.ResourceFinder| checking [/etc/xdg/xdg-ubuntu/yarp/robots/ergoCubGazeboV1_1] (robot YARP_CONFIG_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/etc/xdg/yarp/robots/ergoCubGazeboV1_1] (robot YARP_CONFIG_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [robots/ergoCubGazeboV1_1] (robot YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/ICUBcontrib/robots/ergoCubGazeboV1_1] (robot YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| found /home/acroci/mambaforge/envs/walking/share/ICUBcontrib/robots/ergoCubGazeboV1_1
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/iCub/robots/ergoCubGazeboV1_1] (robot YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/yarp/robots/ergoCubGazeboV1_1] (robot YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [config/path.d] (robot path.d YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| found /home/acroci/mambaforge/envs/walking/share/yarp/config/path.d
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/BipedalLocomotionFramework/robots/ergoCubGazeboV1_1] (robot yarp.d)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/ICUBcontrib/robots/ergoCubGazeboV1_1/from] (robot)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/.config/yarp/from] (YARP_CONFIG_HOME)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/.local/share/yarp/from] (YARP_DATA_HOME)
[DEBUG] |yarp.os.ResourceFinder| checking [/etc/xdg/xdg-ubuntu/yarp/from] (YARP_CONFIG_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/etc/xdg/yarp/from] (YARP_CONFIG_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [from] (YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/ICUBcontrib/from] (YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/iCub/from] (YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/yarp/from] (YARP_DATA_DIRS)
[DEBUG] |yarp.os.ResourceFinder| checking [/home/acroci/mambaforge/envs/walking/share/BipedalLocomotionFramework/from] (yarp.d)
[DEBUG] |yarp.os.ResourceFinder| did not find from
""

```

xela-95, 2024-03-18:
> Probably you can just add the `install/share/ergoCub` folder from the ergocub_ws to `YARP_DATA_DIRS`, so the YARP facilities to find files can find the `model.urdf` of the model. If necessary you can add it there by copying it from ergocub-software .

By adding that to `YARP_DATA_DIRS` made `yarprobotinterface` find the URDF file. Since all the modifications I've done for gz-sim-yarp-plugins are in the SDF file version of model, is there a way to load that file or should I convert it back to a URDF?

traversaro, 2024-03-18:
> Since all the modifications I've done for gz-sim-yarp-plugins are in the SDF file version of model, is there a way to load that file or should I convert it back to a URDF?

iDynTree (the underlying library used by the wholebodydynamics module used to estimate external forces and by the walking controller) does not support loading SDF, see https://github.com/robotology/idyntree/issues/481 . However, as your modifications did not changed the kinematic and dynamics parameters of the models, I think it is sufficient to use the corresponding URDF used to create the SDF.

xela-95, 2024-03-18:
### ✅ Step 5/7
I could run:
```
yarp rpc /wholeBodyDynamics/rpc
>> resetOffset all
```

receiving the `Response: [ok]` answer.

xela-95, 2024-03-18:
### Step 6/7
When I launched the walking module with:
```
YARP_CLOCK=/clock WalkingModule
```

I got the following response:

```
[INFO] |yarp.os.Port|/IITICUBLAP218/WalkingModule/91069/clock:i| Port /IITICUBLAP218/WalkingModule/91069/clock:i active at tcp://192.168.220.201:10085/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/91069/clock:i| Receiving input from /clock to /IITICUBLAP218/WalkingModule/91069/clock:i using tcp
[ERROR] |yarp.os.Property| cannot read from dcm_walking_with_joypad.ini
[ERROR] [getiDynTreeVectorFixSizeFromSearchable] Missing field  goal_port_scaling
[ERROR] [WalkingModule::configure] Failed while reading goal_port_scaling.
[INFO] |yarp.os.RFModule| RFModule failed to open.
[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/91069/clock:i| Removing input from /clock to /IITICUBLAP218/WalkingModule/91069/clock:i

```

xela-95, 2024-03-18:
> ### Step 6/7
> 
> When I launched the walking module with:
> 
> ```
> YARP_CLOCK=/clock WalkingModule
> ```
> 
> I got the following response:
> 
> ```
> [INFO] |yarp.os.Port|/IITICUBLAP218/WalkingModule/91069/clock:i| Port /IITICUBLAP218/WalkingModule/91069/clock:i active at tcp://192.168.220.201:10085/
> [INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
> [INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
> [INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/91069/clock:i| Receiving input from /clock to /IITICUBLAP218/WalkingModule/91069/clock:i using tcp
> [ERROR] |yarp.os.Property| cannot read from dcm_walking_with_joypad.ini
> [ERROR] [getiDynTreeVectorFixSizeFromSearchable] Missing field  goal_port_scaling
> [ERROR] [WalkingModule::configure] Failed while reading goal_port_scaling.
> [INFO] |yarp.os.RFModule| RFModule failed to open.
> [WARNING] |yarp.os.NetworkClock| Destroying network clock
> [INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
> [INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/91069/clock:i| Removing input from /clock to /IITICUBLAP218/WalkingModule/91069/clock:i
> ```
> 
> from https://github.com/robotology/walking-controllers/blob/39fb59969e0219f50e63b39664839f7ed02559af/src/WalkingModule/src/Module.cpp#L151

Solved, just missed to source a setup file I use to configure needed environment variables. Now I get:

```
[INFO] |yarp.os.Port|/IITICUBLAP218/WalkingModule/101245/clock:i| Port /IITICUBLAP218/WalkingModule/101245/clock:i active at tcp://192.168.220.201:10087/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/101245/clock:i| Receiving input from /clock to /IITICUBLAP218/WalkingModule/101245/clock:i using tcp
[DEBUG] |yarp.dev.PolyDriver|remotecontrolboardremapper| Parameters are (REMOTE_CONTROLBOARD_OPTIONS (writeStrict on)) (axesNames (torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device remotecontrolboardremapper) (localPortPrefix "/walking-coordinator/remoteControlBoard") (remoteControlBoards ("/ergocubSim/torso" "/ergocubSim/left_arm" "/ergocubSim/right_arm" "/ergocubSim/left_leg" "/ergocubSim/right_leg"))
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/torso") (remote "/ergocubSim/torso") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o active at tcp://192.168.220.201:10088/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o active at tcp://192.168.220.201:10089/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i active at tcp://192.168.220.201:10090/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/left_arm") (remote "/ergocubSim/left_arm") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o active at tcp://192.168.220.201:10091/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o active at tcp://192.168.220.201:10092/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i active at tcp://192.168.220.201:10093/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o to /ergocubSim/lef [... cut ...]
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/right_leg/stateExt:i| Removing input from /ergocubSim/right_leg/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/right_leg/stateExt:i
[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/101245/clock:i| Removing input from /clock to /IITICUBLAP218/WalkingModule/101245/clock:i

```

xela-95, 2024-03-18:
The error is raised from https://github.com/robotology/walking-controllers/blob/v0.8.0/src/RobotInterface/src/Helper.cpp#L391-L401 and it is due to the fact that the `getEncoderSpeeds` method is not yet implemented in the controlboard plugin.

xela-95, 2024-03-18:
### ✅ Step 6/7

After #123 was merged the `WalkingModule` started by invoking it simply by launching `WalkingModule` in a terminal.

It produced the following logs:

```
WalkingModule
[DEBUG] |yarp.dev.PolyDriver|remotecontrolboardremapper| Parameters are (REMOTE_CONTROLBOARD_OPTIONS (writeStrict on)) (axesNames (torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device remotecontrolboardremapper) (localPortPrefix "/walking-coordinator/remoteControlBoard") (remoteControlBoards ("/ergocubSim/torso" "/ergocubSim/left_arm" "/ergocubSim/right_arm" "/ergocubSim/left_leg" "/ergocubSim/right_leg"))
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/torso") (remote "/ergocubSim/torso") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o active at tcp://192.168.220.201:10088/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o active at tcp://192.168.220.201:10089/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i active at tcp://192.168.220.201:10090/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/left_arm") (remote "/ergocubSim/left_arm") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o active at tcp://192.168.220.201:10091/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o active at tcp://192.168.220.201:10092/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i active at tcp://192.168.220.201:10093/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o to /ergocubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o to /ergocubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i| Receiving input from /ergocubSim/left_arm/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/right_arm") (remote "/ergocubSim/right_arm") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/right_arm/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/right_arm/rpc:o active at tcp://192.168.220.201:10094/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/right_arm/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/right_arm/command:o active at tcp://192.168.220.201:10095/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/right_arm/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/right_arm/stateExt:i active at tcp://192.168.220.201:10096/
[... cut ...]
	 - right_foot_task: IK-SE3Task - Frame name: r_sole Mask: true true true. Use exogenous feedback position: false. Use exogenous feedback orientation: false. Priority: 0.
	 - com_task: CoMTask Mask: true true false Priority: 0.
Note: The lower is the integer associated to the priority, the higher is the priority.
==========================

[INFO] Loading custom PIDs
[INFO] Parsing DEFAULT PID group.
[INFO] DEFAULT PID successfully loaded
[INFO] |yarp.os.Port|/walking-coordinator/logger/data:o| Port /walking-coordinator/logger/data:o active at tcp://192.168.220.201:10116/
[INFO] [WalkingModule::configure] Ready to play! Please prepare the robot.

```

xela-95, 2024-03-18:
### Step 7/7

I then tried the last step, giving commands to the walking module with `yarp rpc /walking-coordinator/rpc`.

The result to the first command, `prepareRobot` was the following:

```
q desired IK     7.90808   0.0515361 -0.00774795    -3.39176     19.4153    -12.9109     40.5797    -3.39175     19.4128    -12.9106     40.5803     5.43723    0.871996    -0.20733    -34.1722    -20.7114   -0.913073     5.45065    0.902982   -0.217773    -34.2003    -20.7262   -0.912138
[ERROR] [RobotInterface::checkMotionDone] The timer is expired but the joint  r_knee  has an error of  0.260427  radians
[ERROR] [WalkingModule::updateModule] Unable to check if the motion is done
```

Then by asking again to `prepareRobot` the result was:

```
[INFO] [WalkingModule::updateModule] Try to prepare again
q desired IK     7.90808   0.0515361 -0.00774795    -3.39176     19.4153    -12.9109     40.5797    -3.39175     19.4128    -12.9106     40.5803     5.43723    0.871996    -0.20733    -34.1722    -20.7114   -0.913073     5.45065    0.902982   -0.217773    -34.2003    -20.7262   -0.912138
[INFO] [WalkingModule::updateModule] rootlink offset  0.0457547 .
[INFO] [WalkingModule::updateModule] The robot is prepared.

``` 

Then when I asked for `startWalking` I got an error:

```
[2024-03-18 14:35:52.899] [thread: 242499] [blf] [error] [GlobalCoPEvaluator::advance] Zero contacts are active, the CoP is not valid.
[ERROR] [WalkingModule::computeGlobalCoP] Unable to compute the global CoP.
[ERROR] [WalkingModule::updateModule] Unable to compute the global CoP.
```

I don't know why the `prepareRobot` is not always working, maybe there are some control gains to be adjusted.  

While for the `startWalking` error I think this is due to https://github.com/robotology/gz-sim-yarp-plugins/issues/122#issuecomment-2002004899

xela-95, 2024-03-18:
Here's a video of the `prepareRobot` action:

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/c495563d-c78e-45bb-9dc5-27a517b6c36e

S-Dafarra, 2024-03-18:
> I don't know why the `prepareRobot` is not always working, maybe there are some control gains to be adjusted.

It is possible, yes. It checks that the robot is in the desired position considering the settling time we set. See https://github.com/robotology/walking-controllers/blob/edc8984998043c98182f726e17822deb2a7b1915/src/RobotInterface/src/Helper.cpp#L980

xela-95, 2024-03-18:
> It is possible, yes. It checks that the robot is in the desired position considering the settling time we set. See [...]

Clear, thanks! 

BTW the method that is failing during the `startWalking` phase comes from the `blf` method `advance` of class `GlobalCoPEvaluator` : https://github.com/ami-iit/bipedal-locomotion-framework/blob/1e9cb3f78b122952a5d4af7b020110911d65ab90/src/Contacts/src/GlobalCoPEvaluator.cpp#L86-L178

S-Dafarra, 2024-03-18:
> BTW the method that is failing during the `startWalking` phase comes from the `blf` method `advance` of class `GlobalCoPEvaluator` : https://github.com/ami-iit/bipedal-locomotion-framework/blob/1e9cb3f78b122952a5d4af7b020110911d65ab90/src/Contacts/src/GlobalCoPEvaluator.cpp#L86-L178

Yes it means that neither of the two local cop is acceptable, either because the normal component is too small, or because the cop is too far from the foot boundaries

xela-95, 2024-03-18:
I'm getting a curios behavior in my last tests: when preparing the robot it is always returning a `Response: [ok]` but it logs the error that `[ERROR] [RobotInterface::checkMotionDone] The timer is expired but the joint  r_elbow  has an error of  0.355696  radians`. and it writes this sometimes for `r_elbow` and sometimes for `l_elbow`.

By connecting to the robot the yarpmotorgui, I saw that the range of the elbow joints is somehow restricted to the position reached with the `prepareRobot`, so maybe the prepare fails because the even if it is asking for a certain joint position, the controlboard see the joint having reached its limit.

[image]

[image]

S-Dafarra, 2024-03-18:
> P.S. is it correct that the return message from the rpc command prepareRobot is OK even if the action did not succeed?

The ``[ok]`` you are seeing is simply an acknowledgment of the fact that the walking has received the command, not that it completed successfully.

> By connecting to the robot the yarpmotorgui, I saw that the range of the elbow joints is somehow restricted to the position reached with the `prepareRobot`, so maybe the prepare fails because even if it is asking for a certain joint position, the controlboard see the joint having reached its limit.

This is weird. Is the range bigger before opening the walking?

xela-95, 2024-03-18:
> The `[ok]` you are seeing is simply an acknowledgment of the fact that the walking has received the command, not that it completed successfully.

ah ok got it!

> This is weird. Is the range bigger before opening the walking?

No, actually it is the same. I'll try to take a look for bugs in the joint limits given by the control board, maybe some wrong conversion or things like that.

xela-95, 2024-03-19:
> No, actually it is the same. I'll try to take a look for bugs in the joint limits given by the control board, maybe some wrong conversion or things like that.

Bug found 🕵🏻! In practice the controlboard was not handling correctly the joint position limits. #126 
Since all the developments of the controlboard plugins have currently been developed and tested on a single pendulum (i.e. single joint) I have to review the code in order to guarantee it works correctly for multiple joint models.

xela-95, 2024-03-19:
@S-Dafarra could you tell me if there's a way to get more info about the cop error arising from blf? 

Like getting the result of the cop computation to understand what is the reason behind the `[blf] [error] [GlobalCoPEvaluator::advance] Zero contacts are active, the CoP is not valid.` error

xela-95, 2024-03-19:
With 
```
yarp read ... /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o
```
it is possible to get the 6D wrench of the foot. Now I got a 6D vector of zeros (values under $10^{-10}$):

```
-2.71864789138483145661e-12 4.49089338582689669406e-15 1.84872575288608347244e-11 8.1555891253485101889e-17 -1.12476848873726913157e-12 -5.38235428355966719317e-16
```

With 
```
yarp read "..." /ergocubSim/right_leg/FT/measures:o 
```
it is possible to read the measures from the leg FT (leg, foot rear, foot front).  I'm getting all zeros:

```
() () () () () (((0.0 0.0 0.0 0.0 0.0 0.0) 71.3400000000000034106) ((0.0 0.0 0.0 0.0 0.0 0.0) 71.3400000000000034106) ((0.0 0.0 0.0 0.0 0.0 0.0) 71.3400000000000034106)) () () () ()
```
(the scalar different from zero is the timestamp of the measurement).

xela-95, 2024-03-19:
Side note: to visualize contact in Gazebo there's a dedicated plugin called "Visualize contact" that can be added from the search bar on the top right corner of the GUI:

[image]

xela-95, 2024-03-19:
@traversaro @S-Dafarra do you know where I can find the configuration of the physics engine used in Gazebo classic when using the walking controllers? Or the defaults provided by Gazebo are used? For example, in Gazebo sim there's the [`<physics>`](http://sdformat.org/spec?ver=1.11&elem=physics#world_physics) element (property of the world element), where all the properties related to physics engine, contacts, etc can be configured.

traversaro, 2024-03-19:
> @traversaro @S-Dafarra do you know where I can find the configuration of the physics engine used in Gazebo classic with the walking controllers? Or the defaults provided by Gazebo are used? For example, in Gazebo sim there's the [`<physics>`](http://sdformat.org/spec?ver=1.11&elem=physics#world_physics) element (property of the world element), where all the properties related to physics engine, contacts, etc can be configured.

I think the walking is typically tested with default Gazebo Classic configuration. However I think @GiulioRomualdi is working on a paper repo with a specific world, I Can share it to you.

traversaro, 2024-03-19:
Are you sure that you are not using a _fixed model where there is a joint between the world and the model?

xela-95, 2024-03-19:
> Are you sure that you are not using a _fixed model where there is a joint between the world and the model?

In principle there was such fixed joint, but now to try the walking tutorial I commented it out.

xela-95, 2024-03-19:
> That seems to indicate a problem in the FT sensor plugin. Also all other FT sensors report zero?

Yep, all the FTs read zero. I've verified that I didn't introduced regression bugs since the controlboard [tutorial on the FT sensor](https://github.com/robotology/gz-sim-yarp-plugins/tree/main/tutorial/forcetorque/model_one_sensor) works. 

I don't know if it's ok but also listing the gazebo topics I get:

```
$ gz topic -l
/clock
/gazebo/resource_paths
/gui/camera/pose
/sensors/marker
/stats
/world/turorial_controlboard/clock
/world/turorial_controlboard/dynamic_pose/info
/world/turorial_controlboard/model/ergocub_fixed/model/ergoCub/link/realsense/sensor/realsense_head_rgb/camera_info
/world/turorial_controlboard/model/ergocub_fixed/model/ergoCub/link/realsense/sensor/realsense_head_rgb/image
/world/turorial_controlboard/pose/info
/world/turorial_controlboard/scene/deletion
/world/turorial_controlboard/scene/info
/world/turorial_controlboard/state
/world/turorial_controlboard/stats
/world/turorial_controlboard/light_config
/world/turorial_controlboard/material_color

```

traversaro, 2024-03-19:
> Yep, all the FTs read zero. I've verified that I didn't introduced regression bugs since the controlboard [tutorial on the FT sensor](https://github.com/robotology/gz-sim-yarp-plugins/tree/main/tutorial/forcetorque/model_one_sensor) works.

In the tutorial the plugin is placed in the xml **before** the sensor, in the ergocub SDF **after**, did you tried to change the order?

xela-95, 2024-03-19:
> In the tutorial the plugin is placed in the xml **before** the sensor, in the ergocub SDF **after**, did you tried to change the order?

Just tried, same result. I'll try to see if the wrappers and remappers are responsible

xela-95, 2024-03-19:
Thank to @traversaro the issue of FTs and IMUs not communicating was solved. This was due to the fact  there were some gazebo plugins missing:

```xml
<plugin
  filename="gz-sim-forcetorque-system"
  name="gz::sim::systems::ForceTorque">
</plugin>
<plugin
  filename="gz-sim-imu-system"
  name="gz::sim::systems::Imu">
</plugin>
```

Now I can read measurements via `yarp read`.

The `/wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o` reports:
```
1.06954221966386464615 47.879756222723536041 131.967328336028657532 -6.80291645844825865197 -4.59942486510437120728 2.04004190919274908111
```

When trying to do `setGoal` to make the robot walk I'm still getting the blf error `[blf] [error] [GlobalCoPEvaluator::advance] Zero contacts are active, the CoP is not valid.` and then the `WalkingModule` stops.

traversaro, 2024-03-19:
> Thank to @traversaro the issue of FTs and IMUs not communicating was solved. This was due to the fact there were some gazebo plugins missing:

We should at least print an error if that is the case.

xela-95, 2024-03-19:
### Physics engine configurations
http://sdformat.org/spec?ver=1.11&elem=physics

#### Engine: DART (default)

| `collision_detector` | Result |
|--------|--------|
| fcl (default) | Some steps performed and then fell down. | 
| ode | Some steps performed and then fell down. Simulation slower than tcl case. |
| dart | Cannot be used with collision meshes, only with simple shapes. It gives the error: `Error [DARTCollide.cpp:1649] [DARTCollisionDetector] Attempting to check for an unsupported shape pair: [MeshShape] - [BoxShape]. Returning false.` |
| bullet | Unstable behavior. It wasn't able to do any steps before getting the blf error for the CoP |

Videos:

**FCL**

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/d44f6bd6-75ad-427b-8c8c-6d430aa127ba

**ODE**

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/b6517ed2-e3c3-4fc1-a223-1326598bfbe9

**Bullet**

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/15018717-c252-4dc1-9843-8ebaa1187b1e

xela-95, 2024-03-19:
As @S-Dafarra pointed out, the weird behavior for which the `prepareRobot` does not work the first time can be due to the clock not being appropriately synchronized. I'm looking into it.

xela-95, 2024-03-19:
> We should at least print an error if that is the case.

Agree, I have to understand how to check if that plugins are available in the world.

traversaro, 2024-03-19:
> fcl (default)

I am not sure this is true. If you see https://github.com/gazebosim/gz-sim/blob/gz-sim8/tutorials/physics.md?plain=1#L100, there they report that ode is the default collision detector, and I was expecting that as that is what is tested in gz-physics tests (https://github.com/gazebosim/gz-physics/blob/7d30e47fb61fe4db9cba460717a3910df9deabee/dartsim/src/WorldFeatures_TEST.cc#L87). 
Probably sdformat reports the default in Gazebo Classic, while the default in gz-sim/gz-physics is different.

Can you open an issue in https://github.com/gazebosim/gz-sim to report the issue? Thanks!

xela-95, 2024-03-19:
> You may want to update your libode package version, during the weekend we fixed some critical bugs (see [conda-forge/libode-feedstock#22](https://github.com/conda-forge/libode-feedstock/pull/22) and [conda-forge/libode-feedstock#23](https://github.com/conda-forge/libode-feedstock/pull/23)).

is the latest libode version the 0.16.2? https://anaconda.org/conda-forge/libode

traversaro, 2024-03-19:
Yes, but the fix is in the latest build, i.e. **14**, so make sure to have installed `libode==0.16.2=*_14` , just updating libode should install the latest build

xela-95, 2024-03-19:
> Can you open an issue in https://github.com/gazebosim/gz-sim to report the issue? Thanks!

Done here: https://github.com/gazebosim/gz-sim/issues/2338

xela-95, 2024-03-20:
One of the major issues now is to guarantee that the Gazebo clock is set as Yarp clock on the `/clock` port, otherwise the control and the simulator will run at different speed. 

Here are my attempts:
- By setting the environment variable `YARP_CLOCK` when I launch gazebo with `gz sim example.workd --verbose` Gazebo remains stuck (with a black GUI) logging:
```
[INFO] |yarp.os.Port|/IITICUBLAP218/gz/79976/clock:i| Port /IITICUBLAP218/gz/79976/clock:i active at tcp://10.240.2.19:10002/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[ERROR] |yarp.os.NetworkClock| Cannot find time port "/clock" or a time topic "/clock@"
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
```

- If I just try to set `YARP_CLOCK` when launching the WalkingModule by launching it with: `YARP_CLOCK=/clock WalkingModule` I got:
```
[INFO] |yarp.os.Port|/IITICUBLAP218/WalkingModule/82982/clock:i| Port /IITICUBLAP218/WalkingModule/82982/clock:i active at tcp://10.240.2.19:10085/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/82982/clock:i| Receiving input from /clock to /IITICUBLAP218/WalkingModule/82982/clock:i using tcp
[DEBUG] |yarp.dev.PolyDriver|remotecontrolboardremapper| Parameters are (REMOTE_CONTROLBOARD_OPTIONS (writeStrict on)) (axesNames (torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device remotecontrolboardremapper) (localPortPrefix "/walking-coordinator/remoteControlBoard") (remoteControlBoards ("/ergocubSim/torso" "/ergocubSim/left_arm" "/ergocubSim/right_arm" "/ergocubSim/left_leg" "/ergocubSim/right_leg"))
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/torso") (remote "/ergocubSim/torso") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o active at tcp://10.240.2.19:10086/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o active at tcp://10.240.2.19:10087/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i active at tcp://10.240.2.19:10088/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/ergocubSim/left_arm") (remote "/ergocubSim/left_arm") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o active at tcp://10.240.2.19:10089/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o active at tcp://10.240.2.19:10090/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i| Port /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i active at tcp://10.240.2.19:10091/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/rpc:o to /ergocubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o| Sending output from /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/command:o to /ergocubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i| Receiving input from /ergocubSim/left_arm/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/left_arm/stateExt:i using udp
[... cut ...]
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/ergocubSim/right_leg/stateExt:i| Removing input from /ergocubSim/right_leg/stateExt:o to /walking-coordinator/remoteControlBoard/ergocubSim/right_leg/stateExt:i
[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/WalkingModule/82982/clock:i| Removing input from /clock to /IITICUBLAP218/WalkingModule/82982/clock:i
```
and then stops. I think this is the right path, I'm investigating the error of the WalkingModule

traversaro, 2024-03-20:
> By setting the environment variable `YARP_CLOCK` when I launch gazebo with `gz sim example.workd --verbose` Gazebo remains stuck (with a black GUI) logging:

This is probably a bug, ideally if the network clock is set we should avoid requiring the clock in the plugin that publish it, can you open an issue for that?

xela-95, 2024-03-20:
> This is probably a bug, ideally if the network clock is set we should avoid requiring the clock in the plugin that publish it, can you open an issue for that?

Sorry I'm not sure to have fully understood, can you rephrase?

xela-95, 2024-03-20:
> [ERROR] [configure] Unable to read encoders.

This error comes from https://github.com/robotology/walking-controllers/blob/edc8984998043c98182f726e17822deb2a7b1915/src/RobotInterface/src/Helper.cpp#L391-L401

The strange behavior thing is that when I tried logging when the `getEncoders` and `getEncoderSpeeds` method si invoked, the method never logged anything.
https://github.com/robotology/gz-sim-yarp-plugins/blob/6b25488fb793743ff90d00d9c9184e7e86f5031c/plugins/controlboard/src/ControlBoardDriver.cpp#L1612-L1629

xela-95, 2024-03-20:
I've also made an attempt by launching the yarprobotinterface using the YARP_CLOCK variable and this is what I got:

<details><summary>Logs</summary>

```
[INFO] |yarp.os.Port|/IITICUBLAP218/yarprobotinterface/32264/clock:i| Port /IITICUBLAP218/yarprobotinterface/32264/clock:i active at tcp://10.240.2.19:10045/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP218/yarprobotinterface/32264/clock:i| Receiving input from /clock to /IITICUBLAP218/yarprobotinterface/32264/clock:i using tcp
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[DEBUG] Reading file /home/acroci/ergocub_ws/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[WARNING] Invalid syntax while loading /home/acroci/ergocub_ws/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml at line 1 . Expected document of type robot . Found devices
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /home/acroci/ergocub_ws/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[INFO] Yarprobotinterface was started using the following enable_tags: 
[INFO] Yarprobotinterface was started using the following disable_tags: 
[DEBUG] List of all enable attributes found in the include tags: 
[DEBUG] List of all disable attributes found in the include tags: 
[DEBUG] Preprocessor complete in:  0 s
[WARNING] Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
[WARNING] *************************************************************************************
* yarprobotinterface 'portprefix' parameter does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name    *
*     name:    full port prefix name with leading '/',  e.g.  /robotName      *
* A temporary automatic fix will be done for you, but please fix your config file   *
*************************************************************************************
[INFO] |yarp.os.Port|/ergoCubGazeboV1/yarprobotinterface| Port /ergoCubGazeboV1/yarprobotinterface active at tcp://10.240.2.19:10046/
[INFO] startup phase starting...
[INFO] Opening device torso_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/torso"), ("local" = "/wholeBodyDynamics/torso")]
[DEBUG] |yarp.dev.PolyDriver|torso_mc| Parameters are (device remote_controlboard) (id torso_mc) (local "/wholeBodyDynamics/torso") (remote "/ergocubSim/torso") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/rpc:o| Port /wholeBodyDynamics/torso/rpc:o active at tcp://10.240.2.19:10047/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/command:o| Port /wholeBodyDynamics/torso/command:o active at tcp://10.240.2.19:10048/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/stateExt:i| Port /wholeBodyDynamics/torso/stateExt:i active at tcp://10.240.2.19:10049/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/rpc:o| Sending output from /wholeBodyDynamics/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/command:o| Sending output from /wholeBodyDynamics/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|torso_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device left_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/left_arm"), ("local" = "/wholeBodyDynamics/left_arm")]
[DEBUG] |yarp.dev.PolyDriver|left_arm_mc| Parameters are (device remote_controlboard) (id left_arm_mc) (local "/wholeBodyDynamics/left_arm") (remote "/ergocubSim/left_arm") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/rpc:o| Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://10.240.2.19:10050/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/command:o| Port /wholeBodyDynamics/left_arm/command:o active at tcp://10.240.2.19:10051/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/stateExt:i| Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://10.240.2.19:10052/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/rpc:o| Sending output from /wholeBodyDynamics/left_arm/rpc:o to /ergocubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/command:o| Sending output from /wholeBodyDynamics/left_arm/command:o to /ergocubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/left_arm/stateExt:i| Receiving input from /ergocubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|left_arm_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device right_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/right_arm"), ("local" = "/wholeBodyDynamics/right_arm")]
[DEBUG] |yarp.dev.PolyDriver|right_arm_mc| Parameters are (device remote_controlboard) (id right_arm_mc) (local "/wholeBodyDynamics/right_arm") (remote "/ergocubSim/right_arm") (robotName ergoCubGazeboV1)
[... cut ...]
[ERROR] |yarp.device.multipleanalogsensorsclient| Sensor of type SixAxisForceTorqueSensors with index 0 has non-MAS_OK status.
[DEBUG] yarprobotinterface running happily
[ERROR] |yarp.device.multipleanalogsensorsclient| No data received in the last 3000000.000000 seconds, timeout enabled.
[ERROR] |yarp.device.multipleanalogsensorsclient| Sensor of type ThreeAxisLinearAccelerometers with index 0 has non-MAS_OK status.
[ERROR] |yarp.device.multipleanalogsensorsclient| Sensor of type ThreeAxisGyroscopes with index 0 has non-MAS_OK status.
[DEBUG] yarprobotinterface running happily
```

</details>

xela-95, 2024-03-20:
Thanks to @traversaro we found out that the format of the timestamps published on the `/clock` port by the clock plugin is not what the Yarp Clock and the other modules were expecting.

We found this by comparing the signal through `yarp read ... /clock` with a second clock created through:

```console
YARP_CLOCK=/clock2 yarp clock /clock2 
```

By reading this clock the format is `seconds nanoseconds` like: `63 936749458`

I then updated the clock plugin to use this format and then done the walking tutorial being careful to launch both yarprobotinterface and Walking module with the YARP_CLOCK environment variable set to `/clock`.

Here's a video of the result 🥳 

https://github.com/robotology/gz-sim-yarp-plugins/assets/57228872/03d230e1-5dec-485c-a871-0b7dae804b1e

The simulation is slow (sim2real ratio of 10-30 %) but that's matter for future optimizations to the plugins :)

xela-95, 2024-03-20:
CC @S-Dafarra @GiulioRomualdi let me know if you note something strange in the simulation or if you want me to try change some simulation settings :)

traversaro, 2024-03-20:
> The simulation is slow (sim2real ratio of 10-30 %) but that's matter for future plugins optimizations :)

My guess is that is not something plugin-related, but rather that the model is quite complex. Probably once we integrate the plugin in the ergoCub creo generation pipeline we can also test the minContact model.

S-Dafarra, 2024-03-20:
In my opinion, some possible investigation routes could be:
- how fast does it go without the visualization?
- are the collisions active? If so, what about removing them all manually except those in the feet?
- is it possible to chop away part of the robot (like the head and the forearms) and check how much of an improvement it is?

xela-95, 2024-03-20:
I've tried to change physics engine: in addition to Dartsim, bullet and bullet-featherstone are the ones officially supported.

By launching gazebo with 
```console
gz sim example.world --physics-engine gz-physics-bullet-plugin --verbose
```

Gazebo crashes with the following logs:

<details><summary>Details</summary>

```
[Wrn] [Physics.cc:1191] Model's parent entity [8] not found on world / model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[Wrn] [Physics.cc:1242] Link's parent entity [9] not found on model map.
[... 206 lines cut ...]
[Err] [Physics.cc:2930] Internal error: link [175] not in entity map
[Err] [Physics.cc:2930] Internal error: link [178] not in entity map
[Err] [Physics.cc:2930] Internal error: link [181] not in entity map
[Err] [Physics.cc:2930] Internal error: link [184] not in entity map
[Err] [Physics.cc:2930] Internal error: link [187] not in entity map
[Err] [Physics.cc:2930] Internal error: link [190] not in entity map
[Err] [Physics.cc:2930] Internal error: link [193] not in entity map
[Err] [Physics.cc:2930] Internal error: link [196] not in entity map
[Err] [Physics.cc:2930] Internal error: link [199] not in entity map
[Err] [Physics.cc:2930] Internal error: link [202] not in entity map
[Err] [Physics.cc:2930] Internal error: link [205] not in entity map
[Err] [Physics.cc:2930] Internal error: link [208] not in entity map
[Err] [Physics.cc:2930] Internal error: link [211] not in entity map
[Err] [Physics.cc:2930] Internal error: link [214] not in entity map
terminate called recursively
[Msg] Loading plugin [gz-rendering-ogre2]
terminate called recursively
terminate called recursively
terminate called recursively
Stack trace (most recent call last) in thread 300336:
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]

```

</details>

traversaro, 2024-03-20:
> Gazebo crashes with the following logs:

Cool, probably it make sense to check this happens without any plugin, and then report the issue in gz-sim .

xela-95, 2024-03-20:
> In my opinion, some possible investigation routes could be:
>
>     * how fast does it go without the visualization?
>
>     * are the collisions active? If so, what about removing them all manually [...]

I think I can close this issue since its acceptance criteria have been met.

I will open a new issue to track how the simulation changes by following these investigation routes.

closed 2024-03-20 by xela-95