robotology/gz-sim-yarp-plugins #129: Investigate how to speed up the simulation of ergocub with walking-controllers

xela-95, 2024-03-20:


traversaro, 2024-03-20:
>     * [ ]  Remove collisions except for feet

We already have this model in ergocub-models, it is called `ergoCubGazeboV1_1_minContacts` . At this point, probably you can consider looking into adding support for gz-sim-yarp-plugins into the models generated by https://github.com/icub-tech-iit/ergocub-software, and then you will get that model to test for free (fyi @Nicogene).

xela-95, 2024-03-21:
> We already have this model in ergocub-models, it is called `ergoCubGazeboV1_1_minContacts` . At this point, probably you can consider looking into adding support for gz-sim-yarp-plugins into the [...]

Opened an issue for this: #130