robotology/gz-sim-yarp-plugins #130: Integrate plugins into the ergoCub urdf generation process

xela-95, 2024-03-21:


xela-95, 2024-03-21:
CC @traversaro

traversaro, 2024-03-21:
I discussed with @Nicogene, and at the moment the process is only mentioned in https://github.com/icub-tech-iit/ergocub-software's README, but it is not documented in detail.

Just reporting some info I know:
* You need to install Creo 9 on your Windows partition, see https://github.com/icub-tech-iit/cad-mechanics/wiki/Install-and-configure-PTC-Creo-Parametric (sorry, this is an internal IIT repo)
* You need to install creo2urdf following the README in https://github.com/icub-tech-iit/creo2urdf, probably you can try the binaries available in each release, or compile from source

Once you have that, you probably need this info:

| CAD model in `cad-mechanics` | yaml file | csv file | YARP_ROBOT_NAME |
|:----------:|:---------:|:----------:|:--------------:|
| `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` | `ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options.yaml` | `ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options.csv` | `ergoCubSN000`  |
| `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` | `ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.yaml` | `ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.csv` | `ergoCubGazeboV1`  |
| `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` | `ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.yaml` | `ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.csv` | `ergoCubGazeboV1_minContacts`  |
| `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-1.asm.1` | `ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options.yaml` | `ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options.csv` | `ergoCubSN001`  |
| `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-1.asm.1` | `ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.yaml` | `ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.csv` | `ergoCubGazeboV1_1`  |
| `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` | `ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.yaml` | `ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.csv` | `ergoCubGazeboV1_1_minContacts`  |

traversaro, 2024-03-21:
> I discussed with @Nicogene, and at the moment the process is only mentioned in https://github.com/icub-tech-iit/ergocub-software's README, but it is not documented in detail.
> 
> Just reporting some info I know:
> 
>     * You need to install Creo 9 on your Windows partition, see https://github.com/icub-tech-iit/cad-mechanics/wiki/Install-and-configure-PTC-Creo-Parametric (sorry, this is an internal IIT repo)
> 
>     * You need to install creo2urdf following the README in https://github.com/icub-tech-iit/creo2urdf, probably you can try the binaries available in each release, or compile from source
> 
>     * You need to install the repos that contain the CAD models, i.e. cad-libraries (see https://github.com/icub-tech-iit/cad-libraries/wiki/Configure-PTC-Creo-with-cad-libraries) and cad-mechanics (sorry for anyone reading this, it is private): https://github.com/icub-tech-iit/cad-mechanics/
> 
> 
> Once you have that, you probably need this info:
> CAD model in `cad-mechanics` 	yaml file 	csv file 	YARP_ROBOT_NAME
> `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options.yaml` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options.csv` 	`ergoCubSN000`
> `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.yaml` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.csv` 	`ergoCubGazeboV1`
> `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.yaml` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_0/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.csv` 	`ergoCubGazeboV1_minContacts`
> `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-1.asm.1` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options.yaml` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options.csv` 	`ergoCubSN001`
> `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-1.asm.1` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.yaml` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_gazebo.csv` 	`ergoCubGazeboV1_1`
> `cad-mechanics/projects/simulation_model/ergocub_1-0/sim_ergocub_1-0.asm.1` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.yaml` 	`ergocub-software/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options/ERGOCUB_all_options_minContacts.csv` 	`ergoCubGazeboV1_1_minContacts`

I ported most of this docs in https://github.com/icub-tech-iit/ergocub-software/pull/229 .

xela-95, 2024-03-22:
@traversaro which tool can I use to convert an SDF to URDF? I've always done the inverse with the `gz sdf` command. Having a URDF will help me understand the correct way to specify the elements to add in the yaml file for the `creo2urdf ` process.

traversaro, 2024-03-22:
> @traversaro which tool can I use to convert an SDF to URDF? I've always done the inverse with the `gz sdf` command. Having a URDF will help me understand the correct way to specify the elements to add in the yaml file for the `creo2urdf ` process.

I do not think there exist any tool that does that, while correctly producing `<gazebo>` extensions elements.

traversaro, 2024-03-22:
> @traversaro where is the assembly I need to load in CREO?

See the previous comment:

> You need to install the repos that contain the CAD models, i.e. cad-libraries (see https://github.com/icub-tech-iit/cad-libraries/wiki/Configure-PTC-Creo-with-cad-libraries) and cad-mechanics (sorry for anyone reading this, it is private): https://github.com/icub-tech-iit/cad-mechanics/

then you can get the exact assembly corresponding to each model in the README of https://github.com/icub-tech-iit/ergocub-software/tree/master/urdf/creo2urdf/data/ergocub1_1 .

xela-95, 2024-03-22:
> See the previous comment:
>
>
> then you can get the exact assembly corresponding to each model in the README of [...]

🤦🏻‍♂️ sorry I forgot you had already answered to this

xela-95, 2024-03-22:
Draft PR added: https://github.com/icub-tech-iit/ergocub-software/pull/230

xela-95, 2024-03-25:
@traversaro I have a doubt on whether to include the clock plugin inside the model or leave it to be defined by the users of ergoCub at a "world" level. In principle, if the clock plugin is defined at model level and there is only on ergocub present in the world it works, but in presence of multiple models there could be issues since all the plugins want to write on the same `/clock` port.

traversaro, 2024-03-25:
> @traversaro I have a doubt on whether to include the clock plugin inside the model or leave it to be defined by the users of ergoCub at a "world" level. In principle, if the clock plugin is defined at model level and there is only on ergocub present in the world it works, but in presence of multiple models there could be issues since all the plugins want to write on the same `/clock` port.
> 
> To me the best approach is to define it in the world in which the model is loaded, but the user should be aware of it (e.g. in the walking-controllers [tutorial ](https://github.com/robotology/walking-controllers?tab=readme-ov-file#computer-how-to-run-the-simulation) it is explicitly written to launch gazebo-classic with the clock plugin as: `gazebo -slibgazebo_yarp_clock.so`

Yes, the clock should not be in the model. In general, it seems to me a good idea for reproducibility to strongly suggest users to use `.world` files to launch gz-sim simulations, as "dragging models manually from the gui" is an inherently non-reproducible operation. I guess when we will migrate the walking-controllers documentation to use gz-sim, we will change the `gazebo -slibgazebo_yarp_clock.so` command to be something like `gazebo ergocub_with_clock.world` , where the `ergocub_with_clock.world` file is available somewhere.

xela-95, 2024-03-25:
> Yes, the clock should not be in the model. In general, it seems to me a good idea for reproducibility to strongly suggest users to use `.world` files to launch gz-sim simulations, as "dragging [...]

Thanks, totally agree.

xela-95, 2024-04-04:
PR https://github.com/icub-tech-iit/ergocub-software/pull/230 has been merged. Closing

closed 2024-04-04 by xela-95