robotology/gz-sim-yarp-plugins #131: Insert the repo into the robotology superbuild

xela-95, 2024-03-21:
Robotology superbuild: https://github.com/robotology/robotology-superbuild

xela-95, 2024-03-21:
CC @traversaro

traversaro, 2024-03-21:
See the docs in https://github.com/robotology/robotology-superbuild/blob/master/doc/developers-faqs.md#how-to-add-a-new-package . As we plan to add the package in conda-forge, notice that we need to let know that to the robotology-superbuild by setting the `gz-sim-yarp-plugins_CONDA_PKG_CONDA_FORGE_OVERRIDE` option, see https://github.com/robotology/robotology-superbuild/blob/master/doc/developers-faqs.md#how-to-ensure-that-binary-packages-are-correctly-generated-for-a-new-package . This is useful to ensure that no `gz-sim-yarp-plugins` package is added in the robotology conda channel.

xela-95, 2024-03-29:
Where I should put the dependencies of the package? The `DEPENDS` field of `ycm_ep_helper` is only for packages inside the superbuild? https://github.com/xela-95/robotology-superbuild/blob/056c18b2c0ca5e953705870b278463bdd1415eb2/cmake/Buildgz-sim-yarp-plugins.cmake#L15-L16

In general what other files should I modify other than the [Buildgz-sim-yarp-plugins.cmake](https://github.com/xela-95/robotology-superbuild/blob/056c18b2c0ca5e953705870b278463bdd1415eb2/cmake/Buildgz-sim-yarp-plugins.cmake)?

traversaro, 2024-03-29:
> Where I should put the dependencies of the package? The `DEPENDS` field of `ycm_ep_helper` is only for packages inside the superbuild? https://github.com/xela-95/robotology-superbuild/blob/056c18b2c0ca5e953705870b278463bdd1415eb2/cmake/Buildgz-sim-yarp-plugins.cmake#L15-L16

Yes. Other dependencies go in the documentation and in the ci script (unfortunately duplicated).

> In general what other files should I modify other than the [Buildgz-sim-yarp-plugins.cmake](https://github.com/xela-95/robotology-superbuild/blob/056c18b2c0ca5e953705870b278463bdd1415eb2/cmake/Buildgz-sim-yarp-plugins.cmake)?

I would say:
* All places where `ROBOTOLOGY_USES_GAZEBO` is used (as you should have `ROBOTOLOGY_USES_GZ_SIM` instead (see https://github.com/search?q=repo%3Arobotology%2Frobotology-superbuild%20ROBOTOLOGY_USES_GAZEBO&type=code)
* All places where `GazeboYARPPlugins` is used (see https://github.com/search?q=repo%3Arobotology%2Frobotology-superbuild+GazeboYarpPlugins&type=code), except old releases files, for what regards releases you only need to modify `latest.releases.yaml`

xela-95, 2024-03-31:
In order to test if I made all the necessary updates, is it sufficient to check the CI is passing or should I try some of the methods in which the robotology can be installed?

traversaro, 2024-03-31:
> In order to test if I made all the necessary updates, is it sufficient to check the CI is passing or should I try some of the methods in which the robotology can be installed?

I guess CI is enough, as long as you check that CI actually enables the option you want to test.

traversaro, 2024-04-03:
To avoid the mutual exclusivity between gazebo-classic and gz-sim8, we can use the following changes:
* Experimental packages that permit side-by-side installation in apt: https://gist.github.com/j-rivero/430c2dc6c73860dd72116ad362413ac3
* Side-by-side installation in conda-forge : https://github.com/conda-forge/gazebo-feedstock/pull/208

xela-95, 2024-04-05:
https://github.com/robotology/robotology-superbuild/pull/1628 has been merged. Closing!

closed 2024-04-05 by xela-95