robotology/gz-sim-yarp-plugins #132: Publish package on conda-forge

xela-95, 2024-03-21:


xela-95, 2024-03-21:
CC @traversaro

traversaro, 2024-03-21:
Once we have a working release, we can prepare a PR to `staged-recipes` as documented in https://conda-forge.org/docs/maintainer/adding_pkgs/#creating-recipes . For a reference, you can see the PR that added gazebo-yarp-plugins: https://github.com/conda-forge/staged-recipes/pull/21891 .

traversaro, 2024-03-21:
> For a reference, you can see the PR that added gazebo-yarp-plugins: https://github.com/conda-forge/staged-recipes/pull/21891 .

Note that since that has been created, the recipe changed, you can see the latest in https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/main/recipe/meta.yaml . I suggest to use the recipe in https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/main/recipe/meta.yaml as a model, so we can have both `libgz-sim-yarp-plugins` package (where `lib` is required by naming rules of conda-forge) and `gz-sim-yarp-plugins` meta-package (that just depends on `libgz-sim-yarp-plugins`, but does not have any run_exports and it is easier for users to remember).

xela-95, 2024-03-29:
@traversaro I'm compiling the requirements section of the `meta.yaml` file. I was thinking to use the dependencies in the `ci.yaml` we use for the conda CI. Other than that, in our CI we install additional dependencies for Linux:
https://github.com/robotology/gz-sim-yarp-plugins/blob/fe4bbe64f151ffa68d934cc7da0dea8651775cd3/.github/workflows/conda-forge.yml#L40-L44
SHould I add those?

xela-95, 2024-03-29:
I have other doubts:

- What is and where I can find the corresponding file of `GazeboYARPPluginsConfig.cmake`: 
https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/68e63851eff50994ccdf94a9872aebc40542a4c7/recipe/meta.yaml#L54
- There's a reason to test only for existence of `common.h` in this output and not for all plugins as done above?

xela-95, 2024-03-29:
@traversaro do you know what are the correct locations on Windows to which the env. vars `GZ_SIM_SYSTEM_PLUGIN_PATH` and `GZ_SIM_RESOURCE_PATH` has to be set? I've seen in https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/68e63851eff50994ccdf94a9872aebc40542a4c7/recipe/activate.msh that for the variables needed by gazebo-classic they are different from Linux (of course).

xela-95, 2024-03-29:
Do I need to have the [conda_build_config.yaml](https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/68e63851eff50994ccdf94a9872aebc40542a4c7/recipe/conda_build_config.yaml) with macOS deployment configurations?

traversaro, 2024-03-29:
> Should I add those?

Those are OS-specific dependencies. conda-forge CI does not use an apt-based Linux distribution, but one based on yum. So you can add them via `yum_requirements.txt`, like we do in gazebo-yarp-plugins: https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/main/recipe/yum_requirements.txt .

traversaro, 2024-03-29:
> Do I need to have the [conda_build_config.yaml](https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/68e63851eff50994ccdf94a9872aebc40542a4c7/recipe/conda_build_config.yaml) with macOS deployment configurations?

Probably not. You can start without, and it the CI complains we add them.

traversaro, 2024-03-29:
> * What is and where I can find the corresponding file of `GazeboYARPPluginsConfig.cmake`:
>       https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/68e63851eff50994ccdf94a9872aebc40542a4c7/recipe/meta.yaml#L54

We are not installing any at the moment, so you can just skip that.

traversaro, 2024-03-29:
> @traversaro do you know what are the correct locations on Windows to which the env. vars `GZ_SIM_SYSTEM_PLUGIN_PATH` and `GZ_SIM_RESOURCE_PATH` has to be set? I've seen in [...]

For `GZ_SIM_SYSTEM_PLUGIN_PATH`, we need to check where are we installing the plugins on Windows. For `GZ_SIM_RESOURCE_PATH`, as we are not installing any model in this repo probably we do not need to set them?

xela-95, 2024-03-29:
> For `GZ_SIM_SYSTEM_PLUGIN_PATH`, we need to check where are we installing the plugins on Windows. For `GZ_SIM_RESOURCE_PATH`, as we are not installing any model in this repo probably we do not need to set them?

Ok I will try to install it on windows and check the installation path. for the `GZ_SIM_RESOURCE_PATH` you're right, I was thinking about the tutorials but they don't need it.

xela-95, 2024-03-29:
> Those are OS-specific dependencies. conda-forge CI does not use an apt-based Linux distribution, but one based on yum. So you can add them via `yum_requirements.txt`, like we do in [...]

Ok thanks! in the `yum_requirements.txt` should I report only the apt-based dependencies or also the ones installed only for linux in https://github.com/robotology/gz-sim-yarp-plugins/blob/fe4bbe64f151ffa68d934cc7da0dea8651775cd3/.github/workflows/conda-forge.yml#L40?

traversaro, 2024-03-29:
> Ok I will try to install it on windows and check the installation path.

You can just check the CI logs.

traversaro, 2024-03-29:
> Ok thanks! in the `yum_requirements.txt` should I report only the apt-based dependencies or also the ones installed only for linux in [...]

Only the one related to OpenGL or similar, not all the dependency. You can just copy the one used in gazebo-yarp-plugins, in general yum and apt packages have different names (as conda-forge packages and apt packages have different names).

xela-95, 2024-04-10:
https://github.com/conda-forge/staged-recipes/pull/25910 has been merged and the package is available. Closing.

closed 2024-04-10 by xela-95