robotology/gz-sim-yarp-plugins #133: Launch gazebo on WSL

xela-95, 2024-03-22:
This issue tracks all the necessary steps to make gazebo and gz-sim-yarp-plugins work on WSL, with particular reference to the case of ErgoCub robot (see #122 )

xela-95, 2024-03-22:
# Preliminary steps

## WSL
- Install WSL2: https://learn.microsoft.com/en-us/windows/wsl/install
- Install Ubuntu 22.04 LTS from the Windows store: https://www.microsoft.com/store/productId/9PN20MSR04DW?ocid=pdpshare

## Mamba and repositories
- Install mamba: https://github.com/robotology/robotology-superbuild/blob/master/doc/install-mambaforge.md
- Create conda environment: https://github.com/robotology/gz-sim-yarp-plugins/tree/main
- Clone this repository

xela-95, 2024-03-22:
## Import ErgoCub
- Clone the temporary repository https://github.com/xela-95/ergocub_ws
- Update in the `setup.sh` the env. vars `GZ_SIM_RESOURCE_PATH` and `YARP_DATA_DIRS`
- Activate conda environment
- Launch `yarp server`
- cd to the ergocub_ws folder
- `source setup.sh`
- Launch `gz sim example.world -v 4`

xela-95, 2024-03-22:
fyi @traversaro

xela-95, 2024-03-22:
I'm facing an issue since gazebo cannot find the gz-sim-yarp-plugins shared libraries, in fact when I execute:
```console
gz sim example.world -v 4
```

I get the following logs:
<details><summary>Details</summary>

```
[Msg] Gazebo Sim GUI    v8.1.0
[Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:263] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [example.world] from the GUI.
[Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[Msg] Gazebo Sim Server v8.1.0
[Msg] Loading SDF world file[/home/acroci/repos/ergocub_ws/example.world].
QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
[Dbg] [Application.cc:102] Initializing application.
[Dbg] [Application.cc:176] Qt using OpenGL graphics interface
[GUI] [Dbg] [Application.cc:663] Create main window
[GUI] [Dbg] [PathManager.cc:68] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[Msg] Serving entity system service on [/entity/system/add]
[GUI] [Dbg] [PathManager.cc:57] Received resource paths.
[GUI] [Dbg] [Gui.cc:401] Requesting GUI from [/world/default/gui/info]...
[GUI] [Dbg] [GuiRunner.cc:149] Requesting initial state from [/world/default/state]...
[GUI] [Msg] Loading config [/home/acroci/.gz/sim/8/gui.config]
[GUI] [Dbg] [Application.cc:534] Loading plugin [MinimalScene]
[GUI] [Dbg] [MinimalScene.cc:800] Creating gz-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:800] Creating gz-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:982] Creating render thread interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:800] Creating gz-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:982] Creating render thread interface for OpenGL
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [/home/acroci/mambaforge/envs/walking/lib/gz-gui-8/plugins/libMinimalScene.so]
[GUI] [Dbg] [Application.cc:534] Loading plugin [EntityContextMenuPlugin]
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:905] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Msg] Added plugin [Entity Context Menu] to main window
[GUI] [Msg] Loaded plugin [EntityContextMenuPlugin] from path [/home/acroci/mambaforge/envs/walking/lib/gz-sim-8/plugins/gui/libEntityContextMenuPlugin.so]
[GUI] [Dbg] [Application.cc:534] Loading plugin [GzSceneManager]
[... cut ...]
[Dbg] [CameraSensor.cc:786] Camera info for [ergocub_fixed::ergoCub::realsense::realsense_head_rgb] advertised on [/world/tutorial_controlboard/model/ergocub_fixed/model/ergoCub/link/realsense/sensor/realsense_head_rgb/camera_info]
[Wrn] [Ogre2Camera.cc:421] Ogre2Camera::SetVisibilityMask: Mask bits c0000000 are set but will be ignored as they conflict with the reserved bits used internally by the ogre2 backend.
[Wrn] [RenderPassSystem.cc:54] RenderPass of typeid 'N2gz9rendering2v814DistortionPassE' is not registered
[Wrn] [ImageBrownDistortionModel.cc:112] ImageBrownDistortionModel is not supported in ogre2
````

</details>

traversaro, 2024-03-22:
> mong other the conda environment will install Gazebo. In order to work properly it needs some low-level dependencies already installed on Ubuntu through apt:
> 
>     * `libopengl0`
> 
>     * `libEGL.so.1`
> 
>     * `libegl1`
> 
>     * `libgl1-mesa-glx`

Just a copy&paste friendly command:

~~~
sudo apt install libgl1-mesa-glx libegl1 libopengl0
~~~

xela-95, 2024-03-22:
Side note: I find really cool to be able to use vscode from windows attached natively to WSL! See https://code.visualstudio.com/docs/remote/wsl

xela-95, 2024-03-27:
# Running Gazebo on a VM

Little off-topic (I didn't want to open explicitly an issue for this): up to now I'm stuck with the solution of using WSL to test ergoCub walking since when I launch the world containing it, it is not able to find the shared libraries of the plugins, even if the `GZ_SIM_SYSTEM_PLUGIN_PATH` is set up.

I've then tried to setup a VM (using VirtualBox) to avoid continuously restarting my PC to switch between Windows and Linux os. After having set up all the stuff needed (including installing the graphics packages with https://github.com/robotology/gz-sim-yarp-plugins/issues/133#issuecomment-2015144181 ), when I started Gazebo I saw a black area in place of the rendered world, with no errors regarding it:
[image]

For the moment I was able to make it work by disabling the HW acceleration from the VirtualBox settings:
[image]

Unfortunately, this caused the simulation to be quite slow.

I've seen there are `gz-sim` issues related to this that I've not read yet:
- https://github.com/gazebosim/gz-sim/issues/1492
- https://github.com/gazebosim/gz-sim/issues/1116

xela-95, 2024-03-27:
> In the end when configuring WSL I made a mistake in setting the `GZ_SIM_SYSTEM_PLUGIN_PATH ` environment variable, and fixing this I was able to make gazebo start and load ergoCub.
> 
> When I try to run the [walking controller tutorial](https://github.com/robotology/walking-controllers/tree/v0.8.0?tab=readme-ov-file#computer-how-to-run-the-simulation) using v0.8.0) I get an error at step 7:
> 
> ```
> [INFO] Loading custom PIDs
> [INFO] Parsing DEFAULT PID group.
> [INFO] DEFAULT PID successfully loaded
> [INFO] |yarp.os.Port|/walking-coordinator/logger/data:o| Port /walking-coordinator/logger/data:o active at tcp://172.21.28.41:10110/
> [INFO] [WalkingModule::configure] Ready to play! Please prepare the robot.
> Exception of type: DYNAMIC_LIBRARY_FAILURE in file "../../src/Common/IpLibraryLoader.cpp" at line 67:
>  Exception message: libhsl.so: cannot open shared object file: No such file or directory
> [ERROR] WalkingIK: Failed in finding a solution.
> [ERROR] [WalkingModule::prepareRobot] Inverse Kinematics failed while computing the initial position.
> ```

The issue was due to missing HSL solvers.