robotology/gz-sim-yarp-plugins #142: Document how to use gz-sim-yarp-plugins with ergoCub

xela-95, 2024-04-05:


xela-95, 2024-04-05:
I created a wiki page at https://github.com/robotology/gz-sim-yarp-plugins/wiki/How-to-use-%60gz%E2%80%90sim%E2%80%90yarp%E2%80%90plugins%60-with-ergoCub to document the workflow.

@traversaro let me know if for you is clear enough or there's something missing. I will ask to some lab colleague to try to follow it and get feedbacks.

traversaro, 2024-04-05:
> @traversaro let me know if for you is clear enough or there's something missing. I will ask to some lab colleague to try to follow it and get feedbacks.

Ok! For the time being I will remove the conda part until the package is actually available, otherwise it risks being confusing. When you are ready I would make an announcement in the internal AMI chat (for now) to start collecting users, thanks!

traversaro, 2024-04-05:
Not sure if this is the right place, but we need also to sort out where to host this docs. As it is robot-specific, I would avoid to commit in gz-sim-yarp-plugins, but on the other hand being simulator specific @Nicogene wanted to avoid to have it in ergocub-software . Indeed in the current form the world file is quite massive for a minimal example. Let's try to iterate a bit, and then we decide. 

One thing that I think we can easily do in ergocub-software is to have a table like https://github.com/robotology/icub-models?tab=readme-ov-file#model-details, and then probably have somewhere something like https://github.com/robotology/icub-models?tab=readme-ov-file#usage section that refers to how the model can be used. I think that this point the `Classic Gazebo` section can easily be modified to also refer to Modern Gazebo (gz-sim), as the modifications required in the `world.` are the same.

xela-95, 2024-04-05:
> Indeed in the current form the world file is quite massive for a minimal example. Let's try to iterate a bit, and then we decide.

Ok I will try to reduce its size.

traversaro, 2024-04-05:
> > Indeed in the current form the world file is quite massive for a minimal example. Let's try to iterate a bit, and then we decide.
> 
> Ok I will try to reduce its size.

No need to, sorry about the misunderstanding. I meant that we can start using it, and then look how it goes. The tutorial as it is is perfect, let's start promoting it!

xela-95, 2024-04-05:
> No need to, sorry about the misunderstanding. I meant that we can start using it, and then look how it goes. The tutorial as it is is perfect, let's start promoting it!

Ah ok sorry for the misunderstanding! 🚀

xela-95, 2024-04-10:
Now the conda-forge package of `gz-sim-yarp-plugins` is available, but it still cannot be used with the `ergocub-software` package since the latter has not bumper a version in which the support for Modern Gazebo is available. Maybe it's better to maintain this issue in review until this happens and the documentation can be updated.

traversaro, 2024-04-10:
> Now the conda-forge package of `gz-sim-yarp-plugins` is available, but it still cannot be used with the `ergocub-software` package since the latter has not bumper a version in which the support for [...]

@Nicogene can we do a release of ergocub-software so we can get advertise users of the gz-sim + ergocub models simulation using just conda packages? Thanks!

Nicogene, 2024-04-10:
Here it is:
- https://github.com/icub-tech-iit/ergocub-software/releases/tag/v0.7.4

traversaro, 2024-04-10:
Thanks, now I need to fix https://github.com/robotology/robotology-superbuild/issues/1634 . :D , or we just migrate ergocub-software to conda-forge as well.

closed 2024-05-09 by xela-95