robotology/gz-sim-yarp-plugins #144: Implement `IVelocityControl` control mode in controlboard plugin

xela-95, 2024-04-05:
See https://www.yarp.it/latest/classyarp_1_1dev_1_1IVelocityControl.html

traversaro, 2024-04-05:
I guess no active use case from AMI is doing this. However this is necessary to properly simulate iCub's eyes and the iCub's Gaze controller (https://robotology.github.io/robotology-documentation/doc/html/icub_gaze_interface.html), that uses the so-called position-velocity mixed mode, see the `VOCAB_CM_MIXED` mode in https://github.com/robotology/icub-main/blob/4595c48eb71a0ce64d61ea309bea329fb0c8bff7/src/modules/iKinGazeCtrl/src/controller.cpp#L576 .

traversaro, 2025-03-05:
> I guess no active use case from AMI is doing this. However this is necessary to properly simulate iCub's eyes and the iCub's Gaze controller [...]

Going through code I noticed that this issue is still unsolved, and this is mostly a problem for iCub's Gaze controller. After coupling support was added both here (https://github.com/robotology/gz-sim-yarp-plugins/pull/221) and in https://github.com/robotology/gz-sim-yarp-plugins/pull/221, I guess this is the only missing piece required to actually run the Gaze controller. fyi @Nicogene @martinaxgloria

Nicogene, 2025-03-05:
> > I guess no active use case from AMI is doing this. However this is necessary to properly simulate iCub's eyes and the iCub's Gaze controller ([robotology.github.io/robotology-documentation/doc/html/icub_gaze_interface.html](https://robotology.github.io/robotology-documentation/doc/html/icub_gaze_interface.html)), that uses the so-called position-velocity mixed mode, see the `VOCAB_CM_MIXED` mode in [robotology/icub-main@`4595c48`/src/modules/iKinGazeCtrl/src/controller.cpp#L576](https://github.com/robotology/icub-main/blob/4595c48eb71a0ce64d61ea309bea329fb0c8bff7/src/modules/iKinGazeCtrl/src/controller.cpp#L576) .
> 
> Going through code I noticed that this issue is still unsolved, and this is mostly a problem for iCub's Gaze controller. After coupling support was added both here ([#221](https://github.com/robotology/gz-sim-yarp-plugins/pull/221)) and in [#221](https://github.com/robotology/gz-sim-yarp-plugins/pull/221), I guess this is the only missing piece required to actually run the Gaze controller. fyi [@Nicogene](https://github.com/Nicogene) [@martinaxgloria](https://github.com/martinaxgloria)

ccing @pattacini

pattacini, 2025-03-05:
I think we could address it with low priority.
Just let's make sure we won't forget about it.