robotology/gz-sim-yarp-plugins #146: Document how to use each plugin

xela-95, 2024-04-05:
To facilitate the adoption of of `gz-sim-yarp-plugins` is useful to provide documentation on how to use and configure the plugins.

Hence, we want to add to each of them a brief README page documenting:
- how to add the plugin in a model;
- how to configure the plugin via `yarpConfigurationString` or `yarpConfigurationFile`, explaining what the required and optional parameters are;
- refer to the tutorials using it;

traversaro, 2025-02-07:
In https://github.com/robotology/gz-sim-yarp-plugins/pull/232 I add a `README` for the new `gzyarp::ConfigurationOverride` plugin, I think we can use it as an example of the `README.md` to add to each plugin folder.

xela-95, 2025-02-07:
Yes that is perfect to me! I will try to add the README for the other as soon as I have time!