robotology/gz-sim-yarp-plugins #149: Write the code that extracts a list of devices from a given simulation instance from gzyarp::Handler and pass it to BipedalLocomotion::RobotInterface::YarpRobotControl::setDriver() and BipedalLocomotion::RobotInterface::YarpSensorBridge::setDriversList

traversaro, 2024-04-05:
To do so, we can build on top of the test in https://github.com/robotology/gz-sim-yarp-plugins/issues/147 and add some similar tests that instead uses `BipedalLocomotion::RobotInterface::YarpRobotControl` and `BipedalLocomotion::RobotInterface::YarpSensorBridge`.

However, we can't add a dependency on blf here. So we need to understand if this code can go in blf, or in a third package that depends on both blf and gz-sim . In both cases, we need to install the headers corresponding to the  gzyarp::Handler , so that downstream projects can use it.

xela-95, 2024-04-19:
> However, we can't add a dependency on blf here. So we need to understand if this code can go in blf, or in a third package that depends on both blf and gz-sim . In both cases, we need to install the headers corresponding to the gzyarp::Handler , so that downstream projects can use it.

Now that the blocking issue #150  has been closed, and with #147 almost finished, I'm starting to look at this issue. Where can I put the code? @traversaro

traversaro, 2024-04-19:
Actually I forgot to update the issue. We discussed about this with @GiulioRomualdi , and given that blf methods of interest takes (or can take) in input a PolyDriverList, probably we can just ad a function/method in the public interface of gz-sim-yarp-plugins

xela-95, 2024-04-23:
Useful pointers:
- https://ami-iit.github.io/bipedal-locomotion-framework/classBipedalLocomotion_1_1RobotInterface_1_1YarpRobotControl.html
- https://ami-iit.github.io/bipedal-locomotion-framework/classBipedalLocomotion_1_1RobotInterface_1_1YarpSensorBridge.html
- https://github.com/ami-iit/paper_romualdi_viceconte_2024_icra_dnn-mpc-walking/blob/docker_mann_async_mpc_closed_loop_without_neck/src/centroidal-mpc-walking/config/robots/ergoCubSN001/dnn-mpc/links_with_imu.ini
- https://github.com/ami-iit/paper_romualdi_viceconte_2024_icra_dnn-mpc-walking/blob/docker_mann_async_mpc_closed_loop_without_neck/src/centroidal-mpc-walking/config/robots/ergoCubGazeboV1_1/dnn-mpc/sensor_bridge.ini

xela-95, 2024-04-26:
> Actually I forgot to update the issue. We discussed about this with @GiulioRomualdi , and given that blf methods of interest takes (or can take) in input a PolyDriverList, probably we can just ad a function/method in the public interface of gz-sim-yarp-plugins, no need to have a new repo or add dependencies between blf and this repo

@traversaro so if I understood it correctly, I should implement these methods in the public interface of gz-sim-yarp-plugins (with no additional dependency), but then I should have a separate project where I can call the new interface methods of gz-sim-yarp-plugins but also the `BipedalLocomotion::RobotInterface::YarpRobotControl`/`BipedalLocomotion::RobotInterface::YarpSensorBridge` ones?

traversaro, 2024-04-27:
> @traversaro so if I understood it correctly, I should implement these methods in the public interface of gz-sim-yarp-plugins (with no additional dependency), but then I should have a separate [...]

Yes, I think that can be just an example on how to use BLF + gz-sim, the code will me quite limited so no need to have a dedicated project for it.

xela-95, 2024-05-03:
I created a temporary repository to host the example at https://github.com/xela-95/blf-gz-sim-example

When it works I will put it into a more meaningful place, like an example folder of gz-sim-yarp-plugins.