robotology/gz-sim-yarp-plugins #150: Install headers and CMake config files to  gzyarp::Handler so that it can be used by downstream projects

traversaro, 2024-04-05:


traversaro, 2024-04-05:
This is useful for https://github.com/robotology/gz-sim-yarp-plugins/issues/149 but also to permit to migrate plugins such as https://github.com/ami-iit/ironcub-mk1-software/tree/d002a12528dcae47dbd6fe61e34d7c16ef5368d8/lib/gazebo/plugins/jets-plugin to gz-sim-yarp-plugins, in a way that they can add devics to the gzyarp::Handle so that the robotinterface xml can find devices created by third-party gz-sim plugins that are not contained in gz-sim-yarp-plugins.

xela-95, 2024-04-08:
@traversaro can you give me some examples of projects in which this has already been done?

To make the headers of the library available to downstream projects, is it sufficient to have the `target_include_directories` CMake command with the `PUBLIC` attribute, like in:
https://github.com/robotology/gz-sim-yarp-plugins/blob/f08dd84cfa188d1971abcf9912ff09241f3fc1c0/libraries/singleton-devices/CMakeLists.txt#L2
?

What about the CMake config files?

traversaro, 2024-04-08:
> @traversaro can you give me some examples of projects in which this has already been done?
>
> To make the headers of the library available to downstream projects, is it sufficient to have the [...]

You can use https://github.com/robotology/how-to-export-cpp-library/blob/master/CMakeLists.txt as an example. In particular, you need to add a call to the `install_basic_package_files` function, see https://github.com/robotology/how-to-export-cpp-library/blob/master/CMakeLists.txt#L140-L145 .

traversaro, 2024-04-08:
Note that you need to modify the `target_include_directories` to also add to the target the directory where the include will be installed, via the `$<INSTALL_INTERFACE`, see https://github.com/robotology/how-to-export-cpp-library/blob/afb21efb655e7b1cc11636c3d42aad9e02fb626f/src/LibTemplateCMake/CMakeLists.txt#L74 or https://github.com/robotology/how-to-export-cpp-library/blob/afb21efb655e7b1cc11636c3d42aad9e02fb626f/src/LibHeaderOnlyTemplateCMake/CMakeLists.txt#L47 .

PR #160 (2024-04-12): Install headers and CMake config files to gzyarp::DeviceRegistry so that it can be used by downstream projects
Closes #150
closed 2024-04-12 by xela-95