robotology/gz-sim-yarp-plugins #158: Do new release (April 2024)

traversaro, 2024-04-10:
This is useful to include https://github.com/robotology/gz-sim-yarp-plugins/pull/141 in a released version. cc @xela-95

xela-95, 2024-04-10:
Concerning the conda-forge package, do I just have to update the recipe `meta.yaml` with the new package version in a fork of the feedstock repo and merge it through a PR, is it correct @traversaro?

https://github.com/conda-forge/gz-sim-yarp-plugins-feedstock/blob/6ebe595692fe19858c784e08fabe85a4badbaac5/recipe/meta.yaml#L4

xela-95, 2024-04-10:
Release v0.2.0 done: https://github.com/robotology/gz-sim-yarp-plugins/releases/tag/v0.2.0

traversaro, 2024-04-10:
> Concerning the conda-forge package, do I just have to update the recipe `meta.yaml` with the new package version in a fork of the feedstock repo and merge it through a PR, is it correct [...]

This is done automatically by the bot, no need to do this manually. The bot will take some hours to do that, if you want to speed up the process, you can ask the bot to do the version update as soon as possible with the `@conda-forge-admin please update version` command bot, see for example https://github.com/conda-forge/cmake-package-check-feedstock/issues/5 and https://conda-forge.org/docs/maintainer/infrastructure/#conda-forge-admin-please-update-version .

xela-95, 2024-04-10:
conda-forge package also updated: https://anaconda.org/conda-forge/gz-sim-yarp-plugins

Closing.

closed 2024-04-10 by xela-95