robotology/gz-sim-yarp-plugins #163: Ensure that enable_tags and disable_tags options can be passed in a world SDF file to the gazebo_yarp_robotinterface of an included model

traversaro, 2024-04-22:
Similar to https://github.com/robotology/gazebo-yarp-plugins/issues/672, but for gz-sim-yarp-plugins .

As we do not have any `gazebo_yarp_configurationoverride`-like plugin in `gz-sim-yarp-plugins`, probably we first need to solve this before this one, see https://github.com/robotology/gz-sim-yarp-plugins/issues/159 that is kind of related (even if focusing just on initial joint positions).

xela-95, 2024-04-22:
@traversaro do you mean that having a `gazebo_yarp_configurationoverride`-like plugin is prioritary with respect to this? Do I have understood correctly?

xela-95, 2024-04-22:
Ok clear. After an initial investigation I will report what I've found and then we can decide how to proceed.

traversaro, 2025-02-17:
A fix for this issue was proposed in https://github.com/robotology/gz-sim-yarp-plugins/pull/238 .

closed 2025-02-19 by traversaro