robotology/gz-sim-yarp-plugins #170: downgrading ipopt when installing through conda.

LoreMoretti, 2024-07-31:
I have followed the recommended installation procedure, i.e.:

```
conda install -c conda-forge gz-sim-yarp-plugins

```

To complete the installation, I am asked to downgrade some libraries:

``` diff

  Downgrade:
─────────────────────────────────────────────────────────────────────────────────────────────────────

  - sqlite                                    3.45.1  h2c6b66d_0             conda-forge     Cached
  + sqlite                                    3.32.3  hcee41ef_1             conda-forge        1MB
  - ipopt                                    3.14.14  h04b96a2_1             conda-forge     Cached
  + ipopt                                     3.13.3  h559d178_1             conda-forge        2MB
  - gst-plugins-good                          1.22.7  h16e77d8_0             conda-forge     Cached
  + gst-plugins-good                          1.18.5  h0661c57_2             conda-forge        2MB

```

among which there is `ipopt`.

Is there a way to prevent `ipopt` to be downgraded?

LoreMoretti, 2024-07-31:
Just to be precise, I am using `mamba`

traversaro, 2024-07-31:
So happy to see you installing this. : )

So, I tried to install just gz-sim-yarp-plguins and ipopt 3.14.14, and it is working fine:

~~~
traversaro@IITBMP014LW012:~$ conda create -n test gz-sim-yarp-plugins ipopt==3.14.14
WARNING: A conda environment already exists at '/home/traversaro/miniforge3/envs/test'
Remove existing environment (y/[n])? y

Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
    current version: 24.5.0
    latest version: 24.7.1

Please update conda by running

    $ conda update -n base -c conda-forge conda

## Package Plan ##

  environment location: /home/traversaro/miniforge3/envs/test

  added / updated specs:
    - gz-sim-yarp-plugins
    - ipopt==3.14.14

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    aws-c-auth-0.7.22          |      hbd3ac97_10         104 KB  conda-forge
    aws-c-cal-0.7.1            |       h87b94db_1          46 KB  conda-forge
    aws-c-event-stream-0.4.2   |      h7671281_15          53 KB  conda-forge
    aws-c-http-0.8.2           |       he17ee6b_6         190 KB  conda-forge
    aws-c-io-0.14.10           |       h826b7d6_1         154 KB  conda-forge
    aws-c-mqtt-0.10.4          |       hcd6a914_8         160 KB  conda-forge
    aws-c-s3-0.6.0             |       h365ddd8_2         108 KB  conda-forge
    aws-crt-cpp-0.27.3         |       hda66527_2         337 KB  conda-forge
    aws-sdk-cpp-1.11.329       |       h46c3b66_9         3.5 MB  conda-forge
    azure-core-cpp-1.13.0      |       h935415a_0         330 KB  conda-forge
    azure-identity-cpp-1.8.0   |       hd126650_2         195 KB  conda-forge
    azure-storage-blobs-cpp-12.12.0|       hd2e3451_0         511 KB  conda-forge
    azure-storage-common-cpp-12.7.0|       h10ac4d7_1         140 KB  conda-forge
    dartsim-6.14.2             |       hdf3b901_0        13.4 MB  conda-forge
    geos-3.12.2                |       he02047a_1         1.7 MB  conda-forge
    geotiff-1.7.3              |       hf7fa9e8_2         129 KB  conda-forge
    glib-2.80.3                |       h8a4344b_1         585 KB  conda-forge
    glib-tools-2.80.3          |       h73ef956_1         111 KB  conda-forge
    gz-sim-yarp-plugins-0.2.0  |       ha770c72_2           8 KB  conda-forge
    ipopt-3.14.14              |       h04b96a2_1         997 KB  conda-forge
    json-c-0.17                |       h1220068_1          82 KB  conda-forge
    libclang13-18.1.8          |default_h9def88c_1        10.5 MB  conda-forge
    libcurl-8.9.0              |       hdb1bdb2_0         406 KB  conda-forge
    libgcrypt-1.11.0           |       h4ab18f5_1         668 KB  conda-forge
    libgdal-3.9.1              |      ha770c72_10         411 KB  conda-forge
    libgdal-core-3.9.1         |      h8f9377d_10         9.8 MB  conda-forge
    libgdal-fits-3.9.1         |      hdd6600c_10         464 KB  conda-forge
    libgdal-grib-3.9.1         |      h5f34788_10         704 KB  conda-forge
    libgdal-hdf4-3.9.1         |      ha39a594_10         562 KB  conda-forge
    libgdal-hdf5-3.9.1         |      ha2ed5f0_10         623 KB  conda-forge
    libgdal-jp2openjpeg-3.9.1  |      h2ebfdf0_10         456 KB  conda-forge
    libgdal-kea-3.9.1          |      h2b45729_10         468 KB  conda-forge
    libgdal-netcdf-3.9.1       |      h94e7027_10         713 KB  conda-forge
    libgdal-pdf-3.9.1          |      h562c687_10         647 KB  conda-forge
    libgdal-pg-3.9.1           |      he047751_10         512 KB  conda-forge
[... 345 lines cut ...]
  xorg-libxi         conda-forge/linux-64::xorg-libxi-1.7.10-h4bc722e_1
  xorg-libxmu        conda-forge/linux-64::xorg-libxmu-1.1.3-h4ab18f5_1
  xorg-libxpm        conda-forge/linux-64::xorg-libxpm-3.5.17-hd590300_0
  xorg-libxrandr     conda-forge/linux-64::xorg-libxrandr-1.5.2-h7f98852_1
  xorg-libxrender    conda-forge/linux-64::xorg-libxrender-0.9.11-hd590300_0
  xorg-libxt         conda-forge/linux-64::xorg-libxt-1.3.0-hd590300_1
  xorg-randrproto    conda-forge/linux-64::xorg-randrproto-1.5.0-h7f98852_1001
  xorg-renderproto   conda-forge/linux-64::xorg-renderproto-0.11.1-h7f98852_1002
  xorg-xextproto     conda-forge/linux-64::xorg-xextproto-7.3.0-h0b41bf4_1003
  xorg-xf86vidmodep~ conda-forge/linux-64::xorg-xf86vidmodeproto-2.3.1-h7f98852_1002
  xorg-xproto        conda-forge/linux-64::xorg-xproto-7.0.31-h7f98852_1007
  xz                 conda-forge/linux-64::xz-5.2.6-h166bdaf_0
  yaml               conda-forge/linux-64::yaml-0.2.5-h7f98852_2
  ycm-cmake-modules  conda-forge/linux-64::ycm-cmake-modules-0.16.4-hac33072_0
  zeromq             conda-forge/linux-64::zeromq-4.3.5-h75354e8_4
  zlib               conda-forge/linux-64::zlib-1.3.1-h4ab18f5_1
  zstd               conda-forge/linux-64::zstd-1.5.6-ha6fb4c9_0
  zziplib            conda-forge/linux-64::zziplib-0.13.69-he45264a_2

Proceed ([y]/n)? n

CondaSystemExit: Exiting.
~~~

Can you share the environment in which you are trying install `gz-sim-yarp-plugins` ?

LoreMoretti, 2024-07-31:
Sure, here the output of the `mamba list`:

```

mamba list
# packages in environment at /home/lomoretti/mambaforge/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_14    conda-forge
ace                       7.1.3                h59595ed_2    conda-forge
aiohttp                   3.9.3           py312h98912ed_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.7.1                h59595ed_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
bison                     3.8.2                h59595ed_0    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.82.0               h8da182e_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.27.0               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.7.4             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
catch2                    3.5.2                h00ab1b0_0    conda-forge
catkin-pkg                1.0.0                    pypi_0    pypi
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.0                hbdc6101_0    conda-forge
charset-normalizer        3.3.2                    pypi_0    pypi
cmake                     3.28.3               hcfe8598_0    conda-forge
coinhsl                   2019.05.21.1         h11d5e8e_0    <unknown>
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.0           py312h8572e83_0    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.1               h7abe67e_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
docutils                  0.21.2                   pypi_0    pypi
double-conversion         3.3.0                h59595ed_0    conda-forge
doxygen                   1.10.0               h661eb56_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
expat-cos7-x86_64         2.1.0             h9b0a68f_1105    conda-forge
fcl                       0.7.0                hadc09e8_4    conda-forge
ffmpeg                    6.1.1           gpl_h186bccc_100    conda-forge
filelock                  3.15.4                   pypi_0    pypi
flann                     1.9.2                h2b5ea80_0    conda-forge
[... 375 lines cut ...]
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-libxtst              1.2.3             h7f98852_1002    conda-forge
xorg-libxxf86vm           1.1.5                h0b41bf4_0    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.4           py312h98912ed_0    conda-forge
zeromq                    4.3.5                h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

```

traversaro, 2024-07-31:
You can also try yourself to run `conda install -c conda-forge gz-sim-yarp-plugins ipopt==3.14.14` to understand if it complains.

LoreMoretti, 2024-07-31:
> Interesting, I have no idea what is happening, can you share the list with conda list --explicit so that I can easily re-create it on my machine?

Here:

```

# This file may be used to create an environment using:
# $ conda create --name <env> --file <this file>
# platform: linux-64
@EXPLICIT
https://conda.anaconda.org/conda-forge/linux-64/bash-completion-2.11-ha770c72_1.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/_libgcc_mutex-0.1-conda_forge.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/_sysroot_linux-64_curr_repodata_hack-3-h69a702a_14.conda
https://conda.anaconda.org/conda-forge/linux-64/ca-certificates-2024.7.4-hbcca054_0.conda
https://conda.anaconda.org/conda-forge/noarch/font-ttf-dejavu-sans-mono-2.37-hab24e00_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/font-ttf-inconsolata-3.000-h77eed37_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/font-ttf-source-code-pro-2.038-h77eed37_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/font-ttf-ubuntu-0.83-h77eed37_1.conda
https://conda.anaconda.org/conda-forge/linux-64/ld_impl_linux-64-2.40-h41732ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/libboost-headers-1.82.0-ha770c72_6.conda
https://conda.anaconda.org/conda-forge/noarch/libgcc-devel_linux-64-12.3.0-h8bca6fd_105.conda
https://conda.anaconda.org/conda-forge/noarch/libstdcxx-devel_linux-64-12.3.0-h8bca6fd_105.conda
https://conda.anaconda.org/conda-forge/linux-64/libstdcxx-ng-13.2.0-h7e041cc_5.conda
https://conda.anaconda.org/conda-forge/linux-64/mumps-include-5.6.2-ha770c72_4.conda
https://conda.anaconda.org/conda-forge/linux-64/nlohmann_json-3.11.2-h27087fc_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/poppler-data-0.4.12-hd8ed1ab_0.conda
https://conda.anaconda.org/conda-forge/noarch/pybind11-abi-4-hd8ed1ab_3.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/python_abi-3.12-4_cp312.conda
https://conda.anaconda.org/conda-forge/noarch/tzdata-2024a-h0c530f3_0.conda
https://conda.anaconda.org/conda-forge/linux-64/utfcpp-4.0.5-ha770c72_0.conda
https://conda.anaconda.org/conda-forge/noarch/fonts-conda-forge-1-0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/kernel-headers_linux-64-3.10.0-h4a8ded7_14.conda
https://conda.anaconda.org/conda-forge/linux-64/libgomp-13.2.0-h807b86a_5.conda
https://conda.anaconda.org/conda-forge/linux-64/_openmp_mutex-4.5-2_gnu.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/fonts-conda-ecosystem-1-0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/sysroot_linux-64-2.17-h4a8ded7_14.conda
https://conda.anaconda.org/conda-forge/linux-64/binutils_impl_linux-64-2.40-hf600244_0.conda
https://conda.anaconda.org/conda-forge/noarch/expat-cos7-x86_64-2.1.0-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libdrm-cos7-x86_64-2.4.97-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/libgcc-ng-13.2.0-h807b86a_5.conda
https://conda.anaconda.org/conda-forge/noarch/libglvnd-cos7-x86_64-1.0.1-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libsepol-cos7-x86_64-2.5-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libx11-common-cos7-x86_64-1.6.7-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libxau-cos7-x86_64-1.0.8-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libxcb-cos7-x86_64-1.13-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libxdamage-cos7-x86_64-1.1.4-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libxshmfence-cos7-x86_64-1.2-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/mesa-khr-devel-cos7-x86_64-18.3.4-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/mesa-libglapi-cos7-x86_64-18.3.4-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/ace-7.1.3-h59595ed_2.conda
https://conda.anaconda.org/conda-forge/linux-64/alsa-lib-1.2.10-hd590300_0.conda
https://conda.anaconda.org/conda-forge/linux-64/aom-3.7.1-h59595ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/asio-1.29.0-h59595ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/attr-2.5.1-h166bdaf_1.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/binutils-2.40-hdd6e379_0.conda
https://conda.anaconda.org/conda-forge/linux-64/binutils_linux-64-2.40-hbdbef99_2.conda
https://conda.anaconda.org/conda-forge/linux-64/bzip2-1.0.8-hd590300_5.conda
https://conda.anaconda.org/conda-forge/linux-64/c-ares-1.27.0-hd590300_0.conda
https://conda.anaconda.org/conda-forge/linux-64/catch2-3.5.2-h00ab1b0_0.conda
https://conda.anaconda.org/conda-forge/linux-64/console_bridge-1.0.2-h924138e_1.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/dav1d-1.2.1-hd590300_0.conda
https://conda.anaconda.org/conda-forge/linux-64/double-conversion-3.3.0-h59595ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/eigen-3.4.0-h00ab1b0_0.conda
https://conda.anaconda.org/conda-forge/linux-64/fmt-10.2.1-h00ab1b0_0.conda
https://conda.anaconda.org/conda-forge/linux-64/fribidi-1.0.10-h36c2ea0_0.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/geos-3.12.0-h59595ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/gettext-0.21.1-h27087fc_0.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/giflib-5.2.1-h0b41bf4_3.conda
https://conda.anaconda.org/conda-forge/linux-64/glm-0.9.9.8-h00ab1b0_0.conda
https://conda.anaconda.org/conda-forge/linux-64/gmp-6.3.0-h59595ed_0.conda
[... 366 lines cut ...]
[... cut ...]
https://conda.anaconda.org/conda-forge/linux-64/pyqtwebengine-5.15.9-py312hbd9a932_5.conda
https://conda.anaconda.org/conda-forge/linux-64/vtk-io-ffmpeg-9.2.6-qt_py312h1234567_219.conda
https://conda.anaconda.org/conda-forge/linux-64/gazebo-11.14.0-h4154485_3.conda
https://conda.anaconda.org/conda-forge/linux-64/opencv-4.8.1-py312h74ac831_5.conda
https://conda.anaconda.org/conda-forge/linux-64/vtk-9.2.6-qt_py312h1234567_219.conda
https://conda.anaconda.org/conda-forge/linux-64/pcl-1.14.0-h4836831_0.conda
```

LoreMoretti, 2024-07-31:
> You can also try yourself to run conda install -c conda-forge gz-sim-yarp-plugins ipopt==3.14.14 to understand if it complains.

It complains with the following message:

<details>
  <summary>click to expland</summary>

  ```

(robsub) lomoretti@IITICUBLAP226:~/robotology/robotology-superbuild (master)$ mamba install -c conda-forge gz-sim-yarp-plugins ipopt==3.14.14

Looking for: ['gz-sim-yarp-plugins', 'ipopt==3.14.14']

conda-forge/noarch                                  15.8MB @   5.8MB/s  2.8s
conda-forge/linux-64                                36.6MB @   4.7MB/s  7.8s

Pinned packages:
  - python 3.12.*

warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
Could not solve for environment specs
The following packages are incompatible
├─ dartsim is installable with the potential options
│  ├─ dartsim [6.10.1|6.11.0|...|6.13.1] would require
│  │  └─ tinyxml2 >=9.0.0,<10.0a0 , which can be installed;
│  ├─ dartsim [6.10.0|6.10.1|6.9.4|6.9.5] would require
│  │  └─ tinyxml2 >=8.0.0,<8.1.0a0 , which can be installed;
│  ├─ dartsim [6.12.2|6.13.1] would require
│  │  └─ flann >=1.9.2,<1.9.3.0a0 , which can be installed;
│  ├─ dartsim 6.12.2 would require
│  │  ├─ flann >=1.9.2,<1.9.3.0a0 , which can be installed;
│  │  └─ urdfdom >=3.1.1,<3.2.0a0 , which can be installed;
│  ├─ dartsim [6.12.2|6.14.0|6.14.1|6.14.2] would require
│  │  ├─ assimp >=5.4.1,<5.4.2.0a0 , which requires
│  │  │  └─ libboost >=1.84.0,<1.85.0a0  with the potential options
│  │  │     ├─ libboost 1.84.0, which can be installed;
│  │  │     └─ libboost 1.84.0 would require
│  │  │        ├─ icu >=75.1,<76.0a0 , which can be installed;
│  │  │        └─ libzlib >=1.3.1,<2.0a0 , which can be installed;
│  │  └─ libboost >=1.84.0,<1.85.0a0  with the potential options
│  │     ├─ libboost 1.84.0, which can be installed;
│  │     └─ libboost 1.84.0, which can be installed (as previously explained);
│  ├─ dartsim 6.13.2 would require
│  │  ├─ assimp >=5.3.1,<5.3.2.0a0  but there are no viable options
│  │  │  ├─ assimp 5.3.1 would require
│  │  │  │  ├─ libboost >=1.82.0,<1.83.0a0  but there are no viable options
│  │  │  │  │  ├─ libboost 1.82.0 would require
│  │  │  │  │  │  └─ boost-cpp 1.82.0* , which can be installed;
│  │  │  │  │  ├─ libboost 1.82.0 would require
│  │  │  │  │  │  └─ icu >=72.1,<73.0a0 , which can be installed;
│  │  │  │  │  └─ libboost 1.82.0 conflicts with any installable versions previously reported;
│  │  │  │  └─ libzlib >=1.2.13,<1.3.0a0 , which conflicts with any installable versions previously reported;
│  │  │  └─ assimp 5.3.1 conflicts with any installable versions previously reported;
│  │  └─ flann >=1.9.2,<1.9.3.0a0 , which can be installed;
│  ├─ dartsim 6.13.2 would require
│  │  ├─ assimp >=5.4.1,<5.4.2.0a0 , which can be installed (as previously explained);
│  │  └─ flann >=1.9.2,<1.9.3.0a0 , which can be installed;
│  └─ dartsim [6.14.2|6.14.4] would require
│     └─ ipopt >=3.14.16,<3.14.17.0a0 , which can be installed;
├─ expat is installable with the potential options
│  ├─ expat [2.5.0|2.6.1], which can be installed;
│  ├─ expat [2.1.0|2.2.1|...|2.4.9], which can be installed;
│  └─ expat 2.6.2 would require
│     └─ libexpat 2.6.2 h59595ed_0, which requires
│        └─ expat 2.6.2.* , which conflicts with any installable versions previously reported;
├─ gazebo is installable with the potential options
│  ├─ gazebo 11.14.0 would require
│  │  ├─ assimp >=5.3.1,<5.3.2.0a0  but there are no viable options
│  │  │  ├─ assimp 5.3.1, which cannot be installed (as previously explained);
│  │  │  └─ assimp 5.3.1 conflicts with any installable versions previously reported;
[... 2237 lines cut ...]
   │  └─ vtk-base [9.3.0 egl_py312h1234567_0|9.3.1 osmesa_py312h1234567_100|9.3.1 osmesa_py312hb25cddc_101], which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg [9.3.0|9.3.1], which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.0 would require
   │  ├─ ffmpeg >=6.1.1,<7.0a0 , which cannot be installed (as previously explained);
   │  └─ vtk-base 9.3.0 qt_py312h1234567_200, which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.1, which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.1 would require
   │  ├─ ffmpeg >=6.1.1,<7.0a0 , which cannot be installed (as previously explained);
   │  └─ vtk-base 9.3.1 egl_py312h16fb7b0_1, which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.1 would require
   │  ├─ ffmpeg >=6.1.1,<7.0a0 , which cannot be installed (as previously explained);
   │  └─ vtk-base [9.3.1 egl_py312hc69c4c9_2|9.3.1 osmesa_py312h5b0cd0c_102], which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.1 would require
   │  ├─ ffmpeg >=7.0.1,<8.0a0 , which cannot be installed (as previously explained);
   │  └─ vtk-base 9.3.1 egl_py312h92a6ba4_2, which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.1 would require
   │  ├─ ffmpeg >=7.0.1,<8.0a0 , which cannot be installed (as previously explained);
   │  └─ vtk-base 9.3.1 osmesa_py312hde7e526_102, which cannot be installed (as previously explained);
   ├─ vtk-io-ffmpeg 9.3.1 would require
   │  ├─ ffmpeg >=6.1.1,<7.0a0 , which cannot be installed (as previously explained);
   │  └─ vtk-base [9.3.1 qt_py312h1234567_200|9.3.1 qt_py312h9816b39_201|9.3.1 qt_py312hf523409_202], which cannot be installed (as previously explained);
   └─ vtk-io-ffmpeg 9.3.1, which cannot be installed (as previously explained).

  ```
</details>

traversaro, 2024-07-31:
Not sure what is the difference w.r.t. to your setup, but in my case this is the output:

~~~
(testlo24) traversaro@IITBMP014LW012:~$ conda install -c conda-forge gz-sim-yarp-plugins
Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package gz-sim-yarp-plugins-0.2.0-ha770c72_0 requires libgz-sim-yarp-plugins 0.2.0 h41785cb_0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ gz-sim-yarp-plugins is installable with the potential options
│  ├─ gz-sim-yarp-plugins [0.1.1|0.2.0] would require
│  │  └─ libgz-sim-yarp-plugins [0.1.1 h41785cb_0|0.2.0 h41785cb_0|0.2.0 h41785cb_1], which requires
│  │     └─ libgz-sim8 >=8.1.0,<9.0a0  with the potential options
│  │        ├─ libgz-sim8 [8.1.0|8.2.0|8.3.0] would require
│  │        │  └─ libprotobuf >=4.25.3,<4.25.4.0a0 , which can be installed;
│  │        └─ libgz-sim8 8.1.0 would require
│  │           └─ tinyxml2 >=10.0.0,<11.0a0 , which can be installed;
│  └─ gz-sim-yarp-plugins 0.2.0 would require
│     └─ libgz-sim-yarp-plugins 0.2.0 h6e67749_2, which requires
│        └─ libprotobuf >=4.25.3,<4.25.4.0a0 , which can be installed;
├─ libprotobuf 4.24.4.*  is not installable because it conflicts with any installable versions previously reported;
└─ tinyxml2 9.0.0.*  is not installable because it conflicts with any installable versions previously reported.
~~~

so there error is more clear, you have some old versions of protobuf and tinyxml2 . My conda version is:
~~~
traversaro@IITBMP014LW012:~$ conda info

     active environment : None
            shell level : 0
       user config file : /home/traversaro/.condarc
 populated config files : /home/traversaro/miniforge3/.condarc
                          /home/traversaro/.condarc
          conda version : 24.5.0
    conda-build version : 24.5.1
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.5.0=0
                          __cuda=12.3=0
                          __glibc=2.39=0
                          __linux=5.15.153.1=0
                          __unix=0=0
       base environment : /home/traversaro/miniforge3  (writable)
      conda av data dir : /home/traversaro/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/traversaro/miniforge3/pkgs
                          /home/traversaro/.conda/pkgs
       envs directories : /home/traversaro/miniforge3/envs
                          /home/traversaro/.conda/envs
               platform : linux-64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.10.14 Linux/5.15.153.1-microsoft-standard-WSL2 ubuntu/24.04 glibc/2.39 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1002:1002
             netrc file : None
           offline mode : False
~~~

LoreMoretti, 2024-07-31:
I was installing `gz-sim-yarp-plugins` in a conda environment which was not empty, since it was containing all the dependencies of the `robotology-superbuild`.

I will proceed by installing `gz-sim-yarp-plugins` from the `robotology-superbuild`.

Closing this issue, thanks @traversaro.

closed 2024-07-31 by LoreMoretti