robotology/gz-sim-yarp-plugins #171: issue with IDLE button in yarpmotorgui

LoreMoretti, 2024-08-01:
I am trying the [single_pendulum](https://github.com/robotology/gz-sim-yarp-plugins/tree/main/tutorial/single_pendulum) tutorial.

When I run the `gz sim model.sdf`, the model gets correctly loaded.

Then, I run `yarpmotorgui`, which succesfully loads the yarpmotorgui, but the defeaul control mode is **position** and not **idle**.

After, I started the simulation in gz by clicking the play button.

So then, I clicked on the **idle** button (the one below **home**):

[image]

and I get the correct behaviour in gz (i.e. the pendulum is swinging), though the yarpmotorgui "freezes" as per the video below:

https://github.com/user-attachments/assets/03d9d842-41dd-4372-8fea-05b0c3387506

and I cannot switch anymore control mode.

Moreover, I see the following error being printed multiple times in the terminal:

```
[ERROR] Error while updating control references
[WARNING] Joint  upper_joint  control mode  1818519910  is currently not implemented
```

@xela-95 pointed out that, if instead of using the **idle** button, one selects the **idle** control mode from the first drop-down menu, everything runs fine.

xela-95, 2024-08-01:
Thanks @LoreMoretti for testing the tutorials of this repo! I will check what is the behavior of the Idle button of the `yarpmotorgui` and come back to you.

xela-95, 2024-08-01:
By looking at the documentation doc http://wiki.icub.org/wiki/File:ICub_Control_Modes_1_1.pdf (I don't know if there is something more up to date than this 2014 document), in Section 3 it says:

[image]

[image]

[image]

Hence, I think that we missed to implement this safety mechanism, that in `gazebo-yarp-plugins` is implemented [here](https://github.com/robotology/gazebo-yarp-plugins/blob/d296652b7619dbef39e9bd65ae49f6ca3bdd9377/plugins/controlboard/src/ControlBoardDriverControlMode.cpp#L115-L129). Now being addressed in #172

xela-95, 2024-08-02:
Now that #172 is merged we can try to verify if this behavior with yarpmotorgui is solved :)

traversaro, 2024-08-02:
> (I don't know if there is something more up to date than this 2014 document)

I think there was an attempt in re-organizing the docs in https://github.com/robotology/yarp/issues/1139, but I am not sure what is the status of that, so I guess the PDF is probably the most updated source of info.

LoreMoretti, 2024-08-02:
> Now that https://github.com/robotology/gz-sim-yarp-plugins/pull/172 is merged we can try to verify if this behavior with yarpmotorgui is solved :)

Seems to be solved:

https://github.com/user-attachments/assets/656dafce-c98a-45fe-97b6-a72b7a7f4531

xela-95, 2024-08-02:
Perfect! We can close this now.

closed 2024-08-02 by xela-95