robotology/gz-sim-yarp-plugins #173: crash when loading ergoCubGazeboV1_1 in gz sim

LoreMoretti, 2024-08-02:
I am trying to load ergoCubGazeboV1_1 in gz_sim, but I am experiencing a crash.

 I have installed **gz-sim-yarpl-plugins** as part of the **robotology-superbuild** (installed from source with conda dependencies).
 

<details>

<summary>robotology git status</summary>

```
 (robsub_gzsim) lomoretti@IITICUBLAP226:~/robotology/robotology-superbuild (master)$ ./scripts/robotologyGitStatus.sh 
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild: master
3 days ago|The KMP_DUPLICATE_LIB_OK problem is present also on Windows so document and apply the workaround also on Windows (#1658)|84f1b1d9207de0856b7f79d5c7795fdba224f2de|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
 M .vscode/settings.json
?? .cache/
diff --git a/.vscode/settings.json b/.vscode/settings.json
index 8c9319f..a95c43a 100644
--- a/.vscode/settings.json
+++ b/.vscode/settings.json
@@ -1,7 +1,8 @@
 {
-    "C_Cpp.default.compileCommands": "${config:cmake.buildDirectory}/compile_commands.json",
-    "cmake.mergedCompileCommands": "${command:cmake.buildDirectory}/compile_commands.json",
+    "C_Cpp.default.compileCommands": "${workspaceFolder}/build/compile_commands.json",
+    "cmake.mergedCompileCommands": "${workspaceFolder}/build/compile_commands.json",
     "clangd.arguments": [
-        "--compile-commands-dir=${config:cmake.buildDirectory}/"
+        "--compile-commands-dir=${workspaceFolder}/build/"
     ],
-}
+    "cmake.copyCompileCommands": "",
+}
\ No newline at end of file
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/BayesFilters/: master
9 months ago|Merge pull request #103 from GiulioRomualdi/patch-1|b180d03df2614eeb7749e130c70391f59d91408c|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/bipedal-locomotion-framework/: master
9 days ago|Add method to get joint limits from control board (#868)|4ea37800355531fe4c78f12a4ed0aa426a748ed3|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/BlockFactory/: master
6 months ago|Bump version to  0.8.5 (#76)|22f62816df31a39b972ac135d186d4150458c3b1|HEAD -> master, tag: v0.8.5, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/casadi/: 3.6.5.backport3724
4 weeks ago|Issue #3723 Fixed function signatures for MX::difference|71d40ac6419b603965a0e0882857a0615b428a1c|HEAD -> 3.6.5.backport3724, origin/3.6.5.backport3724
## 3.6.5.backport3724...origin/3.6.5.backport3724
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/casadi-matlab-bindings/: HEAD
5 months ago|Merge pull request #16 from ami-iit/bump365|9d66e0bb56ae4536a7e49c827901f55442b255ab|HEAD, tag: v3.6.5.0
## HEAD (no branch)
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/CppAD/: HEAD
7 months ago|stable/20240000: check_all.sh: run check_version.sh interactively when a stable version.|38f532ec5a5c098c7439d3fa51a2cb91e1b0a11e|HEAD, tag: 20240000.2
## HEAD (no branch)
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/ergocub-software/: master
3 weeks ago|ergoCub1_0: fix mass and inertia of the head (#249)|e4b567e67e746340051e296b0d9ed74c164f454b|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/GazeboYARPPlugins/: master
9 weeks ago|Merge pull request #683 from robotology/traversaro-patch-1|d296652b7619dbef39e9bd65ae49f6ca3bdd9377|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/home/lomoretti/robotology/robotology-superbuild/src/gz-sim-yarp-plugins/: main
84 minutes ago|Add handling of `VOCAB_CM_FORCE_IDLE` control mode (#172)|e681eca66446e3652b0f1f002b84845f27b3c22c|HEAD -> main, origin/main, origin/HEAD
[... 586 lines cut ...]
xorg-libxmu               1.1.3                h4ab18f5_1    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-libxtst              1.2.4                h4bc722e_0    conda-forge
xorg-libxxf86vm           1.1.5                h4bc722e_1    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-recordproto          1.14.2            h7f98852_1002    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.4           py310h2372a71_0    conda-forge
zeromq                    4.3.5                h75354e8_4    conda-forge
zlib                      1.3.1                h4ab18f5_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
zziplib                   0.13.69              he45264a_2    conda-forge

```

</details>

LoreMoretti, 2024-08-02:
Then, I am creating the world containing `ergoCubGazeboV1_1` as reported in the [wiki](https://github.com/robotology/gz-sim-yarp-plugins/wiki/How-to-use-%60gz%E2%80%90sim%E2%80%90yarp%E2%80%90plugins%60-with-ergoCub), which I copy here as well:

<details>

<summary>world_ergoCubGazeboV1_1.sdf</summary>

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~/dnn-mpc/paper_romualdi_viceconte_2024_icra_dnn-mpc-walking/worlds (dnn-mpc/tsid/unicycle)$ cat world_ergoCubGazeboV1_1.sdf 
<?xml version="1.0"?>

<sdf version="1.11">
    <world name="tutorial_controlboard">
        <physics name="1ms" type="ignored">
            <max_step_size>0.001</max_step_size>
            <real_time_factor>1.0</real_time_factor>
        </physics>
        <plugin
            filename="gz-sim-physics-system"
            name="gz::sim::systems::Physics">
        </plugin>
        <plugin
            filename="gz-sim-user-commands-system"
            name="gz::sim::systems::UserCommands">
        </plugin>
        <plugin
            filename="gz-sim-sensors-system"
            name="gz::sim::systems::Sensors">
        </plugin>
        <plugin
            filename="gz-sim-scene-broadcaster-system"
            name="gz::sim::systems::SceneBroadcaster">
        </plugin>
        <plugin
            filename="gz-sim-contact-system"
            name="gz::sim::systems::Contact">
        </plugin>

        <!-- YARP CLOCK -->
        <plugin
            filename="gz-sim-yarp-clock-system"
            name="gzyarp::Clock">
        </plugin>

        <light type="directional" name="sun">
            <cast_shadows>true</cast_shadows>
            <pose>0 0 10 0 0 0</pose>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.2 0.2 0.2 1</specular>
            <attenuation>
                <range>1000</range>
                <constant>0.9</constant>
                <linear>0.01</linear>
                <quadratic>0.001</quadratic>
            </attenuation>
            <direction>-0.5 0.1 -0.9</direction>
        </light>

        <model name="ground_plane">
            <static>true</static>
            <link name="link">
                <collision name="collision">
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                </collision>
                <visual name="visual">
                    <geometry>
[... 627 lines cut ...]
#19   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libffi.so.8.1.0", at 0x7f6f2b009fe9, in ffi_call_int
#18   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libffi.so.8.1.0", at 0x7f6f2b00aa49, in ffi_call_unix64
#17   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8-gz.so.8.3.0", at 0x7f6f2a395bcd, in runServer
#16   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a036b6f, in gz::sim::v8::Server::Server(gz::sim::v8::ServerConfig const&)
#15   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a04459e, in gz::sim::v8::ServerPrivate::CreateEntities()
#14   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a05ce09, in gz::sim::v8::SimulationRunner::SimulationRunner(sdf::v14::World const*, std::shared_ptr<gz::sim::v8::SystemLoader> const&, gz::sim::v8::ServerConfig const&)
#13   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f29fd6f1a, in gz::sim::v8::LevelManager::UpdateLevelsState()
#12   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f29fcb4a8, in gz::sim::v8::LevelManager::LoadActiveEntities(std::set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)
#11   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a017de0, in gz::sim::v8::SdfEntityCreator::CreateEntities(sdf::v14::Model const*)
#10   Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a00dcaf, in void gz::sim::v8::EventManager::Emit<gz::common::EventT<void (unsigned long, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> >), gz::sim::v8::events::LoadPluginsTag>, unsigned long const&, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> > const&>(unsigned long const&, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> > const&)
#9    Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a04e655, in gz::sim::v8::SimulationRunner::LoadPlugins(unsigned long, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> > const&)
#8    Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a06b854, in gz::sim::v8::SystemManager::LoadPlugin(unsigned long, sdf::v14::Plugin const&)
#7    Object "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/libgz-sim8.so.8.3.0", at 0x7f6f2a06b1ff, in gz::sim::v8::SystemManager::AddSystemImpl(gz::sim::v8::SystemInternal, std::shared_ptr<sdf::v14::Element const>)
[... cut ...]
#1    Object "/home/lomoretti/robotology/robotology-superbuild/build/install/lib/yarp/yarp_multipleanalogsensorsserver.so", at 0x7f6eec35e6b8, in MultipleAnalogSensorsServer::populateAllSensorsMetadata()
#0    Object "/home/lomoretti/robotology/robotology-superbuild/build/install/lib/yarp/yarp_multipleanalogsensorsremapper.so", at 0x7f6eec382314, in non-virtual thunk to MultipleAnalogSensorsRemapper::getNrOfThreeAxisGyroscopes() const
Segmentation fault (Address not mapped to object [0x8])

```

</details>

traversaro, 2024-08-02:
Interesting. If you are compiling gz-sim-yarp-plugins, can you try to enable BUILD_TESTING in the CMake build of the repo and run the tests, to understand if those work fine or not in your system?

traversaro, 2024-08-02:
Yes, it is indeed a test dependency, you can install it with the gtest conda-forge package, see https://github.com/robotology/gz-sim-yarp-plugins/blob/e681eca66446e3652b0f1f002b84845f27b3c22c/ci_env.yml#L12 .

LoreMoretti, 2024-08-02:
Here the log of the tests:

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins (master)$ ctest
Test project /home/lomoretti/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins
      Start  1: ForceTorqueTest
 1/13 Test  #1: ForceTorqueTest .............................***Failed    3.09 sec
      Start  2: ImuTest
 2/13 Test  #2: ImuTest .....................................   Passed    7.17 sec
      Start  3: LaserTest
 3/13 Test  #3: LaserTest ...................................***Failed    3.39 sec
      Start  4: CameraTest
 4/13 Test  #4: CameraTest ..................................***Failed    8.19 sec
      Start  5: ClockTest
 5/13 Test  #5: ClockTest ...................................***Exception: SegFault  4.11 sec
      Start  6: ControlBoardTorqueControlTest
 6/13 Test  #6: ControlBoardTorqueControlTest ...............***Exception: SegFault  2.10 sec
      Start  7: ControlBoardPositionDirectControlTest
 7/13 Test  #7: ControlBoardPositionDirectControlTest .......***Failed    0.46 sec
      Start  8: ControlBoardPositionControlTest
 8/13 Test  #8: ControlBoardPositionControlTest .............***Exception: SegFault  0.47 sec
      Start  9: ControlBoardCommonsTest
 9/13 Test  #9: ControlBoardCommonsTest .....................***Exception: SegFault  0.61 sec
      Start 10: ControlBoardOnMultipleGazeboInstancesTest
10/13 Test #10: ControlBoardOnMultipleGazeboInstancesTest ...***Exception: SegFault  0.52 sec
      Start 11: ConfigurationParsingFromFileTest
11/13 Test #11: ConfigurationParsingFromFileTest ............   Passed    0.48 sec
      Start 12: ConfigurationParsingFromStringTest
12/13 Test #12: ConfigurationParsingFromStringTest ..........   Passed    0.42 sec
      Start 13: ConcurrentInstancesTest
13/13 Test #13: ConcurrentInstancesTest .....................   Passed    3.46 sec

31% tests passed, 9 tests failed out of 13

Total Test time (real) =  34.47 sec

The following tests FAILED:
	  1 - ForceTorqueTest (Failed)
	  3 - LaserTest (Failed)
	  4 - CameraTest (Failed)
	  5 - ClockTest (SEGFAULT)
	  6 - ControlBoardTorqueControlTest (SEGFAULT)
	  7 - ControlBoardPositionDirectControlTest (Failed)
	  8 - ControlBoardPositionControlTest (SEGFAULT)
	  9 - ControlBoardCommonsTest (SEGFAULT)
	 10 - ControlBoardOnMultipleGazeboInstancesTest (SEGFAULT)
Errors while running CTest
Output from these tests are in: /home/lomoretti/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```

I have re-run with verbose options the failed one:

<details>

<summary>tests log with verbosity</summary>

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins (master)$ ctest --rerun-failed --output-on-failure
Test project /home/lomoretti/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins
    Start  1: ForceTorqueTest
1/9 Test  #1: ForceTorqueTest .............................***Failed    3.08 sec
Running main() from /home/conda/feedstock_root/build_artifacts/gtest-split_1722457657850/work/googletest/src/gtest_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from ForceTorqueTest
[ RUN      ] ForceTorqueTest.PluginTest
[Msg] Loading SDF world file[/home/lomoretti/robotology/robotology-superbuild/src/gz-sim-yarp-plugins/tests/forcetorque/model.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[... 857 lines cut ...]
[ERROR] Error in gzyarp::DeviceRegistry::getDevicesKeys: gz instance not found
/home/lomoretti/robotology/robotology-superbuild/src/gz-sim-yarp-plugins/tests/controlboard/ControlBoardOnMultipleGazeboInstancesTest.cc:107: Failure
Value of: deviceIds.size() == 1
  Actual: false
Expected: true

0% tests passed, 9 tests failed out of 9

Total Test time (real) =  22.26 sec

The following tests FAILED:
	  1 - ForceTorqueTest (Failed)
	  3 - LaserTest (Failed)
	  4 - CameraTest (Failed)
	  5 - ClockTest (SEGFAULT)
	  6 - ControlBoardTorqueControlTest (SEGFAULT)
	  7 - ControlBoardPositionDirectControlTest (Failed)
	  8 - ControlBoardPositionControlTest (SEGFAULT)
	  9 - ControlBoardCommonsTest (SEGFAULT)
	 10 - ControlBoardOnMultipleGazeboInstancesTest (SEGFAULT)
Errors while running CTest

```

</details>

traversaro, 2024-08-02:
This is interesting, tests are passing fine in CI https://github.com/robotology/gz-sim-yarp-plugins/actions/runs/10208894581/job/28246043512 . 

Probably you can run `conda list --explicit` and `env` and compare them with the output of the same command in CI ("Print environemnt") build step to understand if there is anything strange in your setup.

LoreMoretti, 2024-08-02:
Sure, here it is:

<details>

<summary>conda list --explicit</summary>

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~$ conda list --explicit
# This file may be used to create an environment using:
# $ conda create --name <env> --file <this file>
# platform: linux-64
@EXPLICIT
https://conda.anaconda.org/conda-forge/linux-64/bash-completion-2.11-ha770c72_1.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/_libgcc_mutex-0.1-conda_forge.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/_sysroot_linux-64_curr_repodata_hack-3-h69a702a_16.conda
https://conda.anaconda.org/conda-forge/linux-64/ca-certificates-2024.7.4-hbcca054_0.conda
https://conda.anaconda.org/conda-forge/noarch/font-ttf-dejavu-sans-mono-2.37-hab24e00_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/font-ttf-inconsolata-3.000-h77eed37_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/font-ttf-source-code-pro-2.038-h77eed37_0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/font-ttf-ubuntu-0.83-h77eed37_2.conda
https://conda.anaconda.org/conda-forge/linux-64/ld_impl_linux-64-2.40-hf3520f5_7.conda
https://conda.anaconda.org/conda-forge/linux-64/libboost-headers-1.84.0-ha770c72_3.conda
https://conda.anaconda.org/conda-forge/noarch/libgcc-devel_linux-64-12.4.0-ha4f9413_100.conda
https://conda.anaconda.org/conda-forge/noarch/libstdcxx-devel_linux-64-12.4.0-ha4f9413_100.conda
https://conda.anaconda.org/conda-forge/linux-64/mumps-include-5.7.2-ha770c72_0.conda
https://conda.anaconda.org/conda-forge/linux-64/nlohmann_json-3.11.3-h59595ed_0.conda
https://conda.anaconda.org/conda-forge/noarch/poppler-data-0.4.12-hd8ed1ab_0.conda
https://conda.anaconda.org/conda-forge/noarch/pybind11-abi-4-hd8ed1ab_3.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/python_abi-3.10-4_cp310.conda
https://conda.anaconda.org/conda-forge/noarch/tzdata-2024a-h0c530f3_0.conda
https://conda.anaconda.org/conda-forge/linux-64/utfcpp-4.0.5-ha770c72_0.conda
https://conda.anaconda.org/conda-forge/noarch/fonts-conda-forge-1-0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/kernel-headers_linux-64-3.10.0-h4a8ded7_16.conda
https://conda.anaconda.org/conda-forge/linux-64/libgomp-14.1.0-h77fa898_0.conda
https://conda.anaconda.org/conda-forge/linux-64/_openmp_mutex-4.5-2_gnu.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/fonts-conda-ecosystem-1-0.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/sysroot_linux-64-2.17-h4a8ded7_16.conda
https://conda.anaconda.org/conda-forge/linux-64/binutils_impl_linux-64-2.40-ha1999f0_7.conda
https://conda.anaconda.org/conda-forge/noarch/libdrm-cos7-x86_64-2.4.97-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/libgcc-ng-14.1.0-h77fa898_0.conda
https://conda.anaconda.org/conda-forge/noarch/libglvnd-cos7-x86_64-1.0.1-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libsepol-cos7-x86_64-2.5-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/libxshmfence-cos7-x86_64-1.2-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/mesa-khr-devel-cos7-x86_64-18.3.4-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/noarch/mesa-libglapi-cos7-x86_64-18.3.4-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/alsa-lib-1.2.12-h4ab18f5_0.conda
https://conda.anaconda.org/conda-forge/linux-64/attr-2.5.1-h166bdaf_1.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/aws-c-common-0.9.17-h4ab18f5_0.conda
https://conda.anaconda.org/conda-forge/linux-64/binutils-2.40-h4852527_7.conda
https://conda.anaconda.org/conda-forge/linux-64/binutils_linux-64-2.40-hb3c18ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/bzip2-1.0.8-h4bc722e_7.conda
https://conda.anaconda.org/conda-forge/linux-64/c-ares-1.32.3-h4bc722e_0.conda
https://conda.anaconda.org/conda-forge/linux-64/dav1d-1.2.1-hd590300_0.conda
https://conda.anaconda.org/conda-forge/linux-64/fribidi-1.0.10-h36c2ea0_0.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/gettext-tools-0.22.5-h59595ed_2.conda
https://conda.anaconda.org/conda-forge/linux-64/giflib-5.2.2-hd590300_0.conda
https://conda.anaconda.org/conda-forge/linux-64/json-c-0.17-h1220068_1.conda
https://conda.anaconda.org/conda-forge/linux-64/jxrlib-1.1-hd590300_3.conda
https://conda.anaconda.org/conda-forge/linux-64/keyutils-1.6.1-h166bdaf_0.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/lame-3.100-h166bdaf_1003.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/libbrotlicommon-1.1.0-hd590300_1.conda
https://conda.anaconda.org/conda-forge/linux-64/libdeflate-1.20-hd590300_0.conda
https://conda.anaconda.org/conda-forge/linux-64/libev-4.33-hd590300_2.conda
https://conda.anaconda.org/conda-forge/linux-64/libexpat-2.6.2-h59595ed_0.conda
https://conda.anaconda.org/conda-forge/linux-64/libffi-3.4.2-h7f98852_5.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/libgettextpo-0.22.5-h59595ed_2.conda
https://conda.anaconda.org/conda-forge/linux-64/libgfortran5-14.1.0-hc5f4f2c_0.conda
https://conda.anaconda.org/conda-forge/noarch/libglvnd-glx-cos7-x86_64-1.0.1-h9b0a68f_1105.tar.bz2
https://conda.anaconda.org/conda-forge/linux-64/libiconv-1.17-hd590300_2.conda
https://conda.anaconda.org/conda-forge/linux-64/libjpeg-turbo-3.0.0-hd590300_1.conda
[... 551 lines cut ...]
LC_ALL=en_US.UTF-8
STRIP=/home/lomoretti/mambaforge/envs/robsub_gzsim/bin/x86_64-conda-linux-gnu-strip
[... cut ...]
CXXFILT=/home/lomoretti/mambaforge/envs/robsub_gzsim/bin/x86_64-conda-linux-gnu-c++filt
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
BUILD=x86_64-conda-linux-gnu
ROS_DISTRO=humble
RANLIB=/home/lomoretti/mambaforge/envs/robsub_gzsim/bin/x86_64-conda-linux-gnu-ranlib
CONDA_BUILD_SYSROOT=/home/lomoretti/mambaforge/envs/robsub_gzsim/x86_64-conda-linux-gnu/sysroot
LC_NUMERIC=it_IT.UTF-8
PROJ_NETWORK=ON
OGRE_RESOURCE_PATH=/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/OGRE
_=/usr/bin/env

```

</details>

traversaro, 2024-08-02:
Ok, there is nothing too strange in the env (at least that I see), we can move to actually see the test failures.

For the first one, I see:

~~~
[ERROR] File not found:  ../../../tests/forcetorque/forcetorque_nws.xml
[ERROR] Error while finding yarpRobotInterfaceConfigurationFile:  ../../../tests/forcetorque/forcetorque_nws.xml
[ERROR] gz-sim-yarp-robotinterface-system :yarpRobotInterfaceConfigurationFile not found
[ERROR] gz-sim-yarp-robotinterface-system : Error loading robotinterface configuration file
~~~

Somehow the test in https://github.com/robotology/gz-sim-yarp-plugins/blob/e681eca66446e3652b0f1f002b84845f27b3c22c/tests/forcetorque/ForceTorqueTest.cc#L27 can't find the yarprobotinterface files. The problem seems with the relative path (relative w.r.t. to what?) passed to `yarpRobotInterfaceConfigurationFile`. The problem seems similar to https://github.com/robotology/gz-sim-yarp-plugins/pull/168 . I guess in CI the test pass as somehow this relative path encodes the assumption that the build directory is a child directory of the source directory of gz-sim-yarp-plugins, assumption that is not true when running the test in the superbuild.

traversaro, 2024-08-02:
Anyhow these failures are probably unrelated, we can open a new issue for tracking this @xela-95 .

xela-95, 2024-08-02:
Looking at the logs posted in https://github.com/robotology/gz-sim-yarp-plugins/issues/173#issuecomment-2264879467, here are the lines that trigger me the most:

```
[Err] [SystemLoader.cc:92] Failed to load system plugin [libgazebo_yarp_imu.so] : Could not find shared library.
[Err] [SystemLoader.cc:92] Failed to load system plugin [libgazebo_yarp_controlboard.so] : Could not find shared library.
[Err] [SystemLoader.cc:92] Failed to load system plugin [libgazebo_yarp_robotinterface.so] : Could not find shared library.
[ERROR] Target device l_arm_ft_sensor_inertial_hardware_device (network = left_arm_imu ) does not exist.
[ERROR] Cannot run attach action on device alljoints-inertials_remapper
```

traversaro, 2024-08-02:
The first three are the usual errors you get as the model also contains Gazebo Classic plugins, so you can ignore them.

traversaro, 2024-08-02:
Probably it is some kind of regression of https://github.com/icub-tech-iit/ergocub-software/pull/232/files and https://github.com/icub-tech-iit/ergocub-software/pull/232 and https://github.com/icub-tech-iit/ergocub-software/pull/236 that added the plugins for the FT imus only on Gazebo Classic and not in gz-sim ? I guess the fix is just to add those and regenerate the models, or for the time being just disable those devices from yarprobotinterface .xml files.

xela-95, 2024-08-02:
I just created a new conda environment from which I installed all the dependencies of robotology superbuild, than I created the world SDF and launched gz sim and it opened:
[image]

Here the logs:

<details><summary>Details</summary>

```
    ~/Desktop/debug_ergocub_gz_sim  gz sim ergocub-world.sdf                                                                                                                                                                ✔  robsub_debug   14:38:58  
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
[ERROR] |yarp.os.Network| Cannot connect to port /root
[ERROR] |yarp.os.Port| Port /clock failed to activate (invalid address)
[ERROR] Failed to open port /clock
[Err] [SystemLoader.cc:92] Failed to load system plugin [libgazebo_yarp_imu.so] : Could not find shared library.
[INFO] File is a URI:  model://ergoCub/conf/gazebo_ergocub_waist_inertial.ini
[INFO] Yarp configuration file loaded:  /home/acroci/mambaforge/envs/jaxsim/share/ergoCub/conf/gazebo_ergocub_waist_inertial.ini
[INFO] gz-sim-yarp-imu-system: configuration of sensor  waist_imu_0  loaded
[DEBUG] |yarp.dev.PolyDriver|gazebo_imu| Parameters are (device gazebo_imu) (disableImplicitNetworkWrapper) (parentLinkName torso_1) (sensorName waist_imu_0) (sensorScopedName "model/ergoCub/link/torso_1/sensor/waist_imu_0") (sensor_name waist_imu_0) (yarpDeviceName waist_inertial_hardware_device)
[INFO] |yarp.dev.PolyDriver|gazebo_imu| Created device <gazebo_imu>. See C++ class IMUDriver for documentation.
[INFO] Registered YARP device with instance name: model/ergoCub/waist_inertial_hardware_device
[ERROR] Yarp network does not seem to be available, is the yarpserver running?
[ERROR] Yarp network does not seem to be available, is the yarpserver running?
[Err] [SystemLoader.cc:92] Failed to load system plugin [libgazebo_yarp_controlboard.so] : Could not find shared library.
[INFO] File is a URI:  model://ergoCub/conf/gazebo_ergocub_torso.ini
[INFO] Yarp configuration file loaded:  /home/acroci/mambaforge/envs/jaxsim/share/ergoCub/conf/gazebo_ergocub_torso.ini
[INFO] gz-sim-yarp-controlboard-system: configuration of device  torso_hardware_device  loaded
[DEBUG] gz-sim-yarp-controlboard-system : robot scoped name:  model/ergoCub
[DEBUG] gz-sim-yarp-controlboard-system : yarpDeviceName:  torso_hardware_device
[DEBUG] |yarp.dev.PolyDriver|gazebo_controlboard| Parameters are (IMPEDANCE_CONTROL (controlUnits metric_units) (controlLaw joint_pid_gazebo_v1) (stiffness 0.0 0.0 0.0) (damping 0.0 0.0 0.0)) (LIMITS (jntPosMax 43.0 45.0 23.0) (jntPosMin -43.0 -18.0 -23.0) (jntVelMax 100.0 100.0 100.0)) (POSITION_CONTROL (controlUnits metric_units) (controlLaw joint_pid_gazebo_v1) (kp 70.0 70.0 70.0) (kd 0.149999999999999994449 0.149999999999999994449 0.149999999999999994449) (ki 0.170000000000000012212 0.170000000000000012212 0.170000000000000012212) (maxInt 9999 9999 9999) (maxOutput 9999 9999 9999) (shift 0.0 0.0 0.0) (ko 0.0 0.0 0.0) (stictionUp 0.0 0.0 0.0) (stictionDwn 0.0 0.0 0.0)) (TRAJECTORY_GENERATION (trajectory_type minimum_jerk)) (VELOCITY_CONTROL (velocityControlImplementationType integrator_and_position_pid) (controlUnits metric_units) (controlLaw joint_pid_gazebo_v1) (kp 8.72600000000000086686 8.72600000000000086686 8.72600000000000086686) (kd 0.0350000000000000033307 0.0350000000000000033307 0.0350000000000000033307) (ki 0.00200000000000000004163 0.00200000000000000004163 0.00200000000000000004163) (maxInt 9999 9999 9999) (maxOutput 9999 9999 9999) (shift 0.0 0.0 0.0) (ko 0.0 0.0 0.0) (stictionUp 0.0 0.0 0.0) (stictionDwn 0.0 0.0 0.0)) (controlBoardId "model/ergoCub") (device gazebo_controlboard) (disableImplicitNetworkWrapper) (jointNames torso_roll torso_pitch torso_yaw) (max_damping 100.0 100.0 100.0) (max_stiffness 1000.0 1000.0 1000.0) (max_torques 1000000.0 1000000.0 1000000.0) (min_damping 0.0 0.0 0.0) (min_stiffness 0.0 0.0 0.0) (yarpDeviceName torso_hardware_device)
[INFO] |yarp.dev.PolyDriver|gazebo_controlboard| Created device <gazebo_controlboard>. See C++ class ControlBoardDriver for documentation.
[WARNING] Parameter  jointNames  should be a list, but its format is deprecated as parenthesis are missing.  Please add parentesis to the list, as documented in https://github.com/robotology/yarp/discussions/3092 .
[INFO] Found 3 joints from the plugin configuration.
[INFO] Found 65 joints from the model description.
[INFO] Joint  torso_roll  added to the control board data.
[INFO] Joint  torso_pitch  added to the control board data.
[INFO] Joint  torso_yaw  added to the control board data.
[WARNING] Parameter  jntPosMin  should be a list, but its format is deprecated as parenthesis are missing.  Please add parentesis to the list, as documented in https://github.com/robotology/yarp/discussions/3092 .
[WARNING] Parameter  jntPosMax  should be a list, but its format is deprecated as parenthesis are missing.  Please add parentesis to the list, as documented in https://github.com/robotology/yarp/discussions/3092 .
[INFO] Trajectory generator type set to  minimum_jerk
[ERROR] refSpeed  not found
[WARNING] Parameter refSpeed not found in TRAJECTORY_GENERATION group. Defaults will be applied
[ERROR] refAcceleration  not found
[WARNING] Parameter refAcceleration not found in TRAJECTORY_GENERATION group. Defaults will be applied
[DEBUG] Joint  torso_roll  trajectory generation reference speed:  10  [deg/s]
[DEBUG] Joint  torso_roll  trajectory generation reference acceleration:  10  [deg/s^2]
[... cut ...]
https://conda.anaconda.org/conda-forge/linux-64/pcl-1.14.1-hbf7b2d8_3.conda
https://conda.anaconda.org/conda-forge/linux-64/libgz-gui8-8.0.0-hee5a715_6.conda
https://conda.anaconda.org/conda-forge/linux-64/libgz-sensors8-8.0.0-hf437a2f_5.conda
https://conda.anaconda.org/conda-forge/linux-64/libgz-sim8-8.3.0-h5fd5377_0.conda
https://conda.anaconda.org/conda-forge/linux-64/gz-sim8-python-8.3.0-py310h64a2142_0.conda
https://conda.anaconda.org/conda-forge/linux-64/gz-sim8-8.3.0-ha770c72_0.conda
```
</details>

traversaro, 2024-08-02:
Interesting, perhaps there is a different between the latest released version and the latest master version of ergocub-software? Anyhow, I guess that downgrading ergocub-software to a version before https://github.com/icub-tech-iit/ergocub-software/pull/232 should also be a valid workaround for the problem.

xela-95, 2024-08-02:
I thought I was able to launch correctly gz sim, but I it worked only because I didn't launched `yarp server` before. Launching it I get the same segfault exception of @LoreMoretti.

traversaro, 2024-08-02:
By the way, the fact that is crashing instead of printing errors may be another thing we may want to address eventually.

xela-95, 2024-08-02:
> Anyhow, I guess that downgrading ergocub-software to a version before [icub-tech-iit/ergocub-software#232](https://github.com/icub-tech-iit/ergocub-software/pull/232) should also be a valid workaround for the problem.

I checked out the [ergocub-software](https://github.com/icub-tech-iit/ergocub-software/tree/master) package to commit [49402203de8f1ef3bfd324f6b6f9ac55a1cebf1f](https://github.com/icub-tech-iit/ergocub-software/commit/49402203de8f1ef3bfd324f6b6f9ac55a1cebf1f), compiled and installed and I was able to launch the world file as in the tutorial. 

I had a look at the PR https://github.com/icub-tech-iit/ergocub-software/pull/236 and I'm trying to understand where to make the necessary adjustments. In the meantime using the version suggested above can unlock @LoreMoretti in performing his tests.

P.S.: Sorry for the delay in the response but I had to refresh how to handle this stuff 😅

xela-95, 2024-08-07:
@LoreMoretti now that https://github.com/icub-tech-iit/ergocub-software/pull/250 has been merged you should be able to complete the tutorial 🤞🏻

LoreMoretti, 2024-08-12:
I pulled and recompiled, 

Then, I started `yarpserver` and run `gz sim ergocubGazevoV1_1_gz_sim_world.sdf`, what I get is the following:

[image]

So there is no more a crash, but the `gz-sim` window remains black, and this line `[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...`  is repeatedly printed on the terminal.

Is there anything I should do related to the `clock` port, before launching `gz-sim`?

LoreMoretti, 2024-08-12:
My bad, I was also exporting `$YARP_CLOCK` env variable (like `export YARP_CLOCK=/clock`), as I was used to do it for the `old gazebo`.

Is this not needed anymore for `gz-sim`?

xela-95, 2024-08-12:
> So there is no more a crash, but the `gz-sim` window remains black, and this line `[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...` is repeatedly printed on the terminal.
> 
> Is there anything I should do related to the `clock` port, before launching `gz-sim`?

Uhm that's strange, in theory if you just want to launch the simulation you don't need to set that environment variable, so Gazebo should open normally. However if you want to launch some controllers to interact with the simulation, as it is done in https://github.com/ami-iit/walking-controllers#computer-how-to-run-the-simulation (that I tested with modern Gazebo and gz-sim-yarp-plugins in https://github.com/robotology/gz-sim-yarp-plugins/issues/122) you have to export that variable as you do with Classic Gazebo.

traversaro, 2024-08-12:
YARP_CLOCK is required to make sure that the YARP's network wrapper server indeed publish with  a frequency coherent with the simulation time. Are you including the `gz-sim-yarp-clock` plugin in the world?

LoreMoretti, 2024-08-12:
> YARP_CLOCK is required to make sure that the YARP's network wrapper server indeed publish with a frequency coherent with the simulation time.

Clear, thanks!

> Are you including the gz-sim-yarp-clock plugin in the world?

I see the following lines in the world file:

```
        <!-- YARP CLOCK -->
        <plugin
            filename="gz-sim-yarp-clock-system"
            name="gzyarp::Clock">
        </plugin>
```

So I think yes.

LoreMoretti, 2024-08-12:
Closing this issue since the crash has been fixed.

Continuing the discussion of the `clock` issue in a [separate issue](https://github.com/robotology/gz-sim-yarp-plugins/issues/182).