robotology/gz-sim-yarp-plugins #174: Unit tests crashing

xela-95, 2024-08-02:
In #173 @LoreMoretti tried to run from his machine the unit tests, getting a lot of errors:

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins (master)$ ctest
Test project /home/lomoretti/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins
      Start  1: ForceTorqueTest
 1/13 Test  #1: ForceTorqueTest .............................***Failed    3.09 sec
      Start  2: ImuTest
 2/13 Test  #2: ImuTest .....................................   Passed    7.17 sec
      Start  3: LaserTest
 3/13 Test  #3: LaserTest ...................................***Failed    3.39 sec
      Start  4: CameraTest
 4/13 Test  #4: CameraTest ..................................***Failed    8.19 sec
      Start  5: ClockTest
 5/13 Test  #5: ClockTest ...................................***Exception: SegFault  4.11 sec
      Start  6: ControlBoardTorqueControlTest
 6/13 Test  #6: ControlBoardTorqueControlTest ...............***Exception: SegFault  2.10 sec
      Start  7: ControlBoardPositionDirectControlTest
 7/13 Test  #7: ControlBoardPositionDirectControlTest .......***Failed    0.46 sec
      Start  8: ControlBoardPositionControlTest
 8/13 Test  #8: ControlBoardPositionControlTest .............***Exception: SegFault  0.47 sec
      Start  9: ControlBoardCommonsTest
 9/13 Test  #9: ControlBoardCommonsTest .....................***Exception: SegFault  0.61 sec
      Start 10: ControlBoardOnMultipleGazeboInstancesTest
10/13 Test #10: ControlBoardOnMultipleGazeboInstancesTest ...***Exception: SegFault  0.52 sec
      Start 11: ConfigurationParsingFromFileTest
11/13 Test #11: ConfigurationParsingFromFileTest ............   Passed    0.48 sec
      Start 12: ConfigurationParsingFromStringTest
12/13 Test #12: ConfigurationParsingFromStringTest ..........   Passed    0.42 sec
      Start 13: ConcurrentInstancesTest
13/13 Test #13: ConcurrentInstancesTest .....................   Passed    3.46 sec

31% tests passed, 9 tests failed out of 13

Total Test time (real) =  34.47 sec

The following tests FAILED:
	  1 - ForceTorqueTest (Failed)
	  3 - LaserTest (Failed)
	  4 - CameraTest (Failed)
	  5 - ClockTest (SEGFAULT)
	  6 - ControlBoardTorqueControlTest (SEGFAULT)
	  7 - ControlBoardPositionDirectControlTest (Failed)
	  8 - ControlBoardPositionControlTest (SEGFAULT)
	  9 - ControlBoardCommonsTest (SEGFAULT)
	 10 - ControlBoardOnMultipleGazeboInstancesTest (SEGFAULT)
Errors while running CTest
Output from these tests are in: /home/lomoretti/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```

xela-95, 2024-08-02:
Here are the logs from @LoreMoretti 

<details><summary>Details</summary>

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins (master)$ ctest --rerun-failed --output-on-failure
Test project /home/lomoretti/robotology/robotology-superbuild/build/src/gz-sim-yarp-plugins
    Start  1: ForceTorqueTest
1/9 Test  #1: ForceTorqueTest .............................***Failed    3.08 sec
Running main() from /home/conda/feedstock_root/build_artifacts/gtest-split_1722457657850/work/googletest/src/gtest_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from ForceTorqueTest
[ RUN      ] ForceTorqueTest.PluginTest
[Msg] Loading SDF world file[/home/lomoretti/robotology/robotology-superbuild/src/gz-sim-yarp-plugins/tests/forcetorque/model.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ForceTorque] for entity [1]
[Msg] Create service on [/world/sensors/create]
[Msg] Remove service on [/world/sensors/remove]
[Msg] Pose service on [/world/sensors/set_pose]
[Msg] Pose service on [/world/sensors/set_pose_vector]
[Msg] Light configuration service on [/world/sensors/light_config]
[Msg] Physics service on [/world/sensors/set_physics]
[Msg] SphericalCoordinates service on [/world/sensors/set_spherical_coordinates]
[Msg] Enable collision service on [/world/sensors/enable_collision]
[Msg] Disable collision service on [/world/sensors/disable_collision]
[Msg] Material service on [/world/sensors/visual_config]
[Msg] Material service on [/world/sensors/wheel_slip]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[INFO] Yarp configuration string loaded:  (yarpDeviceName forcetorque_plugin_device) (jointName joint_12) (sensorName force_torque_sensor)
[INFO] gz-sim-yarp-forcetorque-system: configuration of sensor  force_torque_sensor  loaded
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (jointName joint_12) (sensorName force_torque_sensor) (sensorScopedName "model/force_torque_model/joint/joint_12/sensor/force_torque_sensor") (sensor_name force_torque_sensor) (yarpDeviceName forcetorque_plugin_device)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class ForceTorqueDriver for documentation.
[INFO] Registered YARP device with instance name: model/force_torque_model/forcetorque_plugin_device
[Dbg] [SystemManager.cc:74] Loaded system [gzyarp::ForceTorque] for entity [8]
[ERROR] File not found:  ../../../tests/forcetorque/forcetorque_nws.xml
[ERROR] Error while finding yarpRobotInterfaceConfigurationFile:  ../../../tests/forcetorque/forcetorque_nws.xml
[ERROR] gz-sim-yarp-robotinterface-system :yarpRobotInterfaceConfigurationFile not found
[ERROR] gz-sim-yarp-robotinterface-system : Error loading robotinterface configuration file
[Wrn] [TestFixture.cc:208] Fixture has not been finalized, any functions you attemptedto hook into will not be run. It is recommended to call Finalize()before accessing the server.
[Wrn] [SimulationRunner.cc:725] Found additional publishers on /stats, using namespaced stats topic only
[Wrn] [SimulationRunner.cc:759] Found additional publishers on /clock, using namespaced clock topic only
[Dbg] [SystemManager.cc:74] Loaded system [gzyarp::RobotInterface] for entity [8]
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/sensors/control], [/world/sensors/control/state] and [/world/sensors/playback/control]
[Msg] Serving GUI information on [/world/sensors/gui/info]
[Msg] World [sensors] initialized with [1ms] physics profile.
[Msg] Serving world SDF generation service on [/world/sensors/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Dbg] [SimulationRunner.cc:727] Publishers [Address, Message Type]:
[Dbg] [SimulationRunner.cc:732]   tcp://172.20.10.5:34809, gz.msgs.WorldStatistics
[Dbg] [SimulationRunner.cc:761] Publishers [Address, Message Type]:
[Dbg] [SimulationRunner.cc:766]   tcp://172.20.10.5:34809, gz.msgs.Clock
[Dbg] [SimulationRunner.cc:547] Creating PostUpdate worker threads: 4
[Dbg] [SimulationRunner.cc:558] Creating postupdate worker thread (0)
[Dbg] [SimulationRunner.cc:558] Creating postupdate worker thread (1)
[Dbg] [SimulationRunner.cc:558] Creating postupdate worker thread (2)
[Msg] Serving scene information on [/world/sensors/scene/info]
[Msg] Serving graph information on [/world/sensors/scene/graph]
[Msg] Serving full state on [/world/sensors/state]
[Msg] Serving full state (async) on [/world/sensors/state_async]
[Msg] Publishing scene information on [/world/sensors/scene/info]
[... 804 lines cut ...]
[... cut ...]
0% tests passed, 9 tests failed out of 9

Total Test time (real) =  22.26 sec

The following tests FAILED:
	  1 - ForceTorqueTest (Failed)
	  3 - LaserTest (Failed)
	  4 - CameraTest (Failed)
	  5 - ClockTest (SEGFAULT)
	  6 - ControlBoardTorqueControlTest (SEGFAULT)
	  7 - ControlBoardPositionDirectControlTest (Failed)
	  8 - ControlBoardPositionControlTest (SEGFAULT)
	  9 - ControlBoardCommonsTest (SEGFAULT)
	 10 - ControlBoardOnMultipleGazeboInstancesTest (SEGFAULT)
Errors while running CTest

```

</details>

traversaro, 2024-08-21:
After:
* https://github.com/robotology/gz-sim-yarp-plugins/issues/175
* https://github.com/robotology/gz-sim-yarp-plugins/pull/176
* https://github.com/robotology/gz-sim-yarp-plugins/pull/193
* https://github.com/robotology/gz-sim-yarp-plugins/pull/194

I do not have anymore any crash locally (in WSL I need to set `export LIBGL_ALWAYS_SOFTWARE=1` to avoid segfaults on exit, but that is a WSL-specific thing) or in CI, so @LoreMoretti when you have time feel free to try this in an updated version of gz-sim-yarp-plugins to check if the problems have been solved.

xela-95, 2024-08-29:
Thank you @LoreMoretti ! We can close this now.

closed 2024-08-29 by xela-95