robotology/gz-sim-yarp-plugins #233: Define controlboard for nested model

traversaro, 2025-02-09:
In Gazebo Classic/gazebo-yarp-plugins, it was possible to have a plugin-less model, and then include it in a model in which the gazebo controlboard plugin for its joints, for example something like:

~~~xml
<model name="pendulum_with_yarp_plugins">
    <include>
        <uri>model://single_pendulum_no_initial_configuration.sdf</uri>
    </include>
    <plugin name="gzyarp::ControlBoard" filename="gz-sim-yarp-controlboard-system">
        <yarpConfigurationFile>
            model://conf/gazebo_controlboard.ini
        </yarpConfigurationFile>
    </plugin>
</model>
~~~

This is not possible anymore with gz-sim-yarp-plugins, and for example created duplication in test files added in https://github.com/robotology/gz-sim-yarp-plugins/pull/232 . It would be great to support this pattern also in gz-sim-yarp-plugins .

traversaro, 2025-02-09:
See https://github.com/gazebosim/gz-sim/issues/1844 for a similar bug in gz-sim.

xela-95, 2025-02-10:
@traversaro do you think this is a current limitation of the plugins or it depends on gz-sim itself? 

Issue https://github.com/gazebosim/gz-sim/issues/1844 is closed as completed and the related PRs merged if I'm not mistaken, so it gz-sim should allow this, right?

traversaro, 2025-02-10:
I would guess mostly a current limitation of the plugin. In https://github.com/robotology/gz-sim-yarp-plugins/blob/ea02b61a2c46940fc9db51b78bb035e363752593/plugins/controlboard/src/ControlBoard.cpp#L252 we should make sure that we also search for the joints of the nested models, and I guess somehow gz-sim allows that as https://github.com/gazebosim/gz-sim/issues/1844 was solved.