robotology/gz-sim-yarp-plugins #235: Implement ISystemReset interface in all plugins and ensure that a system reset does not result in YARP ports being closed and reopened

traversaro, 2025-02-12:
See https://github.com/robotology/gz-sim-yarp-plugins/issues/206#issuecomment-2652966078 .

LoreMoretti, 2025-03-10:
It would be extremely useful that when the simulation gets reset the robot will start from the initial configuration set through the `gzyarp::ConfigurationOverride` plugin in the world file.

traversaro, 2025-03-10:
> It would be extremely useful that when the simulation gets reset the robot will start from the initial configuration set through the `gzyarp::ConfigurationOverride` plugin in the world file.

It does not? That's a bug that I think is independent from the one tracked here.

traversaro, 2025-03-10:
> > It would be extremely useful that when the simulation gets reset the robot will start from the initial configuration set through the `gzyarp::ConfigurationOverride` plugin in the world file.
> 
> It does not? That's a bug that I think is independent from the one tracked here.

Follow up in https://github.com/robotology/gz-sim-yarp-plugins/issues/252 .