robotology/gz-sim-yarp-plugins #245: Torque control mode: cannot set reference torque

PasMarra, 2025-03-04:
https://github.com/user-attachments/assets/d8369dff-a3c2-4ada-94bc-260d70b67416

When switching to torque control mode, the joint does not keep the measured torque as reference. Furthermore, it is not possible to set a new reference.

traversaro, 2025-03-04:
> When switching to torque control mode, the joint does not keep the measured torque as reference.

Thanks! That is probably a bug in the switching of control modes. The related code in gazebo-yarp-plugins is https://github.com/robotology/gazebo-yarp-plugins/blob/7206f84eca17d83f6b9d5b90e3f1b05d0be7a78e/plugins/controlboard/src/ControlBoardDriverControlMode.cpp#L139-L181 , but probably there is no equivalent code in gz-sim-yarp-plugins. fyi @xela-95 

> Furthermore, it is not possible to set a new reference.

In which control mode? Can you point to the part of the video where this happens?

PasMarra, 2025-03-04:
> In which control mode? 

Torque control

> Can you point to the part of the video where this happens?

`shoulder_roll`:
1- Switching to torque control mode: min 0:06.
2- Setting new reference for the torque control mode: min 0:09 - 0:15.

`shoulder_pitch`:
1- Switching to torque control mode: min 0:21.
2- Setting new reference for the torque control mode: min 0:24 - 0:28.

traversaro, 2025-03-05:
Thanks a lot @PasMarra for reporting this critical regression, this may be of interest for other users of gz-sim @LoreMoretti . As the main problem was the torque reference not being correctly set, I would split the other problem ("When switching to torque control mode, the joint does not keep the measured torque as reference. ") in a separate issue: https://github.com/robotology/gz-sim-yarp-plugins/issues/249 .

closed 2025-03-05 by traversaro