robotology/gz-sim-yarp-plugins #247: VOCAB_CM_TORQUE control not supported in coupled joints

traversaro, 2025-03-04:
The code that converts desired quantities from actuated axis to physical joints and measured quantities from physical joints to actuated axis seems only consider desired positions and measured position and velocities, see:
* https://github.com/robotology/gz-sim-yarp-plugins/blob/13720141e6f95574d83042db1a5572298c44085f/plugins/controlboard/src/ControlBoard.cpp#L382-L390
* https://github.com/robotology/gz-sim-yarp-plugins/blob/13720141e6f95574d83042db1a5572298c44085f/plugins/controlboard/src/ControlBoard.cpp#L472-L481

So, clearly the `VOCAB_CM_TORQUE` control mode is not supported for coupled joints. At the moment we should document this limitation, and ideally return an error if someone asks to set in torque a coupling joint. We could also try to understand if it make sense to support this (for example if it is supported in any physical robot) and if it exists a case in which it make sense to support this, implement it.

traversaro, 2025-03-05:
> I guess the simplest solution was to implement it: [#248](https://github.com/robotology/gz-sim-yarp-plugins/pull/248) .

This works fine for coupling devices that implement the torque mapping methods, but some do not (see https://github.com/robotology/gz-sim-yarp-plugins/pull/248#pullrequestreview-2660646495). In that case we need to understand how to handle. For the control I think the best solution is return error if somebody tries to switch the coupled joints in torque control mode, while I am not sure how to handle the torque feedback case. In theory we could return false in `getTorques`, but I am afraid this would not be handled nicely by the various remappers logic. Probably the easiest solution is to return the same value returned in the real robot for the ergocub heands, @PasMarra as you are probably the person with most experience with the real robot, can you check that? Thanks!

traversaro, 2025-03-05:
Another problem in the torque control mode in coupled joints is that given how it is implemented now is possible that in underactuated systems that approximate system in which there is come kind of "position constraint" among the different physical joints, controlling the coupled device in torque will result in a drift that will break the position constraints. This does not happen in other control modes as there eventually everything is converted to a reference position on the actuated axis side, that is converted via position mapping to a physical joint reference that then is passed to a position PID, so the enforcing of any position constraint is done implicitly by the PID position control on the physical joint side.

PasMarra, 2025-03-24:
Hi @traversaro, I've tried  the torque control mode for the fingers of ergoCub002, but also on the real robot this control mode is not properly handled.

It happens that `yarp::dev::IControlMode::setControlMode()` returns **true** but the control mode **does not change**; then, if i set a reference with `yarp::dev::ITorqueControl::setRefTorques()`, still the return value is  **true** but the reference values do not change. Analogous situation for `yarp::dev::ITorqueControl::getRefTorques()`.