robotology/gz-sim-yarp-plugins #252: Allow to restart from initial configuration when resetting the simulation

LoreMoretti, 2025-03-10:
I am using [this version](https://github.com/robotology/gz-sim-yarp-plugins/tree/9691fa7f632172d93890c104167d3f6828749cb8) of `gz-sim-yarp-plugins`.

What I usually do is:

1. run `yarpserver`
2. start gz-sim like `YARP_CLOCK=/clock gz sim -r ./world/world.sdf `
3. start `WBD`
4. run a controller.

Then, I do:

1. stop the controller
2. reset the gz simulation (`ctrl+R` or clicking the reset icon)

But, I noticed that the robot it is not placed back to the initial configuration specified through the `gzyarp::ConfigurationOverride` plugin in the world file.

cc @traversaro

traversaro, 2025-03-10:
Ok, but the initial configuration specified from the `gzyarp::ConfigurationOverride` works fine the first time, right?

LoreMoretti, 2025-03-10:
> Ok, but the initial configuration specified from the `gzyarp::ConfigurationOverride` works fine the first time, right?

yes, it works fine.

traversaro, 2025-03-11:
I was able to reproduce the problem in the case of https://github.com/robotology/gz-sim-yarp-plugins/blob/main/tutorial/single_pendulum/single_pendulum_world.sdf, but I had to add to the world also the clock plugin and to set `YARP_CLOCK=/clock`.

traversaro, 2025-03-11:
However, I am not sure if the problem of `YARP_CLOCK=/clock gz sim -r ./world/world.sdf` is that the initial position is not reset. It seems to me that if I launch  `YARP_CLOCK=/clock gz sim -r ./world/world.sdf` and then I try to reset, the simulation does not restart? Probably in https://github.com/robotology/gz-sim-yarp-plugins/pull/236 I was a bit too extremistic, and instead it make sense to have the reset method, at least for the Clock plugins so that the port of the clock is never closed? Anyhow, the CLock plugin is not related at all with the robotinterface, so I guess that it is not a problem if only that one has a reset properly implemented.

traversaro, 2025-03-11:
Even implementing the reset, during reset the gz sim process blocks after the message:

~~~
[INFO] Executing detach action, level 5 on device single_pendulum_nws_yarp with parameters []
~~~

and the following message never appears:

~~~
[INFO] All actions for action level 5 of shutdown phase started. Waiting for unfinished actions.
~~~

something somehow is blocked while waiting for the clock. Possible solution are:
* make sure that nothing is blocked on the problematic clock
* find another way instead of `YARP_CLOCK=/clock` to ensure that gz-sim publishers are synchronized with the simulation.

traversaro, 2025-03-12:
xref https://github.com/robotology/gz-sim-yarp-plugins/issues/182 that was basically the same issue but on first run, not reset.

traversaro, 2025-03-15:
> something somehow is blocked while waiting for the clock. Possible solution are:

While blocked, the bt is:

~~~
Download failed: Invalid argument.  Continuing without source file ./nptl/./nptl/futex-internal.c.
0x00007ffff714fd71 in __futex_abstimed_wait_common64 (private=128, cancel=true, abstime=0x0, op=265, expected=6143, futex_word=0x7fffaa7fc990)
    at ./nptl/futex-internal.c:57
warning: 57     ./nptl/futex-internal.c: No such file or directory
(gdb) bt
#0  0x00007ffff714fd71 in __futex_abstimed_wait_common64 (private=128, cancel=true, abstime=0x0, op=265, expected=6143, futex_word=0x7fffaa7fc990)
    at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=128, abstime=0x0, clockid=0, expected=6143, futex_word=0x7fffaa7fc990) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x7fffaa7fc990, expected=6143, clockid=clockid@entry=0, abstime=abstime@entry=0x0,
    private=private@entry=128) at ./nptl/futex-internal.c:139
#3  0x00007ffff71557a3 in __pthread_clockjoin_ex (threadid=140736053888704, thread_return=0x0, clockid=0, abstime=0x0, block=<optimized out>)
    at ./nptl/pthread_join_common.c:102
#4  0x00007ffff73bba17 in __gthread_join (__threadid=<optimized out>, __value_ptr=0x0)
    at /home/conda/feedstock_root/build_artifacts/gcc_compilers_1729025086244/work/build/x86_64-conda-linux-gnu/libstdc++-v3/include/x86_64-conda-linux-gnu/bits/gthr-default.h:682
#5  std::thread::join (this=0x5555572dd4b0) at ../../../../../libstdc++-v3/src/c++11/thread.cc:134
#6  0x00007ffff7638f56 in yarp::os::impl::ThreadImpl::join (this=0x5555572dd490, seconds=-1)
    at /home/traversaro/robotology-superbuild/src/YARP/src/libYARP_os/src/yarp/os/impl/ThreadImpl.cpp:138
#7  0x00007fffb9c17dc9 in non-virtual thunk to ControlBoard_nws_yarp::detach() ()
    at /home/traversaro/robotology-superbuild/src/YARP/src/devices/networkWrappers/controlBoard_nws_yarp/ControlBoard_nws_yarp.h:129
#8  0x00007fffb9c797e3 in yarp::robotinterface::Device::detach (this=0x555557234d40)
    at /home/traversaro/robotology-superbuild/src/YARP/src/libYARP_robotinterface/src/yarp/robotinterface/Device.cpp:471
#9  0x00007fffb9c85a53 in yarp::robotinterface::Robot::enterPhase (this=this@entry=0x5555572b9848, phase=phase@entry=yarp::robotinterface::ActionPhaseShutdown)
    at /home/traversaro/robotology-superbuild/src/YARP/src/libYARP_robotinterface/src/yarp/robotinterface/Robot.cpp:730
#10 0x00007fffb9c481ef in gzyarp::RobotInterface::CloseRobotInterface (this=this@entry=0x5555572b9830)
    at /home/traversaro/robotology-superbuild/src/gz-sim-yarp-plugins/plugins/robotinterface/RobotInterface.cc:61
#11 0x00007fffb9c48415 in gzyarp::RobotInterface::OnDeviceRemoved (this=0x5555572b9830, removeDeviceRegistryDatabaseKey=...)
    at /home/traversaro/robotology-superbuild/src/gz-sim-yarp-plugins/plugins/robotinterface/RobotInterface.cc:79
#12 0x00007fffb9c423b6 in std::__invoke_impl<void, void (gzyarp::RobotInterface::*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), gzyarp::RobotInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__f=<optimized out>, __t=<optimized out>)
    at /home/traversaro/robotology-superbuild/.pixi/envs/default/lib/gcc/x86_64-conda-linux-gnu/13.3.0/include/c++/bits/invoke.h:74
#13 std::__invoke<void (gzyarp::RobotInterface::*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), gzyarp::RobotInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__fn=<optimized out>)
    at /home/traversaro/robotology-superbuild/.pixi/envs/default/lib/gcc/x86_64-conda-linux-gnu/13.3.0/include/c++/bits/invoke.h:96
#14 std::_Bind<void (gzyarp::RobotInterface::*(gzyarp::RobotInterface*, std::_Placeholder<1>))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>::__call<void, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, 0ul, 1ul>(std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&>&&, std::_Index_tuple<0ul, 1ul>) (__args=..., this=<optimized out>)
    at /home/traversaro/robotology-superbuild/.pixi/envs/default/lib/gcc/x86_64-conda-linux-gnu/13.3.0/include/c++/functional:506
#15 std::_Bind<void (gzyarp::RobotInterface::*(gzyarp::RobotInterface*, std::_Placeholder<1>))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>::operator()<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) (this=<optimized out>) at /home/traversaro/robotology-superbuild/.pixi/envs/default/lib/gcc/x86_64-conda-linux-gnu/13.3.0/include/c++/functional:591
#16 std::__invoke_impl<void, std::_Bind<void (gzyarp::RobotInterface::*(gzyarp::RobotInterface*, std::_Placeholder<1>))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__invoke_other, std::_Bind<void (gzyarp::RobotInterface::*(gzyarp::RobotInterface*, std::_Placeholder<1>))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) (__f=...)
[... cut ...]
#18 std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), std::_Bind<void (gzyarp::RobotInterface::*(gzyarp::RobotInterface*, std::_Placeholder<1>))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)> >::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) (__functor=..., __args#0=...)
    at /home/traversaro/robotology-superbuild/.pixi/envs/default/lib/gcc/x86_64-conda-linux-gnu/13.3.0/include/c++/bits/std_function.h:290
~~~

traversaro, 2025-03-15:
I am not sure what is happening exactly, but the overall logic (I guess) is that the main physics thread is setting the `closing` varialble for the controlboard_nws_yarp publishing thread to close in https://github.com/robotology/yarp/blob/b9c4888aef50b2014221d53ce8ee9c45175a74dd/src/devices/networkWrappers/controlBoard_nws_yarp/ControlBoard_nws_yarp.cpp#L342 and , and then waiting for the thread to close in https://github.com/robotology/yarp/blob/b9c4888aef50b2014221d53ce8ee9c45175a74dd/src/libYARP_os/src/yarp/os/impl/ThreadImpl.cpp#L155-L159, but the thread is blocked in the delay function in https://github.com/robotology/yarp/blob/b9c4888aef50b2014221d53ce8ee9c45175a74dd/src/libYARP_os/src/yarp/os/PeriodicThread.cpp#L266, that would be unblocked only by a clock tick. I wonder how it was possible to it to work fine in Gazebo Classic.

traversaro, 2025-03-15:
I pushed a solution for the reset problem in https://github.com/robotology/gz-sim-yarp-plugins/commit/fc9ac9da4bedc5c94363f28f4af1a35be3f34035 . The basic idea is to continue to publish increasing time values to `/clock` when a reset is ongoing, to ensure that any YARP thread that may be blocked on it (due to `YARP_CLOCK=/clock` is not blocked doing the reset. That works fine, but the problem is there again at the shutdown instead of the reset, and so the problem is still not solved.

traversaro, 2025-03-15:
> The main problem is that we can't easily enforce that the Clock constructor is the last to be called, in particular if a plugin is specified in a nested model, its destructor will always be called after the one of the plugin of the nested model.

In a sense, the problem is similar to https://github.com/robotology/gz-sim-yarp-plugins/pull/186, but in this case the problem is more subtle as the relation between plugins is established in an indirect way by the `YARP_CLOCK` env variable, so it is not possible to track the relation explicitly.

traversaro, 2025-03-15:
One thing that could simplify the problem is that the only plugin that can spawn yarp::os::PeriodicThread is the RobotInterface plugin, so probably we only need to worry about the relation between Clock and RobotInterface instances, and not about any other plugin.

traversaro, 2025-03-17:
> In a sense, the problem is similar to [#186](https://github.com/robotology/gz-sim-yarp-plugins/pull/186), but in this case the problem is more subtle as the relation between plugins is established [...]

Indeed, I tried to solve the problem in a way similar to https://github.com/robotology/gz-sim-yarp-plugins/pull/253, by adding in the clock destructor a callback that is used by any robotinterface plugin to close when the clock plugin is being destroyed.

traversaro, 2025-03-17:
@LoreMoretti the issue should be fixed by https://github.com/robotology/gz-sim-yarp-plugins/pull/253 .