robotology/gz-sim-yarp-plugins #279: Update to Yarp4

randaz81, 2025-08-29:
The branch YARP4 is ready.

traversaro, 2025-08-30:
> The branch YARP4 is ready.

Thanks @randaz81 ! I adapted it in https://github.com/robotology/gz-sim-yarp-plugins/tree/yarp4support to add support for yarp 4 without breaking backward compatibility with YARP 3.12, so that we can avoid to maintain two different branches.

traversaro, 2025-09-01:
It is deprecated, but people are still using it, so I think it make sense to just easier to update it, it should be easy to do, I can handle it.

I have just one question, in the YARP 4 changelog (https://github.com/robotology/yarp/blob/8c1b42c7b4fabb777476c95f56968858881c0b61/doc/release/v4_0_0.MD?plain=1#L41) I read "Removed deprecated device `RGBDSensorClient` (replaced by RGBDSensor_nwc_yarp)". I was a bit confused by this working, I guess what happened is that `RGBDSensorClient` was renamed to `RGBDSensor_nwc_yarp`, but the device parameters are the same?

traversaro, 2025-09-01:
Ack, but given that if I am not wrong the device was never deprecated in the past (i.e., you still need to use `RGBDSensorClient` if you use YARP 3.12), perhaps writing "The `RGBDSensorClient` YARP devices was renamed to `RGBDSensor_nwc_yarp`" may be more clear? At least, it would be for me.